Koenkk/zigbee2mqtt #5869: Update firmware OTA reported true but the same fw version remains

BogdanDIA, 2021-01-23:
## What happened
I bought a Philips hue remote control, It can join w/o problems and quick. The zigbee2mqtt reported existing new firmware available for it and I triggered OTA update. The update apparently goes from 0% to 100% and eventually reports as being successful although it take like 50-60minutes. However, the version of the firmware reported after is the same as the previous one.

## What did you expect to happen
Have the new firmware version updated.

## How to reproduce it (minimal and precise)
Trigger OTA update. In the log I attach there is an error on block xxx but the process continues until 100%. So the questions are:
- why update takes so much?
- why it is not reported as an error? maybe it is an error that has been recovered and then the device should have the new firmware. 

## Debug info 
Adapter hardware: CC2652RB from Slaesh

Logs: https://pastebin.com/raw/QWPnfZen

PS: this does remember me to this ugly story from 2016 when Philips light bulbs were infected en masse and flashing SOS: https://eprint.iacr.org/2016/1047.pdf

BogdanDIA, 2021-02-24:
Any news with this issue?

Koenkk, 2021-02-24:
Known issue, repairing the device should fix it

BogdanDIA, 2021-02-28:
When I entered the initial post, I did a lot of unpair/pair cycles. However, I did again today with the latest Z2M add-on and still have the same problem:
...
info  2021-02-28 10:07:47: Update of 'RemoteControl1' at 100.00%
info  2021-02-28 10:07:47: MQTT publish: topic 'zigbee2mqtt/RemoteControl1', payload '{"battery":100,"counter":1,"linkquality":81,"update":{"progress":100,"remaining":15,"state":"updating"},"update_available":false}'
info  2021-02-28 10:07:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'RemoteControl1' at 100.00%","meta":{"device":"RemoteControl1","progress":100,"status":"update_progress"},"type":"ota_update"}'
debug 2021-02-28 10:07:49: Received Zigbee message from 'RemoteControl1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
debug 2021-02-28 10:07:49: Received Zigbee message from 'RemoteControl1', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20160302","swBuildId":"5.45.1.17846"}' from endpoint 1 with groupID 0
debug 2021-02-28 10:07:49: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20160302","swBuildId":"5.45.1.17846"}'
info  2021-02-28 10:07:49: Finished update of 'RemoteControl1', from '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}' to '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}'
info  2021-02-28 10:07:49: MQTT publish: topic 'zigbee2mqtt/RemoteControl1', payload '{"battery":100,"counter":1,"linkquality":81,"update":{"state":"idle"},"update_available":false}'
info  2021-02-28 10:07:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"RemoteControl1","transaction":"5lt0a-7"},"meta":{"device":"RemoteControl1","from":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"},"status":"update_succeeded","to":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}},"type":"ota_update"}'
info  2021-02-28 10:07:49: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20160302","software_build_id":"5.45.1.17846"},"id":"RemoteControl1","to":{"date_code":"20160302","software_build_id":"5.45.1.17846"}},"status":"ok","transaction":"5lt0a-7"}'
debug 2021-02-28 10:07:59: Received Zigbee message from 'RemoteControl1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
debug 2021-02-28 10:07:59: Check if update available for '0x0017880108f8625e' (RWL021)
debug 2021-02-28 10:07:59: Is new image available for '0x0017880108f8625e', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":265,"fileVersion":1107314102}', latest meta '{"fileVersion":1107324829,"fileSize":240760,"url":"http://fds.dc1.philips.com/firmware/ZGB_100B_0109/1107324829/Switch-ATmega_6.1.1.28573_0012.sbl-ota","sha512":"d20057d6c4e61d4ccdd3947b8a8c9bf9e126a26c080b9ce3e6839a0615b912c6f93246a60c55a44f2c8b8ec49f5041f46ac38b8a3c0a1c6ab44f062cc62089cf"}'
debug 2021-02-28 10:07:59: Update available for '0x0017880108f8625e': YES
info  2021-02-28 10:07:59: MQTT publish: topic 'zigbee2mqtt/RemoteControl1', payload '{"battery":100,"counter":1,"linkquality":84,"update":{"state":"available"},"update_available":true}'

closed 2021-04-08 by github-actions[bot]