Koenkk/zigbee2mqtt #5874: Trip(p)le action for Aqara WXKG11LM doesn't work

gurbyz, 2021-01-23:
## What happened
I don't see the `tripple` action in my logs for the Aqara WXKG11LM button device. I do see `single` and `double`, but not `tripple` nor `triple`.

I'm on HA 2021.1.4 using the 1.17.0.1 Zigbee2mqtt add-on which is using 1.17.0 of Zigbee2mqtt. I also saw [this issue report](https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/536) for the Domoticz add-on for Zigbee2mqtt, so I guess the issue is not in the add-ons but in Zigbee2mqtt.

Maybe somewhere in the code the `tripple` and `triple` spelling gets mixed up and the action gets lost?

## What did you expect to happen
I want to be able to use the `tripple` action. Or `triple`.

## How to reproduce it (minimal and precise)
I press the button quickly three times and nothing shows up in the logs. If I do it very slowly, a single action or double action or combination of them pops up.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2531, coordinator type: zStack30x
Adapter firmware version: 20190425

schurgan, 2021-01-24:
I can confirm that at Domoticz it doesn't work either. In the service of zigbee2mqtt you can see:

Jan 24 12:27:34 raspberrypi npm[17872]: Zigbee2MQTT:info  2021-01-24 12:27:34: MQTT publish: topic 'zigbee2mqtt/Multischalter Tor', payload '{"action":"triple","battery":100,"linkquality":13,"voltage":3032}'

but nothing happens in Domoticz. Maybe somewhere in the code something has been swapped with "triple" or "tripple"

Koenkk, 2021-01-24:
`tripple` was indeed a typo in the docs (the value `action` was always `triple`) which has already been fixed. Docs will be updated with next release.

In the log of @schurgan I see this action being send so probably there is also a typo in the domoticz addon, please create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin

gurbyz, 2021-01-25:
To be sure I checked again. Only actions for `single` and `double` clicks and `hold` are reported in my zigbee2mqtt logs. `triple` and also `quadruple` click don't have a line in the logs. I have 2 WXKG11LM's and checked both.

I use `legacy: false` as suggested in the docs. My raw settings look like:
```
"0x00158d000 etc": {
                "friendly_name": "bedside_button_gerben",
                "legacy": false
},
```

It's not u huge deal for me and I can live without the extra actions, but maybe I can help solving the bug. Or could I just have bought very outdated Aqara hardware that don't support the extra actions?

Koenkk, 2021-01-25:
Ah turns out triple, quadruple hold and release may not be supported for all models: https://github.com/Koenkk/zigbee-herdsman-converters/blob/4b65499e67c99d2eb88ea7222c4622a7ae444ba1/devices.js#L521 . Due to some refactoring this got lost and is not in the docs anymore. Will re-add it.

closed 2021-01-25 by Koenkk

dheuts90, 2021-05-19:
I'm sure it worked with the 3x WXKG11LM I use. I just noticed today triple and quadruple is not working anymore. I don't even see it back in the logs