Koenkk/zigbee2mqtt #5875: AC25697 LEDVANCE SMART+ CLASSIC MULTICOLOUR 60 10W E27 not reporting the warm white value

esbenr, 2021-01-23:
## What happened
Support for AC25697 LEDVANCE SMART+ CLASSIC MULTICOLOUR 60 10W E27 has a bug.
When using the white temperature scale, the bulb color doesen't change in Home Assistant. I suspect that there is something with the reporting or somehow the handler for the bulb don't support this.

The bulb has two scales. One for RGB and one for W.
[image]

When setting i.e blue color (RGB-scale/color palette):
The HA icon turns the correct color:
[image]

The log shows:
```
Zigbee2MQTT:info  2021-01-23 19:32:58: MQTT publish: topic 'zigbee2mqtt/light_markus_room', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"state":"ON"}'
Zigbee2MQTT:info  2021-01-23 19:32:58: MQTT publish: topic 'zigbee2mqtt/Markus room Ceiling light', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":45,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-23 19:32:58: MQTT publish: topic 'zigbee2mqtt/Markus room Ceiling light', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":45,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-23 19:32:59: MQTT publish: topic 'zigbee2mqtt/Markus room Ceiling light', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":45,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-23 19:33:05: MQTT publish: topic 'zigbee2mqtt/Geek corner Button 1', payload '{"battery":100,"linkquality":111,"voltage":3025}'
```
With the following attributes:
```
min_mireds: 153
max_mireds: 500
effect_list: blink, breathe, okay, channel_change, finish_effect, stop_effect
brightness: 254
color_temp: 400
hs_color: 240, 100
rgb_color: 0, 0, 255
xy_color: 0.136, 0.04
color: 
h: 240
hue: 31
s: 100
saturation: 100
x: 0.476993298233858
'y': 0.413675289942435

linkquality: 48
state: ON
update: 
state: idle

update_available: false
friendly_name: Markus room Ceiling light
supported_features: 63
```

But when setting the light warmth (W-scale/color temperature fader):

The icon in HA doesen't change to the selected color temperature, but stays at the previous RGB color:
[image]

The log shows:
```
Zigbee2MQTT:info  2021-01-23 19:37:13: MQTT publish: topic 'zigbee2mqtt/light_markus_room', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.436929833678155,"y":0.404073616886221},"color_temp":333,"state":"ON"}'
Zigbee2MQTT:info  2021-01-23 19:37:13: MQTT publish: topic 'zigbee2mqtt/Markus room Ceiling light', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.436929833678155,"y":0.404073616886221},"color_temp":333,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-23 19:37:13: MQTT publish: topic 'zigbee2mqtt/light_markus_room', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.436929833678155,"y":0.404073616886221},"color_temp":137,"state":"ON"}'
Zigbee2MQTT:info  2021-01-23 19:37:13: MQTT publish: topic 'zigbee2mqtt/Markus room Ceiling light', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.436929833678155,"y":0.404073616886221},"color_temp":137,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-23 19:37:17: MQTT publish: topic 'zigbee2mqtt/Carport Wall light shed door right', payload '{"brightness":26,"linkquality":105,"state":"ON","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2021-01-23 19:37:17: MQTT publish: topic 'zigbee2mqtt/light_markus_room', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.436929833678155,"y":0.404073616886221},"color_temp":333,"state":"ON"}'
Zigbee2MQTT:info  2021-01-23 19:37:17: MQTT publish: topic 'zigbee2mqtt/Markus room Ceiling light', payload '{"brightness":254,"color":{"h":240,"hue":31,"s":100,"saturation":100,"x":0.436929833678155,"y":0.404073616886221},"color_temp":333,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

With the following attributes:
```
min_mireds: 153
max_mireds: 500
effect_list: blink, breathe, okay, channel_change, finish_effect, stop_effect
brightness: 254
color_temp: 333
hs_color: 240, 100
rgb_color: 0, 0, 255
xy_color: 0.136, 0.04
color: 
h: 240
hue: 31
s: 100
saturation: 100
x: 0.436929833678155
'y': 0.404073616886221

linkquality: 48
state: ON
update: 
state: idle

update_available: false
friendly_name: Markus room Ceiling light
supported_features: 63
```

## What did you expect to happen
- When selecting a RGB color, the bulb color in HA changes to that color:
- When selecting a white temperature on the slider, the bulb icon doesen't change to that, but stays at the previously selected RGB color.

This means the light can be any on the warm white scale, and the status shown is the last selected color of the palette.

## How to reproduce it (minimal and precise)

1) Select a color from the palette: Light and icon in HA, switches to that color.
2) Select a color with the warm white slider: The light changes to that white nuance, but the icon in HA is not updated.

See video of the problem here: 
https://drive.google.com/file/d/12WTJR8lVsN73nBCHmmFEMeZ2c7zySI-6

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: CC26X2R1/ZZH
Adapter firmware version: CC26X2R1_20201026

esbenr, 2021-01-23:
I thin the handler for the bulb don't know when to report the warm white temperature and when to report the RGB color.

Both are in the report sent back to the coordinator, but I only think the handler reads/uses the RGB value, and not the white temperature.
There is also not indicator telling which scale should be used.

In comparison i.e. Trådfri RGB bulbs changes the RGB color when selecting a white temperature. So the white and the color are joined.
In the AC25697 these white temperature and color are like two separate scales and the zigbee2mqtt report don't indicate which is the currently used in the bulb.

Koenkk, 2021-01-24:
> There is also no indicator telling which scale should be used.

True indeed, had a discussion with a HA dev some days ago: https://github.com/home-assistant/core/issues/45145#issuecomment-762972247

The reason that the icon shows the wrong color is because there is no conversion done from hue/sat -> color temp (this is done for XY). I've implemented this now, can you apply the changes from: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2146/files ?

This should show how to find these files: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

closed 2021-03-03 by github-actions[bot]