Koenkk/zigbee2mqtt #5881: Osram Smart+ switch mini without a 'circle-hold' action

Fridolin34, 2021-01-24:
Hello everybody,

I just wanted to make my Osram Smart+ switch mini to listen/respond to a circle long press event, but this event is somehow not existing.

Regarding the documentation, there should be a normal action and a long press action (hold) for each button, but in my system, I only have a long press event for the buttons up and down. I'm using the newest release version of z2m.

[image]

My choices:
[image]

What do I miss or is the documentation outdated? Where can I check, what is available besides from the automation where I wanted to use the function?

Thanks,
Fridolin

Fridolin34, 2021-02-22:
Ok, does somebody know where this entity (or all entities) is defined in z2m? I mean, if the device supports the circle_hold event, it's just like another attribute to be added including getter/setter, or?

hoggerz, 2021-02-23:
Hi Fridolin34,

I have a few of these and I had to physically perform each possible action on the remote before home assistant picked up each one. Try long pressing the circle button and trying again. You might have to restart either or both zigbee2mqtt and home assistant after. I’m using the latest versions of both, so I don’t know if there’s anything else that might effect your setup.

hoggerz

Fridolin34, 2021-02-24:
Hi hoggerz,

thanks for you answer and long story short, it worked. Just checked and the action was not there, did a long press on the circle, simply reloaded the page and I'm now able to create an automation using 'circle hold'.

This is very weird. I'm pretty sure that I tried this so many times and I was expecting that the code needs to be changed in order to provide the method to respond to this specific event. And yes, maybe something change meanwhile and I did my first tries 2 months ago.

But okay, I won't complain.

Thanks for you help.

Best regards,
Fridolin

closed 2021-02-24 by Fridolin34