Koenkk/zigbee2mqtt #5884: Issue adding support for door lock

stolevegen, 2021-01-24:
## What happened

- Paired EasyAccess EasyCodeTouch to CC2531 running Zigbee2mqtt headless on Raspberry Pi. 
- Added config to node_modules/zigbee-herdsman-converters/devices.js as well as devices.js at /opt/zigbee2mqtt/data
- Set log_level: debug
- Got "no converter" messages, but only for information like No converter available for 'easyCodeTouch_v1' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":11}'
- and "No converter available for 'state' (LOCK)"
- Added one converter. 
- No function, battery and lqi "unknown"
- Removed device from network and database. 
- Added zigbee_shepherd_devices: true to data/configuration.yaml 
- Paired device. Log attached. https://pastebin.com/1WUfWRSw . Not supported

## What did you expect to happen

Expected support to be added, and function of sensors and lock action. 

## How to reproduce it (minimal and precise)

Pair EasyAccess EasyCodeTouch

## Debug info
Zigbee2MQTT version: latest, edge. 1.7
Adapter hardware: CC2531
Adapter firmware version: 20190608

stolevegen, 2021-01-31:
I've been trying to modify devices.js to something that works.

At least now I can use "UNLOCK" command from HA frontend, and the door lock unlocks. However, it runs double, so "unlock --> waits x seconds --> locks --> waits x seconds --> unlocks --> waits x seconds --> locks"
Then error in log. 
Battery, link quality and lock status does not update in HA frontend/lovelace. 

The lock has a function where you hold one button and it stays unlocked until button is pushed again. I will try to sniff the traffic while activating this function.  

First pair log : https://pastebin.com/kAaAFYY4
2nd pair log : https://pastebin.com/ERbifWsa
After "10 sec unlock" runs double : https://pastebin.com/B0fmc3f5

From devices.js : https://pastebin.com/BPeZBhSu

stolevegen, 2021-02-03:
Adding user manual describing implementation details, clusters etc. Note ; 

The Lock/Unlock command does not support PIN code in the command, and if a Pin Code parameter is present, it can only have the parameter Length and it must be set to 0. 

[user manual.pdf](https://github.com/Koenkk/zigbee2mqtt/files/5921232/user.manual.pdf)
@Koenkk

Koenkk, 2021-02-04:
Can you share what you've added to devices.js now?

stolevegen, 2021-02-04:
This is what I'm working with now 

`    {
        zigbeeModel: ['easyCodeTouch_v1'],
        model: 'EasyCode Touch',
        vendor: 'EasyAccess',
        description: 'EasyCode Touch',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
        toZigbee: [tz.lock],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(11);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.lock(), e.battery()],
    },`

I've also tried with the two following without much luck ; 

`    {
        zigbeeModel: ['easyCodeTouch_v1'],
        model: 'EasyCode Touch',
        vendor: 'EasyAccess',
        description: 'EasyCode Touch',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery, fz.lock_programming_event, fz.lock_pin_code_response],
        toZigbee: [tz.lock, tz.pincode_lock],
        meta: {configureKey: 4, pinCodeCount: 30},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(11);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        // Note - Keypad triggered deletions do not cause a zigbee event, though Adds work fine.
        onEvent: async (type, data, device) => {
            // When we receive a code updated message, lets read the new value
            if (data.type === 'commandProgrammingEventNotification' &&
                data.cluster === 'closuresDoorLock' &&
                data.data &&
                data.data.userid !== undefined &&
                // Don't read RF events, we can do this with retrieve_state
                (data.data.programeventsrc === undefined || constants.lockSourceName[data.data.programeventsrc] != 'rf')
            ) {
                await device.endpoints[0].command('closuresDoorLock', 'getPinCode', {userid: data.data.userid}, {});
            }
        },
        exposes: [e.lock(), e.battery()],
    },

    {
        zigbeeModel: ['easyCodeTouch_v1'],
        model: 'EasyCode Touch',
        vendor: 'EasyAccess',
        description: 'EasyCode Touch',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
        toZigbee: [tz.lock],
        meta: {configureKey: 3},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(11);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.lock(), e.battery()],
    },`

Koenkk, 2021-02-04:
So what is logged in the debug log once you send `{"state": "LOCK"}` after pairing with the config from: https://pastebin.com/JdTxcSnM

stolevegen, 2021-02-04:
@Koenkk 
"log_level: debug" log for command : https://pastebin.com/f8Ymb8v3
"log_level: debug" log in full : https://pastebin.com/xungGu7e
"DEBUG=zigbee-herdsman* npm start" log for command : https://pastebin.com/wDfpFyRP 

The events are in this order ; 
- I send {"state": "LOCK"}
- Lock responds by sound indication and actuates
- delay
- Lock responds by sound indication and actuates
- delay
- log: device announcement
- log: error

Koenkk, 2021-02-05:
What if you change https://github.com/Koenkk/zigbee-herdsman-converters/blob/f8029fc9a4117c26de60d15c6e87da9e315c6b77/converters/toZigbee.js#L72 to `{disableResponse: true, disableDefaultResponse: true}`

stolevegen, 2021-02-05:
No error message, but the lock responds four times instead of two.

Koenkk, 2021-02-06:
@stolevegen can you share a sniff when sending the lock command **once**, I want to check how many times the lock command is actually send. If it is only send 1 it seems this locks communicates in a non-standard way and sniffing the traffic with the original gateway is needed.

stolevegen, 2021-02-07:
@Koenkk 
Attached is log when sending lock command once.

I think you might be right about non-standard communication. I will reach out to manufacturer. 

[lock command ONCE.zip](https://github.com/Koenkk/zigbee2mqtt/files/5939265/lock.command.ONCE.zip)

Koenkk, 2021-02-07:
@stolevegen this sniff shows that indeed only one lock command is send.

stolevegen, 2021-02-08:
I spoke with the manufacturer and they say there is no "secret" key or non-standard communication. Should be open and ready to go for any zigbee system. 

Any questions I could ask the manufacturer to help move it along? I'm pretty much stuck.

Koenkk, 2021-02-08:
@stolevegen sniffing traffic with the original gateway would be the next thing.

stolevegen, 2021-02-08:
I've been trying both with deconz, deconz-dev and z2m. My current systems, as well as with fresh installs.
It seems the only packets I pick up "doorlock --> z2m" is the following ; 

`Zigbee2MQTT:debug 2021-01-31 20:16:38: Received Zigbee message from '0xf4ce3632a29609ab', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16843009,"imageType":321,"manufacturerCode":123}' from endpoint 11 with groupID 0`

I suppose I could take the lock out of the door, move everything off-site, reset zigbee module, do a fresh Home Assistant install, do a fresh z2m install, fresh CC2531 firmware and start from scratch. Not sure it would be beneficial though?  

I think the original gateway is still in the making.

Koenkk, 2021-02-09:
I don't see how that will make a difference. I expect either the firmware is bugged or non-standard communication is used.

closed 2021-03-19 by github-actions[bot]

stolevegen, 2021-03-22:
Problem was hardware related, bent pin at contact. Fixed now and should see normal operation

stolevegen, 2021-03-22:
@Koenkk The problems I had turned out to be a hardware issue that has been resolved now. I have updated the title. Zigbee module should be same as in other locks from EasyAccess. Can you reopen issue? 

I have unlock/lock and battery percentage successfully integrated. 

I'm missing Operation Notification Response/Lock Operation Event to reflect if the lock was opened by keypad, RFID tag or Zigbee. 
The module also supports ;
- setting AutoRelockTime (0x0023) on(0x01) or off(0x00).
- setting SoundVolume (0x0024) off(0x00), low(0x01) or high(0x02) 

If you see the logs attached, Operation Notification Response/Lock Operation Event will return this or similar ; 
type 'raw', cluster 'closuresDoorLock', data '{"data":[25,5,32,0,2,0,0,0,0,0,0],"type":"Buffer"}

This is which bit corresponds to unlock/lock source ; 
Unlock keypad will return [25,5,32,**0**,2,0,0,0,0,0,0]  _where fourth bit == "0"_
Unlock RFID tag will return [25,12,32,**3**,2,0,0,0,0,0,0] _where fourth bit == "3"_
Unlock Zigbee will return [25,39,32,255,14,0,0,0,0,0,0] _where fifth bit == "14"_
Lock powerbutton/zigbee will return [25,8,32,255,13,0,0,0,0,0,0] _where fifth bit == "13"_

When trying to expose the event to Home Assistant, I tried with following "fromZigbee.js" https://pastebin.com/9EbFgN8Z and following "devices.js" https://pastebin.com/4x1HY367 . I get lookup not defined. 

Herdsman log (“Lock by home assistant”)
https://pastebin.com/946nGkKY

Herdsman log (“Unlocking by home assistant”)
https://pastebin.com/9QnfX3E0

Herdsman log (“Unlocking by RFID tag”)
https://pastebin.com/DiXWVXKj

Herdsman log (“Unlocking by KeyPad”)
https://pastebin.com/fGGp0GkU