Koenkk/zigbee2mqtt #5895: RGBgenie ZB-3009

whoisdecoy, 2021-01-25:
This device is very similar to https://www.zigbee2mqtt.io/devices/ZGRC-KEY-013.html. It even has the correct picture for RGBgenie ZB-3009, but incorrect for ZGRC-KEY-013. Although this link shows the similar device and mentions the same model id https://zigbee.blakadder.com/RGB_Genie_ZB-3007.html.

https://rgbgenie.com/wp-content/uploads/2019/05/ZB-3009-user-manual.pdf
Page 8 shows the output clusters.

I am running zigbee2 on hassio. After adding to the config and putting devices.js in the data_path according to https://github.com/danielwelch/hassio-zigbee2mqtt#using-devicesjs-override-in-add-on it throws a bunch of errors for missing modules.

`> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js
internal/modules/cjs/loader.js:968
  throw err;
  ^
Error: Cannot find module './lib/light'
Require stack:
- /zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman-converters/devices.js
- /zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman-converters/index.js
- /zigbee2mqtt-1.17.0/lib/zigbee.js
- /zigbee2mqtt-1.17.0/lib/controller.js
- /zigbee2mqtt-1.17.0/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:965:15)
    at Function.Module._load (internal/modules/cjs/loader.js:841:27)
    at Module.require (internal/modules/cjs/loader.js:1025:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman-converters/devices.js:44:15)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
    at Module.load (internal/modules/cjs/loader.js:985:32)
    at Function.Module._load (internal/modules/cjs/loader.js:878:14)
    at Module.require (internal/modules/cjs/loader.js:1025:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman-converters/devices.js',
    '/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman-converters/index.js',
    '/zigbee2mqtt-1.17.0/lib/zigbee.js',
    '/zigbee2mqtt-1.17.0/lib/controller.js',
    '/zigbee2mqtt-1.17.0/index.js'
  ]
}
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

`    {
    zigbeeModel: ['RGBgenie ZB-3009'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'RGBgenie ZB-3009', // Vendor model number, look on the device for a model number
    vendor: 'RGB Genie', // Vendor of the device (only used for documentation and startup logging)
    description: '3 Scene Color Touch Panel and Dimmer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [], // Defines what this device exposes, used for e.g. Home Assistant discovery
},`

I added

## data/database.db entry of the device
`{"id":2,"type":"Router","ieeeAddr":"0x000d6f0014cbe75d","nwkAddr":9049,"manufId":4644,"manufName":"RGBgenie","powerSource":"Mains (single phase)","modelId":"RGBgenie ZB-3009","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"RGBgenie ZB-3009","manufacturerName":"RGBgenie","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.3.1_r30"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.3.1_r30","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1611540748151}`

Koenkk, 2021-01-25:
Are you on the edge version of the addon?

whoisdecoy, 2021-01-25:
No I am not on the edge version

Koenkk, 2021-01-25:
Try switching to that. `./lib/light` is new.

whoisdecoy, 2021-01-25:
OK that did that trick. Here is the debug output from buttons, starting from bottom. This is where the documentation loses me with the converters. Can you please assist?

`Zigbee2MQTT:debug 2021-01-25 13:02:20: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":3}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:02:20: No converter available for 'RGBgenie ZB-3009' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":3}'
Zigbee2MQTT:debug 2021-01-25 13:02:29: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":2}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:02:29: No converter available for 'RGBgenie ZB-3009' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":2}'
Zigbee2MQTT:debug 2021-01-25 13:02:33: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:02:33: No converter available for 'RGBgenie ZB-3009' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":1}'
Zigbee2MQTT:debug 2021-01-25 13:04:25: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveHue', cluster 'lightingColorCtrl', data '{"movemode":1,"rate":28}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:04:25: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveHue' and data '{"movemode":1,"rate":28}'
Zigbee2MQTT:debug 2021-01-25 13:04:33: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":21823,"colory":21823,"transtime":0}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:04:33: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveToColor' and data '{"colorx":21823,"colory":21823,"transtime":0}'
Zigbee2MQTT:debug 2021-01-25 13:04:50: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:04:50: No converter available for 'RGBgenie ZB-3009' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":1}'
Zigbee2MQTT:debug 2021-01-25 13:05:03: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":1,"rate":50}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:05:03: No converter available for 'RGBgenie ZB-3009' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":1,"rate":50}'
Zigbee2MQTT:debug 2021-01-25 13:05:04: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:05:04: No converter available for 'RGBgenie ZB-3009' with cluster 'genLevelCtrl' and type 'commandStopWithOnOff' and data '{}'
Zigbee2MQTT:debug 2021-01-25 13:05:09: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-01-25 13:05:10: No converter available for 'RGBgenie ZB-3009' with cluster 'genOnOff' and type 'commandOff' and data '{}'
Zigbee2MQTT:debug 2021-01-25 13:05:18: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-01-25 13:05:18: No converter available for 'RGBgenie ZB-3009' with cluster 'genOnOff' and type 'commandOn' and data '{}'
Zigbee2MQTT:debug 2021-01-25 13:05:23: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":304,"transtime":0}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:05:23: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveToColorTemp' and data '{"colortemp":304,"transtime":0}'
Zigbee2MQTT:debug 2021-01-25 13:05:31: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":450,"transtime":0}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:05:31: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveToColorTemp' and data '{"colortemp":450,"transtime":0}'
Zigbee2MQTT:debug 2021-01-25 13:05:36: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":17891,"colory":46989,"transtime":0}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:05:36: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveToColor' and data '{"colorx":17891,"colory":46989,"transtime":0}'
Zigbee2MQTT:debug 2021-01-25 13:05:45: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":48103,"colory":17367,"transtime":0}' from endpoint 1 with groupID 22785
[... cut ...]
Zigbee2MQTT:debug 2021-01-25 13:06:29: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveToColor' and data '{"colorx":44105,"colory":15532,"transtime":8}'
Zigbee2MQTT:debug 2021-01-25 13:06:30: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":13631,"colory":18743,"transtime":0}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-25 13:06:30: No converter available for 'RGBgenie ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveToColor' and data '{"colorx":13631,"colory":18743,"transtime":0}'`

Koenkk, 2021-01-26:
You can take the type to find the matching `fromZigbee` converters. e.g.:

```
Zigbee2MQTT:debug 2021-01-25 13:02:20: No converter available for 'RGBgenie ZB-3009' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":3}'
```

-> 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/d751f07919cc1122da3f0ac75053a7ba0cfe137b/converters/fromZigbee.js#L846

So this would result in `fromZigbee: [fz.command_recall]`

whoisdecoy, 2021-01-26:
Ok so I believe I added all the required converters. Here is the updated device code:

`    {
    zigbeeModel: ['RGBgenie ZB-3009'],
    model: 'RGBgenie ZB-3009',
    vendor: 'RGB Genie',
    description: '3 scene remote and dimmer ',
    fromZigbee: [fz.command_recall, fz.command_move_hue, fz.command_move,fz.command_stop, fz.command_on, fz.command_off, fz.command_move_to_color_temp, fz.command_move_to_color],
    toZigbee: [],
    exposes: [],
},`

and some examples from logs:

`Zigbee2MQTT:debug 2021-01-26 16:33:11: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":50}' from endpoint 1 with groupID 22785
Zigbee2MQTT:info  2021-01-26 16:33:11: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"brightness_move_up","action_group":22785,"action_rate":50,"linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:11: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"","linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:11: MQTT publish: topic 'homeassistant/device_automation/0x000d6f0014cbe75d/action_brightness_move_up/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x000d6f0014cbe75d"],"manufacturer":"RGB Genie","model":"3 scene remote and dimmer  (RGBgenie ZB-3009)","name":"0x000d6f0014cbe75d","sw_version":"Zigbee2MQTT 1.17.0-dev"},"payload":"brightness_move_up","subtype":"brightness_move_up","topic":"zigbee2mqtt/0x000d6f0014cbe75d/action","type":"action"}'
Zigbee2MQTT:info  2021-01-26 16:33:11: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d/action', payload 'brightness_move_up'
Zigbee2MQTT:debug 2021-01-26 16:33:11: Received MQTT message on 'homeassistant/device_automation/0x000d6f0014cbe75d/action_brightness_move_up/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x000d6f0014cbe75d"],"manufacturer":"RGB Genie","model":"3 scene remote and dimmer  (RGBgenie ZB-3009)","name":"0x000d6f0014cbe75d","sw_version":"Zigbee2MQTT 1.17.0-dev"},"payload":"brightness_move_up","subtype":"brightness_move_up","topic":"zigbee2mqtt/0x000d6f0014cbe75d/action","type":"action"}'
Zigbee2MQTT:debug 2021-01-26 16:33:27: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 22785
Zigbee2MQTT:info  2021-01-26 16:33:27: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"recall_1","action_group":22785,"linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:27: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"","linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:27: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d/action', payload 'recall_1'
Zigbee2MQTT:debug 2021-01-26 16:33:32: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 22785
Zigbee2MQTT:info  2021-01-26 16:33:32: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"recall_1","action_group":22785,"linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:32: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"","linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:32: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d/action', payload 'recall_1'
Zigbee2MQTT:debug 2021-01-26 16:33:42: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":1,"rate":50}' from endpoint 1 with groupID 22785
Zigbee2MQTT:info  2021-01-26 16:33:42: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"brightness_move_down","action_group":22785,"action_rate":50,"linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:42: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d', payload '{"action":"","linkquality":116}'
Zigbee2MQTT:info  2021-01-26 16:33:42: MQTT publish: topic 'homeassistant/device_automation/0x000d6f0014cbe75d/action_brightness_move_down/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x000d6f0014cbe75d"],"manufacturer":"RGB Genie","model":"3 scene remote and dimmer  (RGBgenie ZB-3009)","name":"0x000d6f0014cbe75d","sw_version":"Zigbee2MQTT 1.17.0-dev"},"payload":"brightness_move_down","subtype":"brightness_move_down","topic":"zigbee2mqtt/0x000d6f0014cbe75d/action","type":"action"}'
Zigbee2MQTT:info  2021-01-26 16:33:42: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d/action', payload 'brightness_move_down'
Zigbee2MQTT:debug 2021-01-26 16:33:42: Received MQTT message on 'homeassistant/device_automation/0x000d6f0014cbe75d/action_brightness_move_down/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x000d6f0014cbe75d"],"manufacturer":"RGB Genie","model":"3 scene remote and dimmer  (RGBgenie ZB-3009)","name":"0x000d6f0014cbe75d","sw_version":"Zigbee2MQTT 1.17.0-dev"},"payload":"brightness_move_down","subtype":"brightness_move_down","topic":"zigbee2mqtt/0x000d6f0014cbe75d/action","type":"action"}'
Zigbee2MQTT:debug 2021-01-26 16:33:46: Received Zigbee message from '0x000d6f0014cbe75d', type 'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 22785
[... cut ...]
Zigbee2MQTT:info  2021-01-26 16:34:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014cbe75d/action', payload 'on'
Zigbee2MQTT:debug 2021-01-26 16:34:53: Received MQTT message on 'homeassistant/device_automation/0x000d6f0014cbe75d/action_on/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x000d6f0014cbe75d"],"manufacturer":"RGB Genie","model":"3 scene remote and dimmer  (RGBgenie ZB-3009)","name":"0x000d6f0014cbe75d","sw_version":"Zigbee2MQTT 1.17.0-dev"},"payload":"on","subtype":"on","topic":"zigbee2mqtt/0x000d6f0014cbe75d/action","type":"action"}'`

Koenkk, 2021-01-27:
I've added the device

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Refer to https://www.zigbee2mqtt.io/integration/home_assistant.html#responding-to-button-clicks on how to do something with the actions.

closed 2021-01-27 by Koenkk

whoisdecoy, 2021-01-27:
Thank you. What about the other fields; eposes, meta, configure?
```
exposes: [e.battery(), e.action(['brightness_move_up', 'brightness_move_down', 'brightness_stop', 'on', 'off', 'recall_*'])],
toZigbee: [],
meta: {configureKey: 1, multiEndpoint: true, battery: {dontDividePercentage: true}},
configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
```

Koenkk, 2021-01-27:
@whoisdecoy I've added those also (based on other RGB genie devices)

whoisdecoy, 2021-01-27:
### I missed a hold function on the white circle button. It returns the following logs:

```
Zigbee2MQTT:debug 2021-01-27 16:29:03: Received Zigbee message from 'RGBWallPanel_Basment_Bedroom', type 'commandMoveColorTemp', cluster 'lightingColorCtrl', data '{"maximum":450,"minimum":155,"movemode":1,"rate":60}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-27 16:29:03: No converter available for 'ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveColorTemp' and data '{"maximum":450,"minimum":155,"movemode":1,"rate":60}'
Zigbee2MQTT:debug 2021-01-27 16:29:09: Received Zigbee message from 'RGBWallPanel_Basment_Bedroom', type 'raw', cluster 'lightingColorCtrl', data '{"data":[1,67,71,0,0],"type":"Buffer"}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-27 16:29:09: No converter available for 'ZB-3009' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[1,67,71,0,0],"type":"Buffer"}'
Zigbee2MQTT:debug 2021-01-27 16:46:25: Received Zigbee message from 'RGBWallPanel_Basment_Bedroom', type 'commandMoveColorTemp', cluster 'lightingColorCtrl', data '{"maximum":450,"minimum":155,"movemode":1,"rate":60}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-27 16:46:25: No converter available for 'ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveColorTemp' and data '{"maximum":450,"minimum":155,"movemode":1,"rate":60}'
Zigbee2MQTT:debug 2021-01-27 16:46:29: Received Zigbee message from 'RGBWallPanel_Basment_Bedroom', type 'raw', cluster 'lightingColorCtrl', data '{"data":[1,28,71,0,0],"type":"Buffer"}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-27 16:46:29: No converter available for 'ZB-3009' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[1,28,71,0,0],"type":"Buffer"}'
Zigbee2MQTT:debug 2021-01-27 16:46:44: Received Zigbee message from 'RGBWallPanel_Basment_Bedroom', type 'commandMoveColorTemp', cluster 'lightingColorCtrl', data '{"maximum":450,"minimum":155,"movemode":3,"rate":60}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-27 16:46:44: No converter available for 'ZB-3009' with cluster 'lightingColorCtrl' and type 'commandMoveColorTemp' and data '{"maximum":450,"minimum":155,"movemode":3,"rate":60}'
Zigbee2MQTT:debug 2021-01-27 16:46:47: Received Zigbee message from 'RGBWallPanel_Basment_Bedroom', type 'raw', cluster 'lightingColorCtrl', data '{"data":[1,30,71,0,0],"type":"Buffer"}' from endpoint 1 with groupID 22785
Zigbee2MQTT:debug 2021-01-27 16:46:47: No converter available for 'ZB-3009' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[1,30,71,0,0],"type":"Buffer"}'
```

The `data '{"maximum":450,"minimum":155,"movemode":1,"rate":60}'` is sent upon holding. The following message is sent upon release: `data '{"data":[1,30,71,0,0],"type":"Buffer"`

Referencing the manual it either adjust color temp or white channel. In my case, I am using a RGBWw light so it would be saturation and the following converter would need to be added. Although that does not match the type: commandMoveColorTemp

```
fz.command_step_saturation
```

### Also, because this is a wall panel wired into mains it does not have a battery. Here is what it is in logs when devices is configured.
```
Zigbee2MQTT:info  2021-01-27 16:49:07: Configuring 'RGBWallPanel_Basment_Bedroom'
Zigbee2MQTT:error 2021-01-27 16:49:08: Failed to configure 'RGBWallPanel_Basment_Bedroom', attempt 4 (TypeError: ConfigureReporting 0x000d6f0014cbe75d/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (payload is not iterable)
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:194:28)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:419:26)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
Zigbee2MQTT:info  2021-01-27 16:49:08: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"RGBWallPanel_Basment_Bedroom"},"error":"Failed to configure (ConfigureReporting 0x000d6f0014cbe75d/1 genPowerCfg([{\"attribute\":\"batteryPercentageRemaining\",\"minimumReportInterval\":3600,\"maximumReportInterval\":62000,\"reportableChange\":0}], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (payload is not iterable))","status":"error","transaction":"enu55-1"}'
```

Koenkk, 2021-01-28:
Fixed both.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)