Koenkk/zigbee2mqtt #5901: Zbmini Power On mode

knowsnotmuch, 2021-01-25:
## Is your feature request related to a problem? Please describe
When the zbmini powers on, it is by default set to off. This is frustrating if there is a power cut. Or if you want to toggle the power state via a wall switch. 

## Describe the solution you'd like
Option to set the power state of switches when they boot-up / power on 

## Describe alternatives you've considered
Node Red rule to set the state. But this creates dependency on one more service and I would prefer that this be localised to the zigbee stack to cut out external dependencies 

## Additional context
I am not sure if zbmini supports PowerOn OnOff - as per[ this link ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3632)it appears to but cant get it to work

sjorge, 2021-01-25:
I don't own the device, but we can try to see if it support the genOnoff startup state....

Update devices.js entry to this.

```
    {
        zigbeeModel: ['01MINIZB'],
        model: 'ZBMINI',
        vendor: 'SONOFF',
        description: 'Zigbee two way smart switch',
        extend: preset.switch(),
        toZigbee: preset.light_onoff_brightness_colortemp_colorxy().toZigbee.concat([tz.power_on_behavior]),
        fromZigbee: preset.light_onoff_brightness_colortemp_colorxy().fromZigbee.concat([fz.power_on_behavior]),
        exposes: [
            e.switch(), exposes.enum('power_on_behavior', ea.ALL, ['off', 'previous', 'on'])
                .withDescription('Controls the behaviour when the device is powered on'),
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            // Has Unknown power source: https://github.com/Koenkk/zigbee2mqtt/issues/5362, force it here.
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
```

knowsnotmuch, 2021-01-26:
@sjorge thank you for the suggestion. I have a problem with my rpi at the moment. I am trying to get it replaced. Will try soon as I can. But really appreciate the prompt response and suggestion. Thank you!

knowsnotmuch, 2021-02-02:
@sjorge - sorry about the delay. 

I tried this but got the error below and zigbee2mqtt would not start 

```
/app/node_modules/zigbee-herdsman-converters/devices.js:12629,
        extend: preset.switch(),,
                             ^,
TypeError: preset.switch is not a function,

```

sjorge, 2021-02-02:
What version are you running? I remember presets being switched to functions recently ish, so it might be that the version is too old.

knowsnotmuch, 2021-02-02:
> What version are you running? I remember presets being switched to functions recently ish, so it might be that the version is too old.

using the docker version - I presume it is the latest 1.17.1

sjorge, 2021-02-03:
Are you sure you are on 1.17.1? That was release 2 days ago and should definitely be new enough.

knowsnotmuch, 2021-02-03:
> Are you sure you are on 1.17.1? That was release 2 days ago and should definitely be new enough.

This is what I see: 

> {"commit":"99274c7","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":15,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.1"}

Let me try again to insert those lines.

knowsnotmuch, 2021-02-03:
I tried again and this time I had no error! Sorry about that. 

Now what am I looking for to confirm this made it through @sjorge ? Also, how can I tell if it supports the genOnoff startup status?

knowsnotmuch, 2021-02-03:
@sjorge does not seem to be supported: 

`Error Publish 'set' 'power_on_behavior' to 'DenFanZ' failed: 'Error: Write 0x847127fffe95f41a/1 genOnOff({"startUpOnOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
`

sjorge, 2021-02-03:
Yeah if you get UNSUPPORTED_ATTRIBUTE, it's not support :(

knowsnotmuch, 2021-02-03:
@sjorge yes - it does look like it is implemented but it does not work. Does that mean that this is a bug? or is the original link I sent wrong?

sjorge, 2021-02-03:
z2m is writing the correct attribute, but the device is saying, no, I don't do that.

So the screenshot might be wrong, or it could be readonly instead of rw... hard to say.

knowsnotmuch, 2021-02-03:
Ok - thanks @sjorge. Is there anyway I can write a nodered rule which picks up that a device booted up and set it to power on? Is there some way I can get the "powered up" status ? 

Again, thanks for the help and for explaining things. Much appreciate it.

sjorge, 2021-02-03:
I think you can trigger a read of 'state' if z2m publishes to `/available` with true, if the result is off, you can then set it to on?

knowsnotmuch, 2021-02-03:
Ok - will check that. I just saw one more screenshot. Here also, 0x4003 shows as rw: 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4016#issue-774313669

knowsnotmuch, 2021-02-07:
> I think you can trigger a read of 'state' if z2m publishes to `/available` with true, if the result is off, you can then set it to on?

@sjorge what topic(s) will these show up on? I am not able to spot any message (reliably) to detect device power

sjorge, 2021-02-07:
They get published to `zigbee2mqtt/<friendly_name>/available`, I just subscribe to those, if it goes from offline -> online I trigger a `/get/state` from node-red and set a flow var to indicate I need to check state == ON on the next update.

Then I have a different node that subscribes to `zigbee2mqtt/<friendly_name>` and when the flow var is set and state != ON, I send a `set/state` with payload of ON.

knowsnotmuch, 2021-02-08:
thanks for the detailed explanation @sjorge - much appreciate it!

knowsnotmuch, 2021-02-12:
I am trying this, but it looks like there will be a delay to the extent of availability_timeout - is that correct?

I thought this will generate too much traffic and limited it to 60s and the specific device. Is it ok to set a time of say 1s ?

closed 2021-03-23 by github-actions[bot]