Koenkk/zigbee2mqtt #5902: Danalock V3 BTZB stops reacting after being idle

HASSHUN, 2021-01-25:
## What happened

Even though my Danalock V3 is configured properly : -
  - Interviewing is succesful (I see battery  and lqi info as well)
  - Converter binding is OK.
  - Can turn the lock by web UI -

after being idle for few minutes it stops working at all and only comes back when I replace the battery.
- Please check my debug info: https://pastebin.com/BQTDhDrV

_!Important!
The device is not mounted on any lock lately._ 

I assume calibration might have sg to do with this issue ( as I read in a former topic #4230 ) 
Currently automatic calibration turns the lock a few kilometres :) , but when it stops it seems to work properly.

I'm quite a newbie and I guess that the solution is something trivial, but didn't find this exact case on the forum. 

## What did you expect to happen

- It should maintain the connection as the device is listed as supported and worked for many before.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.0.1 (HomeAssistant)
Adapter hardware: Conbee II
Adapter firmware version:  v2.7.0

smulle48, 2021-01-26:
If needed, I got the tech doc for the lock, with all available commands etc :-)

vytautasgimbutas, 2021-01-31:
I'm having maybe similar issue, but mine Danalock V3 stops working after a couple of weeks. Just updated from it's firmware 0.16.0 from 0.16.1 but yet not sure if that fixes the problem. 

- lock works via Danalock app
- restarting zigbee2mqtt doesn't help. Taking out batteries and putting them back in helps.
- every night it reconfigures (i.e. `mqtt publish to zigbee2mqtt/bridge/configure`). I don't recall exactly why i've done that but I think there was some another issue and doing this helped. I'll try disabling it.
- after updating it started working. Probably, lock restarted itself.

Any ideas are very welcome! :)

Varming73, 2021-02-03:
Just noted these, issue #6051 seems to be related.

Varming73, 2021-02-03:
@vytautasgimbutas could you maybe test if this works for you too: https://github.com/Koenkk/zigbee2mqtt/issues/6051#issuecomment-772585233

vytautasgimbutas, 2021-02-04:
> @vytautasgimbutas could you maybe test if this works for you too: [#6051 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/6051#issuecomment-772585233)

No, it doesn't. I get an error: `afi_status(rawValue: 8) Domain: DMI`. Only restarting lock helps. :(

Varming73, 2021-02-04:
So when you re-activate Zigbee in the Danalock app you see this error in Zigbee2mqtt or ?

vytautasgimbutas, 2021-02-04:
> So when you re-activate Zigbee in the Danalock app you see this error in Zigbee2mqtt or ?

Danalock app. By restarting I meant the lock itself by taking out the batteries :)

HASSHUN, 2021-02-12:
Is there any update in this topic? 
The lock is working properly unless fro some reason the host restarts or Danalock suddenly loses the status. 
For the first cause I set an automation to reconfigure the lock anytime HA restarts, but does not seem to help much. Regarding the second I have absulotely no clue.

philibertc, 2021-03-10:
@HASSHUN Koenkk has updated the "solution" link.

HASSHUN, 2021-03-28:
@philibertc Nothing got changed, reconfiguring can help for a day or two, but once it loses the state it's not possible to reconfigure without battery replacement. Did the rest of you experienced any improvement?

philibertc, 2021-03-28:
Hello @HASSHUN 
Yes, I had the problem with an outlet that could be toggled from a built-in button that did not update its status when toggled with the built-in button. Reconfiguring solved the problem.
You can create an automation on HomeAssistant that updates the status every minute for example. I can paste the yaml of the automation here if you want.

HASSHUN, 2021-03-29:
Hi @philibertc 
I am using a NodeRed inject node which repeats the status update to mqtt in every 30 secs. Still don't think that it's a very elegant solution. And still, kinda randomly, sudden status loss can happen.

BillyMichael, 2021-04-04:
@philibertc yes please

philibertc, 2021-04-06:
@BillyMichael Here is the action to execute: You need one per device (entity_name is the name of the device for which you want to update the status). 
```yaml
service: mqtt.publish
data:
  topic: zigbee2mqtt/entity_name/get
  payload: '{"state": ""}'
  ```

closed 2021-05-14 by github-actions[bot]