Koenkk/zigbee2mqtt #5909: TuYa TS0044 sends payload twice, and hence toggle action toggles twice

james-fry, 2021-01-25:
Rules (don't ignore these, your issue will be closed without further notice):
- English only
- Make sure you are on the latest Zigbee2MQTT version
- Provide a clear description of the problem
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.
- Make sure you are running the latest firmware https://github.com/koenkk/z-stack-firmware.

Did you read the FAQ? **Yes**
- https://www.zigbee2mqtt.io/information/FAQ.html

Zigbee2MQTT fails to start? **No**
- https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

Having issues when using a CC2531? **No**
- Make sure the CC2531 is connected through a USB extension cable
- Try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- With larger networks (30/40+) devices, your CC2531 might not be powerful enough. This will cause weird issues, in this case it's advised to use a more powerful adapter: https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1

Unsupported device? **No**
- https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Device does not pair or interview fails? **No**
- https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

Bug report? 
- If applicable, provide steps how to reproduce the problem.
- Provide the herdsman debug logging: https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

-->

## What happened

The TuYa TS0044 has 4 x momentary push switches supporting single, double and hold actions.
(https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044)
On any button event I am getting the action payload published to HA twice:

Zigbee2MQTT:info  2021-01-25 22:19:33: MQTT publish: topic 'zigbee2mqtt_cc1352p/4 Button Scene Switch', **payload '{"action":"2_single"**,"battery":100,"last_seen":"2021-01-25T22:19:33+00:00","linkquality":30,"voltage":3000}'
Zigbee2MQTT:info  2021-01-25 22:19:33: MQTT publish: topic 'zigbee2mqtt_cc1352p/4 Button Scene Switch', payload '{"action":"","battery":100,"last_seen":"2021-01-25T22:19:33+00:00","linkquality":30,"voltage":3000}'
Zigbee2MQTT:info  2021-01-25 22:19:33: MQTT publish: topic 'zigbee2mqtt_cc1352p/**4 Button Scene Switch/action', payload '2_single'**

## What did you expect to happen

Payload is sent once so that HA entity updates once

## How to reproduce it (minimal and precise)

Click any button and observe logs
I can also see 2 x action events being sent to node-red by HA for a single button click from sensor.0xbc33acfffe28ba98_action

## Debug info
Zigbee2MQTT version: HASSIO 1.17.0.1
Adapter hardware: ,  CC1352P-2
Adapter firmware version: not sure but can check if needed...

Koenkk, 2021-01-26:
This is expected when `homeassistant: true` is set. Notice that one action is send to `zigbee2mqtt_cc1352p/4 Button Scene Switch` for: https://www.zigbee2mqtt.io/integration/home_assistant.html#via-home-assistant-entity , `zigbee2mqtt_cc1352p/4 Button Scene Switch/action` for https://www.zigbee2mqtt.io/integration/home_assistant.html#via-mqtt-device-trigger-recommended

closed 2021-01-26 by Koenkk

james-fry, 2021-01-26:
Thanks for confirming, @Koenkk 
Its weird as I have many other buttons in my network (xiaomi, hue etc) and I dont get the double toggle happening with these...
Time for some more analysis.

james-fry, 2021-01-26:
Solved:
_All_ my button flows in node-red use "Trigger: State" node as the input, and all of them working fine with toggle - one physical click = one toggle action.
I changed to using an "Events: state" node and now it works fine.