Koenkk/zigbee2mqtt #5917: Zigbee2MQTT:error - Failed to set permit join to true

michaljuris, 2021-01-26:
## What happened

Hello, I'm having this strange problem with the new docker installation of z2m:

`Zigbee2MQTT:error 2021-01-23 01:46:49: Failed to set permit join to true`

followed by: https://pastebin.com/L0yLVHve

when I click on 'Permit Join' through the Frontend UI I get `Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'SRSP - ZDO - mgmtPermitJoinReq after 6000ms'` - https://pastebin.com/JiGYuHzH

## Debug info
Zigbee2MQTT version: 1.17.0 (same happens for stable and Dev version [#8c55598])
Adapter hardware: CC2530+CC2591 connected to RPi4 using CP2102 USB to Serial
Adapter firmware version: CC2530_CC2591_DEFAULT_20190608
Docker configuration: https://pastebin.com/PES3T9j3
configuration.yaml: https://pastebin.com/ws0p5upG

Whole log: https://snipboard.io/3Gtpum.jpg 
https://snipboard.io/NYVi8t.jpg
https://snipboard.io/Gl69RK.jpg

Koenkk, 2021-01-26:
This happens when the connection with the CC2530 is lost OR if the firmware got corrupted (try reflashing it)

michaljuris, 2021-01-28:
@Koenkk Thank a lot for the hint. The probable cause was the USB to TTL converter. I've replaced it for a different kind and it just started to work (Both CP2102)

This one _didn't_ work (https://www.aliexpress.com/item/4000375227626.html) I've tried 2 pieces: 
[image]

and this _did_ work great (https://www.aliexpress.com/item/32670594212.html):
[image]

closed 2021-01-28 by michaljuris