Koenkk/zigbee2mqtt #5923: OTA update from URL or local file

SimonThoustrup, 2021-01-26:
## Is your feature request related to a problem? Please describe
Danfoss is publishing OTA files for the Ally thermostat (https://www.zigbee2mqtt.io/devices/014G2461.html) on their website - https://www.danfoss.com/en/products/smart-heating/dhs/smart-heating/danfoss-ally/danfoss-ally-support/#tab-software

## Describe the solution you'd like
To provide the URL for the OTA file or drop the OTA file in a folder in the zigbee2mqtt directory structure and then initiate the OTA upgrade as for other devices where OTA is already supported.

## Additional context
I believe this could be useful for other devices as well ref. https://github.com/Koenkk/zigbee2mqtt/discussions/2921#discussioncomment-312151

closed 2021-03-05 by github-actions[bot]

hellcry37, 2021-04-09:
up for ota update from local folder

ipeacocks, 2021-12-16:
Can't update one TRV. How to force that?