Koenkk/zigbee2mqtt #5927: BlitzWolf BW-SS7 1 gang

william89731, 2021-01-27:
[image]

**debug**

DEBUG: Received Zigbee message from '0x5c0272fffec595d5', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
WARNING: Received message from unsupported device with Zigbee model 'TS0003' and manufacturer name '_TYZB01_aneiicmq'

**database.db**

{"id":31,"type":"Router","ieeeAddr":"0x5c0272fffec595d5","nwkAddr":4432,"manufId":4098,"manufName":"_TYZB01_aneiicmq","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0003","manufacturerName":"_TYZB01_aneiicmq","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0,"onTime":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1611719836244}

pixeldoc2000, 2021-02-04:
Look at: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Try this in devices.js :
```
    {
        fingerprint: [{modelID: 'TS0003', manufacturerName: '_TYZB01_aneiicmq'}],
        model: 'BW-SS7',
        vendor: 'BlitzWolf',
        description: 'Zigbee 3.0 smart light switch module 1 gang',
        extend: preset.switch(),
        exposes: [e.switch().withEndpoint('l1')],
        endpoint: (device) => {
            return {'l1': 1};
        },
        toZigbee: [tz.TYZB01_on_off],
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        },
    },
```

william89731, 2021-02-05:
> Look at: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
>
> Try this in devices.js :
>
> ```
>     {
>         fingerprint: [{modelID: 'TS0003', manufacturerName: [...]

william89731, 2021-02-05:
> Look at: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
>
> Try this in devices.js :
>
> ```
>     {
>         fingerprint: [{modelID: 'TS0003', manufacturerName: [...]

ok perfect now it is perfectly integrated.  thanks!!!

closed 2021-02-05 by william89731

PetersPlace, 2021-02-07:
Is this still needed in HA with Zigbee2mqtt 1.17.1   version ?
My yesterday delivered Blitzwolff are BW-SS7   1 gang   are still listed as  Zemismart ZM-L03E-Z

This device is a little different then the 2 gang device
`debug 2021-02-06 20:59:16: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:16: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:16: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:16: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:17: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:17: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:17: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:17: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:17: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:17: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:17: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:17: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:17: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:17: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:18: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:18: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:18: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:18: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:18: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:18: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:18: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:18: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:18: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:18: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:19: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:19: Skipping message, definition is undefined and still interviewing
debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0003"}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TYZB01_aneiicmq"}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info  2021-02-06 20:59:20: Successfully interviewed '0x5c0272fffec595c2', device has successfully been paired
info  2021-02-06 20:59:20: Device '0x5c0272fffec595c2' is supported, identified as: Zemismart Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)`

william89731, 2021-02-07:
> 
> 
> Is this still needed in HA with Zigbee2mqtt 1.17.1 version ?
> [image]
> 
> ==============
> 
> debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0003"}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TYZB01_aneiicmq"}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:20: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
> debug 2021-02-06 20:59:19: Received Zigbee message from '0x5c0272fffec595c2', type 'readResponse', cluster 'genBasic', data **'{"modelId":"TS0003"}'** from endpoint 1 with groupID 0
> info 2021-02-06 20:59:20: Successfully interviewed '0x5c0272fffec595c2', device has successfully been paired
> info 2021-02-06 20:59:20: Device '0x5c0272fffec595c2' is supported, identified as: Zemismart Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)

to me it works perfectly (zigbee2mqtt 1.17.1, conbee2 zigbee 3.0)

PetersPlace, 2021-02-07:
I added  william89731 commented 2 days ago   to my devices.js and now it works

The 1 already listed is  fingerprint: [{modelID: 'TS0003', manufacturerName: '_TYZB01_digziiav'}],
i made a copy and added  fingerprint: [{modelID: 'TS0003', manufacturerName: '_TYZB01_aneiicmq'}],