Koenkk/zigbee2mqtt #5928: Trust ZYCT-202 Failed to configure

Mr-Groch, 2021-01-27:
## What happened
Each Zigbee2MQTT restart i see in logs:
```
Zigbee2MQTT:error 2021-01-26 17:00:47: Failed to configure 'Pilot do świateł 1', attempt 1 (Error: Bind 0x00158d00038e48fa/1 genOnOff from '0x00124b00215fff6e/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
Zigbee2MQTT:error 2021-01-26 17:01:05: Failed to configure 'Pilot do świateł 1', attempt 2 (Error: Bind 0x00158d00038e48fa/1 genOnOff from '0x00124b00215fff6e/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
Zigbee2MQTT:error 2021-01-26 17:01:20: Failed to configure 'Pilot do świateł 1', attempt 3 (Error: Bind 0x00158d00038e48fa/1 genOnOff from '0x00124b00215fff6e/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
```
The same happens when I click "reconfigure" button from frontend for this device. I'm waking it up before that. Device is working - all button presses are handled by zigbee2mqtt, just no battery level (I'm using this device about 2 months now and still no battery property).

## What did you expect to happen
There should be no error in logs for this device and reconfiguration should success. Also battery level should work

## How to reproduce it (minimal and precise)
1. Force delete ZYCT-202 device from zigbee2mqtt
2. Reset ZYCT-202 device by holding Off button for 10 seconds
3. Enable pairing mode in zigbee2mqtt
4. Pair ZYCT-202 with zigbee2mqtt by holding "two bulbs" button for 5 seconds until leds flashes
5. Observe in zigbee2mqtt logs that configuration of device fails
6. Restart zigbee2mqtt and failed to reconfigure still fails

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20201113

Tyrian81, 2021-01-28:
I have the exact same problem, hope theres a solution
tried resetting device, repairing like 10 x

Koenkk, 2021-01-28:
Battery isn't supported indeed. Can you update the definition in devices.js to below and see if it works after a repair?

```js
{
    zigbeeModel: ['\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
                    '\u0000\u0000\u0000\u0000\u0000', 'ZLL-NonColorController'],
    model: 'ZYCT-202',
    vendor: 'Trust',
    description: 'Remote control',
    fromZigbee: [fz.command_on, fz.command_off_with_effect, fz.legacy.ZYCT202_stop, fz.legacy.ZYCT202_up_down, fz.battery],
    exposes: [e.action(['on', 'off', 'stop', 'brightness_stop', 'brightness_move_up', 'brightness_move_down']), e.battery()],
    toZigbee: [],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
},
```

Mr-Groch, 2021-01-28:
I have some trouble with pairing (but with this device there were always problems during pairing - no all attempts were succesfull). At the end I paired it successfully with no error message, battery is exposed now, but without value for now... I will wait some time if it will appear or not...

Reconfigure seems that do nothing... There is no error but i see only reconfigure message at bridge topic and nothing more...

Tyrian81, 2021-01-29:
I'm running zigbee2mqtt in docker container , where can i find the devices.js only have 3 files in its config folder  ?
configuration.yaml database.db state.json

Mr-Groch, 2021-01-30:
After 2 days still no battery value, and still:

```
Zigbee2MQTT:error 2021-01-28 22:22:26: Failed to configure 'Pilot do świateł 1', attempt 3 (Error: Bind 0x00158d00038e48fa/1 genPowerCfg from '0x00124b00215fff6e/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
```

Tyrian81, 2021-01-30:
> After 2 days still no battery value, and still:
>
> ```
> Zigbee2MQTT:error 2021-01-28 22:22:26: Failed to configure 'Pilot do świateł 1', attempt 3 (Error: Bind 0x00158d00038e48fa/1 genPowerCfg from [...]

Do your buttons work ?¿

Tyrian81, 2021-01-30:
can i packet sniff with zigbee2mqtt ? maybe i can sniff something out ... :)

Mr-Groch, 2021-01-30:
> Do your buttons work ?¿

Yes, buttons are working, action and action_group

Koenkk, 2021-01-30:
@Mr-Groch instead of repairing force a reconfigure via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure . Right before doing wakeup the device so by clicking on it and check the logs if OK.

If that doesn't help share the data/database.db entry of this device.

Mr-Groch, 2021-01-30:
> @Mr-Groch instead of repairing force a reconfigure via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure . Right before doing [...]

First, response was:

```
{
  "data": {
    "id": "Pilot do świateł 1"
  },
  "error": "Failed to configure (Bind 0x00158d00038e48fa/1 genPowerCfg from '0x00124b00215fff6e/1' failed (AREQ - ZDO - bindRsp after 10000ms))",
  "status": "error"
}
```

But next each time I repeat configure request, I've getting instand response:

```
{
  "data": {
    "id": "Pilot do świateł 1"
  },
  "status": "ok"
}
```

Weird thing that this device announce itself always after wakeup:

```
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Pilot do świateł 1","ieee_address":"0x00158d00038e48fa"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Pilot do świateł 1"},"type":"device_announced"}'
```

Now this device in database.db looks like that:

`{"id":18,"type":"EndDevice","ieeeAddr":"0x00158d00038e48fa","nwkAddr":2584,"manufId":4653,"manufName":"Trust International B.V.","powerSource":"Battery","modelId":"ZLL-NonColorController","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2080,"inClusterList":[0,4,3,6,8,4096],"outClusterList":[0,4,3,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"ZLL-NonColorController","manufacturerName":"Trust International B.V.","powerSource":3,"zclVersion":1,"appVersion":3,"stackVersion":1,"hwVersion":1,"dateCode":"20150212","swBuildId":"2000-0003"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":1010,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":1,"hwVersion":1,"dateCode":"20150212","swBuildId":"2000-0003","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1612000432947}
`

Koenkk, 2021-01-30:
The database explains a lot, the device does not support the power cluster (ID: 1 is not in the `inClusterList`) so therefore it cannot provide battery measurements. I removed the `configure()` entirely so issue will be fixed.

closed 2021-01-30 by Koenkk

Tyrian81, 2021-01-30:
so you made a fix in db file , do i need to update zigbee2mqtt ? or edit file devices.js ?

Koenkk, 2021-01-30:
Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Tyrian81, 2021-01-30:
[https://imgur.com/a/HUMjWq1](url)

Have 2 folders with devices.js ? which should be edited ? running in docker

Mr-Groch, 2021-01-30:
> https://imgur.com/a/HUMjWq1
> 
> Have 2 folders with devices.js ? which should be edited ? running in docker

Open bash console in docker container (`docker exec -it container_name bash`) and there find devices.js (`find / -name devices.js`), next edit it with vi (`vi /app/node_modules/zigbee-herdsman-converters/devices.js`). If you don't know vi - i for edit mode, esc to exit edit mode, / to search in file, to exit :q! to save :w )

Tyrian81, 2021-01-30:
just deleting those 6 lines should fix it ?

Tyrian81, 2021-01-30:
removed the six lines and its not erroring in logs any more , but HA doesn't show actions

`Zigbee2MQTT:info  2021-01-30 17:31:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00038e5ff4"},"type":"device_announced"}'
,Zigbee2MQTT:info  2021-01-30 17:30:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4"},"type":"device_announce"}'
,Zigbee2MQTT:info  2021-01-30 17:30:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4"},"type":"device_announce"}'
,Zigbee2MQTT:info  2021-01-30 17:30:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00038e5ff4"},"type":"device_announced"}'
,Zigbee2MQTT:info  2021-01-30 17:30:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00038e5ff4"},"type":"device_announced"}'
,Zigbee2MQTT:info  2021-01-30 17:31:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4"},"type":"device_announce"}'
`

Tyrian81, 2021-01-30:
every button press ?

Tyrian81, 2021-01-30:
`Zigbee2MQTT:info  2021-01-30 18:44:32: Device '0x00158d00038e5ff4' joined
,Zigbee2MQTT:info  2021-01-30 18:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4"},"type":"device_joined"}'
,Zigbee2MQTT:info  2021-01-30 18:44:32: Starting interview of '0x00158d00038e5ff4'
,Zigbee2MQTT:info  2021-01-30 18:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d00038e5ff4"},"type":"pairing"}'
,Zigbee2MQTT:info  2021-01-30 18:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4","status":"started"},"type":"device_interview"}'
,Zigbee2MQTT:info  2021-01-30 18:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d00038e5ff4"},"type":"device_connected"}'
,Zigbee2MQTT:info  2021-01-30 18:44:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4"},"type":"device_announce"}'
,Zigbee2MQTT:info  2021-01-30 18:44:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00038e5ff4"},"type":"device_announced"}'
,Zigbee2MQTT:info  2021-01-30 18:44:37: Successfully interviewed '0x00158d00038e5ff4', device has successfully been paired
,Zigbee2MQTT:info  2021-01-30 18:44:37: Device '0x00158d00038e5ff4' is supported, identified as: Trust Remote control (ZYCT-202)
,Zigbee2MQTT:info  2021-01-30 18:44:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Remote control","exposes":[{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on","off","stop","up-press","down-press"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZYCT-202","vendor":"Trust"},"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4","status":"successful","supported":true},"type":"device_interview"}'
,Zigbee2MQTT:info  2021-01-30 18:44:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Remote control","friendly_name":"0x00158d00038e5ff4","model":"ZYCT-202","supported":true,"vendor":"Trust"},"type":"pairing"}'
,Zigbee2MQTT:info  2021-01-30 18:44:37: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"10":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]},"configured_reportings":[]},"110":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"12":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"242":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"3":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"4":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"47":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"5":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"6":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"8":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"Coordinator","ieee_address":"0x00124b0018e299ec","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"date_code":"20170627","definition":{"description":"Aqara human body movement and illuminance sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates whether the device detected occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Measured illuminance in lux","name":"illuminance_lux","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Measured illuminance in lux","name":"illuminance","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"RTCGQ11LM","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","msOccupancySensing","msIlluminanceMeasurement","ssIasZone","genPowerCfg","genIdentify"],"output":["genBasic","genOta"]},"configured_reportings":[]}},"friendly_name":"0x00158d000394430b","ieee_address":"0x00158d000394430b","interview_completed":true,"interviewing":false,"model_id":"lumi.sensor_motion.aq2","network_address":3087,"power_source":"Battery","software_build_id":"3000-0001","supported":true,"type":"EndDevice"},{"definition":{"description":" [... cut ...] "genIdentify","genOnOff","genLevelCtrl","touchlink"]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":["touchlink"],"output":["touchlink"]},"configured_reportings":[]}},"friendly_name":"0x00158d00038e5ff4","ieee_address":"0x00158d00038e5ff4","interview_completed":true,"interviewing":false,"model_id":"ZLL-NonColorController","network_address":36949,"power_source":"Battery","software_build_id":"2000-0003","supported":true,"type":"EndDevice"}]'
`

Gets added correctly no errors ?

Tyrian81, 2021-01-30:
> Yes, buttons are working, action and action_group

how did you get the buttons to register , because mine  does neatly pair in zigbee2mqtt and gets added in HA , but no buttonpresses, lqi are reported ?

Mr-Groch, 2021-01-31:
> > > > After 2 days still no battery value, and still:
> > > > ```
> > > > Zigbee2MQTT:error 2021-01-28 22:22:26: Failed to configure 'Pilot do świateł 1', attempt 3 (Error: Bind 0x00158d00038e48fa/1 genPowerCfg from '0x00124b00215fff6e/1' failed (AREQ - ZDO - bindRsp after 10000ms)
> > > >     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
> > > >     at listOnTimeout (internal/timers.js:554:17)
> > > >     at processTimers (internal/timers.js:497:7))
> > > > ```
> > > 
> > > 
> > > Do your buttons work ?¿
> > 
> > 
> > Yes, buttons are working, action and action_group
> 
> how did you get the buttons to register , because mine does neatly pair in zigbee2mqtt and gets added in HA , but no buttonpresses, lqi are reported ? i dont have zigbee bulbs so i dont want to pair it directly with the bulbs. just want the button presses to show up in HA zigbee2mqtt

Do you see in state topic `action `and `action_group` after wakeup and press? If yes, restart HA, autodiscovery sometimes is buggy, and entities are added but not visible until restart

Tyrian81, 2021-01-31:
no in the logs of zigbee2mqtt it doesnt register any button presses, it does get interviewed/paired(without errors) and then added in HA with only action and linkquality...?

Mr-Groch, 2021-01-31:
Try to reset remote (press OFF for about 0 seconds), and pair again (press TWO BULBS button for about 5 seconds) - keep close to coordinator during pairing.. I've also had one time such situation - paired but without button presses, but only once... This device have some trouble with pairing...

Tyrian81, 2021-02-01:
Tried everything repairing 10x close distance and 2m but no presses are registered ? did remove the 6 lines which gives the error , but its just not responding to clicks ?
see debug log when pairing , which seems to be ok ...
https://pastebin.ubuntu.com/p/RrffKfVkKH/

Koenkk, 2021-02-01:
@Tyrian81 can you try updating the definition to below and repair it?

```js
{
    zigbeeModel: ['\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
                    '\u0000\u0000\u0000\u0000\u0000', 'ZLL-NonColorController'],
    model: 'ZYCT-202',
    vendor: 'Trust',
    description: 'Remote control',
    fromZigbee: [fz.command_on, fz.command_off_with_effect, fz.legacy.ZYCT202_stop, fz.legacy.ZYCT202_up_down],
    exposes: [e.action(['on', 'off', 'stop', 'brightness_stop', 'brightness_move_up', 'brightness_move_down'])],
    toZigbee: [],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint, logger) => {
        // Device does not support battery: https://github.com/Koenkk/zigbee2mqtt/issues/5928
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

Tyrian81, 2021-02-01:
`
  {                                                                                                  
        zigbeeModel: ['\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
                      '\u0000\u0000\u0000\u0000\u0000', 'ZLL-NonColorController'],                            
        model: 'ZYCT-202',                   
        vendor: 'Trust',                
        description: 'Remote control',  
        fromZigbee: [fz.command_on, fz.command_off_with_effect, fz.legacy.ZYCT202_stop, fz.legacy.ZYCT202_up_down],
        exposes: [e.action(['on', 'off', 'stop', 'up-press', 'down-press'])],
        toZigbee: [],                                                            
        meta: {configureKey: 2},                                            
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);                                   
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);    
            await reporting.onOff(endpoint);
        }, 
` 
looks identical i removed the 6 lines

Tyrian81, 2021-02-01:
```javascript
  {                                                                                                  
        zigbeeModel: ['\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
                      '\u0000\u0000\u0000\u0000\u0000', 'ZLL-NonColorController'],                            
        model: 'ZYCT-202',                   
        vendor: 'Trust',                
        description: 'Remote control',  
        fromZigbee: [fz.command_on, fz.command_off_with_effect, fz.legacy.ZYCT202_stop, fz.legacy.ZYCT202_up_down],
        exposes: [e.action(['on', 'off', 'stop', 'up-press', 'down-press'])],
        toZigbee: [],                                                            
        meta: {configureKey: 2},                                            
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);                                   
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);    
            await reporting.onOff(endpoint);
        }, 
```

Tyrian81, 2021-02-01:
@Koenkk 
ah i see add , brightness
/app/node_modules/zigbee-herdsman-converters /devices.js is the right one ?

Tyrian81, 2021-02-01:
hmm, added the brightness actions and now i get this error again ?

Zigbee2MQTT:error 2021-02-01 21:10:17: Failed to configure '0x00158d00038e5ff4', attempt 1 (Error: Bind 0x00158d00038e5ff4/1 genOnOff from '0x00124b0018e299ec/1' failed (AREQ - ZDO - bindRsp after 10000ms)
,    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
,    at listOnTimeout (internal/timers.js:554:17)
,    at processTimers (internal/timers.js:497:7))
,Zigbee2MQTT:info  2021-02-01 21:10:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00038e5ff4"},"type":"device_announced"}'

Koenkk, 2021-02-02:
@Tyrian81 make sure to wakeup the device and then directly after that configure it via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure

Mr-Groch, 2021-02-02:
@Koenkk But this is the same config as initially - and with this config configure will fail

Koenkk, 2021-02-02:
@Mr-Groch I know, some may require it some don't (it seems?) so if it works I will surround it with a try/catch

Tyrian81, 2021-02-02:
{"id": "0x00158d00038e5ff4"} where do i input this ?

Tyrian81, 2021-02-02:
docker container /bash ?

Mr-Groch, 2021-02-02:
MQTT Explorer or Home Assistant (dev tools, call service mqtt.publish with data):
```
topic: 'zigbee2mqtt/bridge/request/device/configure'
payload: '{"id": "0x00158d00038e5ff4"}'
```

Tyrian81, 2021-02-02:
hmm now it doesnt even get added in HA, configure  gives error ?
Zigbee2MQTT:info  2021-02-02 20:55:23: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00158d00038e5ff4"},"error":"Device '0x00158d00038e5ff4' cannot be configured","status":"error"}'

Tyrian81, 2021-02-02:
@Koenkk @Mr-Groch 

This should be done after pairing , and then right after waking (pressing some buttons)  right ?

Tyrian81, 2021-02-02:
Hmm paired it again with success, got added in HA check, manual configure fails ?
[https://pastebin.ubuntu.com/p/5SnnntCphW/](url)

Tyrian81, 2021-02-02:
It did do some device_announced

Tyrian81, 2021-02-02:
https://imgur.com/a/wDgKryY

Koenkk, 2021-02-02:
What changed between https://github.com/Koenkk/zigbee2mqtt/issues/5928#issuecomment-771126873 and now? In https://github.com/Koenkk/zigbee2mqtt/issues/5928#issuecomment-771126873 you still had the configuring stuff. If you updated to Zigbe2MQTT 1.17.1 make sure to apply the changes as suggested in https://github.com/Koenkk/zigbee2mqtt/issues/5928#issuecomment-771085494. 

Also what coordinator are you using?

Tyrian81, 2021-02-02:
Coordinator;
https://pastebin.ubuntu.com/p/q4J2RkGhhY/

Re-edited devices.js ;
https://imgur.com/a/qmYbfwt

Tyrian81, 2021-02-02:
just repaired it  with error , will try and configure it manually now

Tyrian81, 2021-02-02:
configuring it tries , but alas;
https://pastebin.ubuntu.com/p/r2xYsyRSzT/

Tyrian81, 2021-02-02:
after pressing 2 bulbs it seemed to give an device_announced, after that a manual configure gave status "OK"

https://pastebin.ubuntu.com/p/cHN6g8fnMz/

Koenkk, 2021-02-02:
I see you are running a very old firmware (`20180507`), updating to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default probably fixes it (no changes are required to the code)

Tyrian81, 2021-02-02:
Ah ok, updating firmware via USB possible ? Or would inneed UART/serial device ?

Koenkk, 2021-02-03:
https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html