Koenkk/zigbee2mqtt #5931: Full use of effects - GLEDOPTO (GL-C-007-1ID)

djansen1987, 2021-01-27:
## Is your feature request related to a problem? Please describe
Not able to use the full potential of the [controller](https://www.zigbee2mqtt.io/devices/GL-C-007-1ID.html)

## Describe the solution you'd like
Add the support of the effects in the z2m device

## Describe alternatives you've considered
[https://github.com/Koenkk/zigbee2mqtt/issues/580#issuecomment-449567679](https://github.com/Koenkk/zigbee2mqtt/issues/580#issuecomment-449567679)

`for HA, to have full functionality, try :`
`light:`

`platform: "mqtt"`
`schema: "json"`
`name: "My Office Light RGBW"`
`state_topic: "zigbee2mqtt/your friendly name"`
`availability_topic: "zigbee2mqtt/bridge/state"`
`brightness: true`
`color_temp: true`
`xy: true`
`command_topic: "zigbee2mqtt/your friendly name/set"`
`qos: 1`
`optimistic: false`
`retain: true`
`rgb: true`
`color_temp_min: 0`
`color_temp_max: 100`
`white_value: true`
`effect: true`
`#Effect list 0=Full, 1=White, 2=RGB, 3=Full hard, 4=Disco, 5=Red Flash, 6=Green Flash, 7=Blue Flash, 8=White Flash]`
`effect_list: [0, 1, 2, 3, 4 ,5, 6, 7, 8]`
`transition: true`

## Additional context
`#Effect list 0=Full, 1=White, 2=RGB, 3=Full hard, 4=Disco, 5=Red Flash, 6=Green Flash, 7=Blue Flash, 8=White Flash]`
`effect_list: [0, 1, 2, 3, 4 ,5, 6, 7, 8]`

70MM13, 2021-01-28:
I think these effects are only working when using the (milight?) remote controls and not over zigbee

closed 2021-03-07 by github-actions[bot]

djansen1987, 2022-03-02:
Afraid not. I solved it by creating a script, helpers and an automation 
Be aware when running this mode it has an high impact on your MQTT network as many messages will be generated

Script: (creates random colors, controlling 2 lights in the example)
```yaml
alias: Discolight
sequence:
  - service: light.turn_on
    data:
      rgb_color:
        - '{{ (range(0, 255)|random) }}'
        - '{{ (range(0, 255)|random) }}'
        - '{{ (range(0, 255)|random) }}'
      brightness: '{{ (range(249, 250)|random) }}'
      transition: '{{ states(''input_number.disco_trans'') |float }}'
    target:
      entity_id: light.studio_muur_2
  - service: light.turn_on
    data:
      rgb_color:
        - '{{ (range(0, 255)|random) }}'
        - '{{ (range(0, 255)|random) }}'
        - '{{ (range(0, 255)|random) }}'
      brightness: '{{ (range(249, 250)|random) }}'
      transition: '{{ states(''input_number.disco_trans'') |float }}'
    target:
      entity_id: light.studio_rgbw
mode: single
```

Helpers:
[image]

Automation (on)
```yaml
alias: Studio Disco
description: ''
trigger:
  - platform: state
    entity_id: input_boolean.discoswitch
    to: 'on'
condition: []
action:
  - repeat:
      until:
        - condition: state
          entity_id: input_boolean.discoswitch
          state: 'off'
      sequence:
        - service: script.discolight
          data: {}
        - delay:
            hours: 0
            minutes: 0
            seconds: '{{ states(''input_number.discodelaysec'') | int  }}'
            milliseconds: '{{ states(''input_number.discodelay'') | int  }}'
mode: single
```

Automation (Off, restore original lights):
```yaml
alias: Studio Disco Off
description: ''
trigger:
  - platform: state
    entity_id: input_boolean.discoswitch
    to: 'off'
    for:
      hours: 0
      minutes: 0
      seconds: 5
    from: 'on'
condition: []
action:
  - choose:
      - conditions:
          - condition: device
            type: is_on
            device_id: 3fc5de8058d5cc37d74d07abac24a8c4
            entity_id: light.studio_muur_2
            domain: light
          - condition: device
            type: is_on
            device_id: 40d5ea83137711eb8d54b7cf5125e9ed
            entity_id: light.studio_rgbw
            domain: light
        sequence:
          - service: light.turn_on
            data:
              kelvin: 2300
            target:
              entity_id:
                - light.studio_muur_2
                - light.studio_rgbw
    default: []
mode: single
```

Card:
```yaml
type: entities
entities:
  - entity: input_boolean.discoswitch     # Turn Automation on/off
  - entity: input_number.disco_trans      # Transition time between colors
  - entity: input_number.discodelay       # Delay time in miliseconds (plus seconds) between next color
  - entity: input_number.discodelaysec  # Delay time in second (plus miliseconds) between next color
```

[image]

tenkaibr, 2022-03-03:
Thank you! I will try your solution!