Koenkk/zigbee2mqtt #5932: Livarno Lux E27 bulb RGB color_temp_startup

timoline, 2021-01-27:
I think color_temp_startup is not supported on a Livarno Lux E27 bulb RGB 
TS0505A
https://www.zigbee2mqtt.io/devices/HG06106C.html#lidl-hg06106c
(btw wrong picture ;-))
https://www.lidl-shop.nl/media/cf3443b01701d5eacd0df0ded28bc3d8.jpeg

Debug Publishing 'set' 'color_temp_startup' to 'bedroom/light1'
Debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x588e81fffee2ee07!
Error Publish 'set' 'color_temp_startup' to 'bedroom/light1' failed: 'Error: Write 0x588e81fffee2ee07/1 lightingColorCtrl({"startUpColorTemperature":65535}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug Error: Write 0x588e81fffee2ee07/1 lightingColorCtrl({"startUpColorTemperature":65535}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23) at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:225:26) at Generator.next (<anonymous>) at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)

Koenkk, 2021-01-27:
Disabled color_temp_startup for all Livarno Lux lights.

Regarding the picture please create a pr to update: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/images/devices/HG06106C.jpg

closed 2021-01-27 by Koenkk

timoline, 2021-01-27:
many thx again :-)