Koenkk/zigbee2mqtt #5937: new device: Lidl Home (Silvercrest) Smart Wireless Door Bell

70MM13, 2021-01-27:
So I bought this today and am in the process of making this work... I'm not that experienced so bare with me.
It's a simple button, with tamper status, battery status and battery level

## Information about the device + link
https://www.lidl.nl/nl/p/smart-home/draadloze-deurbel/p44962
https://zigbeealliance.org/zigbee_products/lidl-home-smart-wireless-door-bell/

## data/database.db entry of the device
```
{"id":16,"type":"EndDevice","ieeeAddr":"0x842e14fffef6d66c","nwkAddr":52983,"manufId":4619,"manufName":"_TZ1800_ladpngdx","powerSource":"Battery","modelId":"TS0211","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0211","manufacturerName":"_TZ1800_ladpngdx","powerSource":3,"zclVersion":3,"appVersion":69,"stackVersion":2,"hwVersion":16,"dateCode":"2020.8.4"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff069cbb","zoneState":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff069cbb","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":2,"hwVersion":16,"dateCode":"2020.8.4","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1611768676551}
```

## debug info
### first pairing
(what is this raw "buffer"?)
```
Starting interview of '0x842e14fffef6d66c'
Received Zigbee message from '0x842e14fffef6d66c', type 'raw', cluster 'genBasic', data '{"data":[13,11,18,0,241,200,84,83,48,50,49,49,32,32,32,32,32,32,32,32,32,32,69,2,0,0,0,0,1,2,4,1,255,21,0,4,0,255,255,255,255,255,255,255],"type":"Buffer"}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0211"}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ1800_ladpngdx"}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
Received Zigbee message from 'floodlight_achter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":2,"imageType":0,"manufacturerCode":4687}' from endpoint 11 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"appVersion":69}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID null
Received Zigbee message from 'awareness_terncy', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":142,"imageType":0,"manufacturerCode":4648}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2020.8.4"}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID null
Received Zigbee message from '0x842e14fffef6d66c', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00212effff069cbb","zoneState":1}' from endpoint 1 with groupID null
Successfully interviewed '0x842e14fffef6d66c', device has successfully been paired
Device '0x842e14fffef6d66c' with Zigbee model 'TS0211' and manufacturer name '_TZ1800_ladpngdx' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

### button click and tamper
```
Received Zigbee message from '0x842e14fffef6d66c', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
```
zonestatus =
click: 1; tamper: 4; tamper+click:5

### battery percentage
(200%?)
```
Received Zigbee message from '0x842e14fffef6d66c', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID null
```

## done so far

added to fromZigbee.js:
```
    ias_doorbell: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                action: (zoneStatus & 1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
	},
```

added to devices.js:
```
    {
		fingerprint: [{modelID: 'TS0211', manufacturerName: '_TZ1800_ladpngdx'}],
		model: 'HG06668',
		vendor: 'Lidl',
		description: 'Silvercrest smart door bell',
		fromZigbee: [fz.ias_doorbell, fz.battery],
		toZigbee: [],
		exposes: [e.action(), e.battery_low(), e.tamper(), e.battery()],
		meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        },
	},
```

I based it on similar looking devices. The battery_low status is an assumption also based on what looked pretty common.
Well basically it works. I think you can use this info to add support I guess?

Regards,

Wimpie

timoline, 2021-01-27:
I was also interested in this device, but couldn't find out where the ding-dong should come from? :-)
Is it just an ordinairy onoff button?

70MM13, 2021-01-27:
yup, as far as I can tell, just a single click, no more, no less :-)

Koenkk, 2021-01-27:
Already supported in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-27 by Koenkk

70MM13, 2021-01-27:
ah.. looked in the wrong place to see if it was already supported 
nevermind :-)

timoline, 2021-01-27:
> yup, as far as I can tell, just a single click, no more, no less :-)

ok, thx, that is what I expected, searching for a good gong/dingdong thingy now...

ciB89, 2021-01-28:
Are there any docs for the button yet? I cannot get it to pair although I am running on latest commit.

timoline, 2021-01-28:
> Are there any docs for the button yet? I cannot get it to pair although I am running on latest commit.

https://www.lidl-service.com/cps/rde/xchg/SID-2790666F-2691023C/lsp/hs.xsl/product.html?id=5090969313&title=Draadloze+deurbel&count=1

ciB89, 2021-01-28:
> > Are there any docs for the button yet? I cannot get it to pair although I am running on latest commit.
> 
> https://www.lidl-service.com/cps/rde/xchg/SID-2790666F-2691023C/lsp/hs.xsl/product.html?id=5090969313&title=Draadloze+deurbel&count=1

I was more thinking about z2mqtt documentation. I can put the device in pairing mode but it is not showing anything in the logs.