Koenkk/zigbee2mqtt #5945: LIVARNO LUX battery mood light

BudBundi, 2021-01-27:
At the moment I'm to dumb for the tuya part, also I'm using docker.

## Information about the device + link
https://www.lidl.de/de/livarno-lux-stimmungsleuchte-mit-akku-und-lichtfarbensteuerung-rgb-led-zigbee-smart-home/p365242

The device can run on power adapter and battery, I have no idea if it sends any battery values.

It is show as supported but, it is the wrong device MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"RGB+CCT light controller","friendly_name":"0x842e14fffe32da78","model":"TS0505A","supported":true,"vendor":"TuYa"},"type":"pairing"}'
Also, it throws one error at pairing
Failed to setup reporting for '0x842e14fffe32da78' - Error: Read 0x842e14fffe32da78/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')

## data/database.db entry of the device
{"id":95,"type":"Router","ieeeAddr":"0x842e14fffe32da78","nwkAddr":51013,"manufId":4098,"manufName":"_TZ3000_9cpuaca6","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_9cpuaca6","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca15641","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca15641","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1611776298599}

Koenkk, 2021-01-28:
What is the model of this device? (probably written on the device and probably starts with HG)

BudBundi, 2021-01-28:
It seems to be 14148906L and should be supported by dev branch. https://github.com/Koenkk/zigbee-herdsman-converters/commit/b09704d2df4369038c0d354af30cd2cfd560fb1b

BudBundi, 2021-01-28:
OK after updating the docker container right now, it is recognized correct.
Seems I was too tired yesterday after work.
Thank you for your work and support.

closed 2021-01-28 by BudBundi