Koenkk/zigbee2mqtt #5952: Request for adding support for Adeo and Awox devices

Chartrux, 2021-01-28:
Hi,
I am new to Home Assistant and Zigbee2mqtt, so please excuse me if my request is not in the right place.
I recently bought two light bulbs which are not supported.
I managed to add them manually to my HA with the following configuration, but how could i ask to add them in the next release of zigbee2mqtt ?

## Information about the device + link
The first one is this Adeo color light
https://www.leroymerlin.fr/v3/p/produits/ampoule-connectee-led-standard-e27-intensite-couleur-variables-enki-lexman-e1506793547
```{
        zigbeeModel: ['LXEK-1'],
        model: '9CZA-A806ST-Q1A',
        vendor: 'Adeo',
        description: 'LED E27 color',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
```
The second one is this Awox white bulb
https://www.castorama.fr/ampoule-led-connectee-e27-globe-850lm-63w-blanc-chaud-awox/9002759339517_CAFR.prd
```
    {
        zigbeeModel: ['TLSR82xx'],
        model: 'TLSR82xx',
        vendor: 'AwoX',
        description: 'LED E27 white',
        extend: generic.light_onoff_brightness,
    },
```

## data/database.db entry of the device
I have no idea how i can get this information :(

Koenkk, 2021-01-28:
Added both devices, thanks.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Will also be included in the next z2m release on 1 feb.

closed 2021-01-28 by Koenkk

Chartrux, 2021-01-28:
Wow, that was quick !
Thanks