Koenkk/zigbee2mqtt #5953: Weird issue at brightness 1

gielfeldt, 2021-01-28:
## What happened
When bulb is at brightness 1, the bulb visibly turns off completely, even though the state remains in ON. And in this state, it cannot perform brightness_move. The bulb acts as if it is off, even though the state is ON. The behavior is consist across all my ikea and philips bulbs.

## What did you expect to happen
I would expect the bulb to be on at minimum brightness, when state ON and brightness 1 is reported.

## How to reproduce it (minimal and precise)
1) publish state on
2) publish brightness 1
3) possibly wait for state to settle internally in zigbee2mqtt?
4) publish state off
5) publish state on

other weird state:

1) publish state on
2) publish some brightness
3) publish brightness_move -200 fx
4) wait a little until bulb is at minimum brightness
5) publish brightness_move 0
6) bulb is now visibly on and with state ON and brightness 1
7) publish state off - light turns off
8) publish state on - light does not turn on, but state is ON and brightness is 1

I've tried disable all state cache options in config also, but no change in behavior.
Any ideas anyone? I can't possibly be the only one with this weird issue?

Is minimum brightness supposed to be 2? If so, then I guess it's a problem that brightness_move can go beyond this.

Thanks.

## Debug info
Zigbee2MQTT version: latest-dev
Adapter hardware: CC26X2R1
Adapter firmware version:

Koenkk, 2021-01-28:
Can you provide the debug log of this? For these bulbs `meta: {turnsOffAtBrightness1: true},`  should be set but this is already the case for the Philips ones. https://github.com/Koenkk/zigbee-herdsman-converters/blob/5750e1176c5ee04ae0d4d0ac2716aa2a499cfa1a/devices.js#L2745 . Also provide your data/database.db

gielfeldt, 2021-01-29:
I'll try to provide some logs if I get the time. In the meantime, I've just discovered something.

* It seems that lights having the turnsOffAtBrightness1, automatically sets brightness to 2, if brightness is set to 1
* Even though the light has turnsOffAtBrightness1, you can still provoke the problem and end up at brightness 1, by using brightness_move downwards.
* If you have lights in a group, and those lights have different turnsOffAtBrightness1 settings, it seems to base the group on the first light?

Perhaps the problem that turnsOffAtBrightness1 solves is not sufficient due to brightness_move?

Having a light at brightness "1", state "on" basically makes it unresponsive until you explicitly set a brightness > 1. This is the real problem for me.

I'm now trying to treat these symptoms by a trigger in nodered listening on all lights, and then publish a brightness "2", if state is "on" and brightness is "1". Seems to be working so far, but it sure feels hackish...

jkempson, 2021-01-29:
Having the same issue but thought Home Assistant would be the issue. 

Its my Ikea bulbs showing the behaviours described.

Happy to help with any logs/testing required.

closed 2021-03-09 by github-actions[bot]