Koenkk/zigbee2mqtt #5956: regression for smart9 / S9ZGBRC01

wondyx, 2021-01-28:
Hello,
A problem with remote control Smart9 / S9ZGBRC01 has appeared after zigbee2mqtt version > 1.14.2.
With this version, and previous version, it works fine; but after upgrade events appears few times for any buttons of remote control, except "emergency" button.

before upgrade, an action on button does generate only one event (witch is good)
Zigbee2MQTT:info  2021-01-28 11:49:08: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"disarm","linkquality":18}' 

After upgrade, one action creates many events
You can see logs below:

Zigbee2MQTT:debug 2021-01-28 11:27:11: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-28 11:27:11: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"disarm","action_code":"","action_zone":255,"linkquality":68}'
Zigbee2MQTT:debug 2021-01-28 11:27:15: Device '0xec1bbdfffe0f9df9' announced itself
Zigbee2MQTT:info  2021-01-28 11:27:15: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe0f9df9","ieee_address":"0xec1bbdfffe0f9df9"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-01-28 11:27:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffe0f9df9"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-01-28 11:27:15: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-28 11:27:15: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"disarm","action_code":"","action_zone":255,"linkquality":73}'
Zigbee2MQTT:debug 2021-01-28 11:27:18: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-28 11:27:18: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"disarm","action_code":"","action_zone":255,"linkquality":78}'
Zigbee2MQTT:debug 2021-01-28 11:27:20: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-28 11:27:20: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":78}'
Zigbee2MQTT:debug 2021-01-28 11:27:21: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-28 11:27:21: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"disarm","action_code":"","action_zone":255,"linkquality":78}'
Zigbee2MQTT:debug 2021-01-28 11:27:35: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":70,"imageType":18567,"manufacturerCode":4619}' from endpoint 1 with groupID 0

I've tried ZHA1.2 (cc2531)  and ZHA3 (cc2538+2592), same problem appears.

Thanks for your help.
regards,

wondyx, 2021-01-29:
Hello,
Some precisions: my coordinator run on raspberry pi witch docker.
Same problem appears on docker running on intel cpu.

I've made some tries.

As problem appeared with zigbee2mqtt-1.14.3, I 've tried to build and run version this version with zigbee-herdsman-0.12.108.tgz (version used in zigbee2mqtt-1.14.2)  instead of zigbee-herdsman-0.12.116.tgz (version used in zigbee2mqtt-1.14.3).
The problem did not not appear anymore.
So it is introduced with with new version of zigbee-herdsman.

Could anyone help please ?

thanks

Koenkk, 2021-01-29:
Can you provide the herdsman debug log of a single press?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

wondyx, 2021-01-29:
Hello,
I can be a little more precise.
Problem appears with version zigbee-herdsman-0.12.112.tgz.
Replacing the following files from zigbee-herdsman-0.12.111,"cluster.d.ts.map"  "cluster.js"  "cluster.js.map", makes problem disappear with version zigbee2mqtt-1.14.3.

Logs for one button with version 1.17.0 below, thanks.

 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,1,5,78,20,1,1,0,99,0,211,235,60,0,0,6,1,13,0,1,0,255,78,20,29,85] +15s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,1,5,78,20,1,1,0,99,0,211,235,60,0,0,6,1,13,0,1,0,255,78,20,29,85] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,1,5,78,20,1,1,0,99,0,211,235,60,0,0,6,1,13,0,1,0,255,78,20,29] - 85 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1281,"srcaddr":5198,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":99,"securityuse":0,"timestamp":3992531,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[1,13,0,1,0,255]}} +15s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":13,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"armmode":1,"code":"","zoneid":255}},"address":5198,"endpoint":1,"linkquality":99,"groupID":0}' +15s
Zigbee2MQTT:debug 2021-01-29 23:13:05: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
Zigbee2MQTT:info  2021-01-29 23:13:05: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":99}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,1,5,78,20,1,1,0,141,0,235,16,61,0,0,6,1,14,0,1,0,255,78,20,29,122] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,1,5,78,20,1,1,0,141,0,235,16,61,0,0,6,1,14,0,1,0,255,78,20,29,122] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,1,5,78,20,1,1,0,141,0,235,16,61,0,0,6,1,14,0,1,0,255,78,20,29] - 122 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1281,"srcaddr":5198,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":141,"securityuse":0,"timestamp":4002027,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[1,14,0,1,0,255]}} +3s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":14,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"armmode":1,"code":"","zoneid":255}},"address":5198,"endpoint":1,"linkquality":141,"groupID":0}' +3s
Zigbee2MQTT:debug 2021-01-29 23:13:08: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  2021-01-29 23:13:08: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":141}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,1,5,78,20,1,1,0,141,0,241,54,61,0,0,6,1,15,0,1,0,255,78,20,29,71] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,1,5,78,20,1,1,0,141,0,241,54,61,0,0,6,1,15,0,1,0,255,78,20,29,71] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,1,5,78,20,1,1,0,141,0,241,54,61,0,0,6,1,15,0,1,0,255,78,20,29] - 71 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1281,"srcaddr":5198,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":141,"securityuse":0,"timestamp":4011761,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[1,15,0,1,0,255]}} +3s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":15,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"armmode":1,"code":"","zoneid":255}},"address":5198,"endpoint":1,"linkquality":141,"groupID":0}' +3s
Zigbee2MQTT:debug 2021-01-29 23:13:11: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
Zigbee2MQTT:info  2021-01-29 23:13:11: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":141}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,78,20,78,20,249,157,15,254,255,189,27,236,128,41] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,78,20,78,20,249,157,15,254,255,189,27,236,128,41] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [78,20,78,20,249,157,15,254,255,189,27,236,128] - 41 +0ms
[... cut ...] "groupID":0}' +1s
Zigbee2MQTT:debug 2021-01-29 23:13:21: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
Zigbee2MQTT:info  2021-01-29 23:13:21: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":141}'

hope this helps

Koenkk, 2021-01-30:
In the logs you see the remote send command multiple times:  `Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm'`.

I don't understand how changes in cluster.js could relate to this. Reverting `cluster.d.ts.map` `cluster.js.map` shouldn't make a difference here. So we should focus on `cluster.js`, can you apply changes 1 by 1 and see what causes it?

wondyx, 2021-01-30:
Hello,
you're right: restoring only cluster.js from zigbee-herdsman-0.12.111 fix version zigbee2mqtt-1.4.3.
I had copied all changes because I had no idea of protocol.  It is just empirical method.

Single press button logs in this version, gives a single mqtt event :

 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,1,5,78,20,1,1,0,44,0,134,101,29,0,0,6,1,72,0,1,0,255,78,20,29,165] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,1,5,78,20,1,1,0,44,0,134,101,29,0,0,6,1,72,0,1,0,255,78,20,29,165] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,1,5,78,20,1,1,0,44,0,134,101,29,0,0,6,1,72,0,1,0,255,78,20,29] - 165 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1281,"srcaddr":5198,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":44,"securityuse":0,"timestamp":1926534,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[1,72,0,1,0,255]}} +7s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":72,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"armmode":1,"code":"","zoneid":255}},"address":5198,"endpoint":1,"linkquality":44,"groupID":0}' +7s
Zigbee2MQTT:debug 2021-01-30 16:48:48: Received Zigbee message from '0xec1bbdfffe0f9df9', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-30 16:48:48: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":44,"action":"arm_day_zones"}'
  zigbee-herdsman:controller:endpoint DefaultResponse 0xec1bbdfffe0f9df9/1 1281(0, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":5198,"destendpoint":1,"srcendpoint":1,"clusterid":1281,"transid":2,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,72,11,0,0]}} +7s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,78,20,1,1,1,5,2,0,30,5,24,72,11,0,0,54] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms

If I can help,
thanks

wondyx, 2021-01-30:
No much differences, only one:

ssIasAce: {
        ID: 1281,
        attributes: {},
        commands: {
            arm: {
                ID: 0,
         ====>       response: 0,
                parameters: [
                    { name: 'armmode', type: dataType_1.default.uint8 },
                    { name: 'code', type: dataType_1.default.charStr },
                    { name: 'zoneid', type: dataType_1.default.uint8 },
                ],

removing "response: 0," from version zigbee2mqtt-1.17.0 (zigbee-herdsman": "^0.13.41") makes problem disappear.

Is this a good idea ?
thx

Koenkk, 2021-01-31:
Can you edit `controller/model/device.js` and remove in `onZclData()`.

```js
// Send a default response if necessary.
const isDefaultResponse = frame.isGlobal() && frame.getCommand().name === 'defaultRsp';
const commandHasResponse = frame.getCommand().hasOwnProperty('response');
if (!frame.Header.frameControl.disableDefaultResponse && !isDefaultResponse && !commandHasResponse) {
    try {
        yield endpoint.defaultResponse(frame.getCommand().ID, 0, frame.Cluster.ID, frame.Header.transactionSequenceNumber);
    }
    catch (error) {
        debug.error(`Default response to ${this.ieeeAddr} failed`);
    }
}
```

Is the problem fixed then? (even with `response: 0` in cluster.js)

wondyx, 2021-01-31:
Hello, 
thanks for your answer.

I replaced file device.js with lines removed:

diff device.js device.js.o 
166a167,177
> // Send a default response if necessary.
>             const isDefaultResponse = frame.isGlobal() && frame.getCommand().name === 'defaultRsp';
>             const commandHasResponse = [...]

/app # cp data/device.js ./node_modules/zigbee-herdsman/dist/controller/model/device.js
/app # cp data/device.js ./node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/dist/controller/model/device.js

But It does not correct the problem, as you can see in the logs below.

Zigbee2MQTT:info  2021-01-31 21:15:22: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":47}'
Zigbee2MQTT:info  2021-01-31 21:15:25: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":39}'
Zigbee2MQTT:info  2021-01-31 21:15:28: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":39}'
Zigbee2MQTT:info  2021-01-31 21:15:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe0f9df9","ieee_address":"0xec1bbdfffe0f9df9"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-01-31 21:15:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffe0f9df9"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-01-31 21:15:32: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":10}'
Zigbee2MQTT:info  2021-01-31 21:15:35: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":13}'
Zigbee2MQTT:info  2021-01-31 21:15:37: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":7}'
Zigbee2MQTT:info  2021-01-31 21:15:38: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"battery":0,"linkquality":13}'

regards,

Koenkk, 2021-02-01:
Does changing it to below fix the problem?

```js
// Send a default response if necessary.
const isDefaultResponse = frame.isGlobal() && frame.getCommand().name === 'defaultRsp';
const commandHasResponse = frame.getCommand().hasOwnProperty('response');
if (!frame.Header.frameControl.disableDefaultResponse && !isDefaultResponse) {
    try {
        yield endpoint.defaultResponse(frame.getCommand().ID, 0, frame.Cluster.ID, frame.Header.transactionSequenceNumber);
    }
    catch (error) {
        debug.error(`Default response to ${this.ieeeAddr} failed`);
    }
}
```

wondyx, 2021-02-02:
Hello,
correction seems to fix problem.
Thank you very much.

Koenkk, 2021-02-02:
@wondyx can you discard any made changes and apply the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2193/files , does this also fix it?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

wondyx, 2021-02-02:
Yes it does the job too.
Startup messages from remote control are escaping the mechanism (for example when battery in inserted).
Can you see repetition below in logs.
But for my case this is not a problem.

Thx
Zigbee2MQTT:info  2021-02-02 22:24:06: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":94}'
Zigbee2MQTT:info  2021-02-02 22:24:06: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":99}'
Zigbee2MQTT:info  2021-02-02 22:24:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe0f9df9","ieee_address":"0xec1bbdfffe0f9df9"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-02-02 22:24:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffe0f9df9"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-02-02 22:24:13: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":70}'
Zigbee2MQTT:info  2021-02-02 22:24:16: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":76}'
Zigbee2MQTT:info  2021-02-02 22:24:16: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":70}'
Zigbee2MQTT:info  2021-02-02 22:24:19: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":73}'
Zigbee2MQTT:info  2021-02-02 22:24:26: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe0f9df9', payload '{"battery":0,"linkquality":76}'

closed 2021-02-02 by Koenkk