Koenkk/zigbee2mqtt #5960: Devolco / Frient Energy Meter

willew, 2021-01-28:
## Information about the device + link
Frient / Devolco Energy Meter  https://frient.com/products/electricity-meter-interface/

## data/database.db entry of the device
`{"id":31,"type":"EndDevice","ieeeAddr":"0x0015bc001b021afe","nwkAddr":30039,"manufId":0,"manufName":"Develco","powerSource":"Battery","modelId":"ZHEMI101","epList":[1,2],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[2,3,5],"outClusterList":[1],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":7,"inClusterList":[0,3,1794],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"modelId":"ZHEMI101","manufacturerName":"Develco","powerSource":3,"zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1611864679486}
`

With this device I would like to follow electricity consumption in real time. I added the following into device.js:
`	{
		zigbeeModel: ['lumi.sensor_power'], 
		model: 'ZHEMI101', 
		vendor: 'Develco',
		description: 'Develco energy meter ', 
		supports: 'energy', 
		fromZigbee: [], 
		toZigbee: [], 
		exposes: [e.energy()], 
	},
`

(Did not know what to put into zigbeeModel, just put something)
After this I can see the vendor and model in the db (and zbmqttassistant). But I cannot see any messages in the log (debug enabled) from the device (except device announced). I tried to cause the device to send something by using a blinking led.

Ideas how to proceed?

willew, 2021-01-28:
Found [https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-power-meter.src/zigbee-power-meter.groovy ](this). I'll try to understand what to do next.

willew, 2021-01-31:
I installed a sniffer and while the device is attached to the energy meter, I collected traffic. The only messages I can see are packages sent to the nearest router, must be some kind of ping/pong service
`IEEE 802.15.4 Command, Dst: 0xa765, Src: 0x7557
    Frame Control Field: 0x8863, Frame Type: Command, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 136
    Destination PAN: 0x0451
    Destination: 0xa765
    Source: 0x7557
    Command Identifier: Data Request (0x04)
    TI CC24xx-format metadata: FCS OK
`
The device is `0x7557` and the router (LIDL switch) is ´0xa765´.

Any ideas?

Koenkk, 2021-01-31:
Try with the following:

```js
{
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.on_off, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.activePower(endpoint);
        await reporting.powerFactor(endpoint);
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()],
},
```

willew, 2021-02-01:
After the above change I can now see the sensors in HA, but they do not have values. 

I allowed joining and reset the device. In Z2M log file the following is seen
`	Line 32: info  2021-02-01 17:26:57: sensor_energy (0x0015bc001b021afe): ZHEMI101 - Develco Energy meter (EndDevice)	Line 32: info  2021-02-01 17:26:57: sensor_energy (0x0015bc001b021afe): ZHEMI101 - Develco Energy meter (EndDevice)
	Line 34: info  2021-02-01 17:26:57: Configuring 'sensor_energy'
	Line 61: debug 2021-02-01 17:26:58: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"07cdc9d","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":20,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":1105,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x000d6ffffe2fc406":{"friendly_name":"repeater_livingroom"},"0x00158d0001922b02":{"debounce":0.5,"friendly_name":"thermo_bedroom"},"0x00158d0001922ea8":{"friendly_name":"thermo_hanna"},"0x00158d00019cd372":{"friendly_name":"button_backyard"},"0x00158d0001f389e1":{"friendly_name":"door_bedroom"},"0x00158d0001f38d6a":{"friendly_name":"door_backyard"},"0x00158d00020436b5":{"friendly_name":"door_scullery"},"0x00158d00020438cc":{"friendly_name":"door_frontyard"},"0x00158d000205a624":{"friendly_name":"button_christmas1"},"0x00158d0002118721":{"friendly_name":"button_kitchen"},"0x00158d000255c8a7":{"friendly_name":"button_front_door"},"0x00158d000255c8fa":{"friendly_name":"button_backlight"},"0x00158d000275219b":{"friendly_name":"motion_backyard"},"0x00158d0002889184":{"friendly_name":"smoke_livingroom"},"0x00158d00032f7be2":{"debounce":0.5,"friendly_name":"thermo_kitchen"},"0x00158d00032f7ea7":{"debounce":0.5,"friendly_name":"thermo_living_room"},"0x00158d000544ec5d":{"friendly_name":"sensor_hanna"},"0x00158d00054b077e":{"friendly_name":"sensor_guest_room"},"0x0015bc001b021af ...
	Line 64: info  2021-02-01 17:26:58: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"07cdc9d","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":20,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":1105,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x000d6ffffe2fc406":{"friendly_name":"repeater_livingroom"},"0x00158d0001922b02":{"debounce":0.5,"friendly_name":"thermo_bedroom"},"0x00158d0001922ea8":{"friendly_name":"thermo_hanna"},"0x00158d00019cd372":{"friendly_name":"button_backyard"},"0x00158d0001f389e1":{"friendly_name":"door_bedroom"},"0x00158d0001f38d6a":{"friendly_name":"door_backyard"},"0x00158d00020436b5":{"friendly_name":"door_scullery"},"0x00158d00020438cc":{"friendly_name":"door_frontyard"},"0x00158d000205a624":{"friendly_name":"button_christmas1"},"0x00158d0002118721":{"friendly_name":"button_kitchen"},"0x00158d000255c8a7":{"friendly_name":"button_front_door"},"0x00158d000255c8fa":{"friendly_name":"button_backlight"},"0x00158d000275219b":{"friendly_name":"motion_backyard"},"0x00158d0002889184":{"friendly_name":"smoke_livingroom"},"0x00158d00032f7be2":{"debounce":0.5,"friendly_name":"thermo_kitchen"},"0x00158d00032f7ea7":{"debounce":0.5,"friendly_name":"thermo_living_room"},"0x00158d000544ec5d":{"friendly_name":"sensor_hanna"},"0x00158d00054b077e":{"friendly_name":"sensor_guest_room"},"0x0015bc001b021afe":{"f ...
	Line 167: info  2021-02-01 17:26:58: MQTT publish: topic 'homeassistant/switch/0x0015bc001b021afe/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/sensor_energy/set","device":{"identifiers":["zigbee2mqtt_0x0015bc001b021afe"],"manufacturer":"Develco","model":"Energy meter (ZHEMI101)","name":"sensor_energy","sw_version":"Zigbee2MQTT 1.17.0"},"json_attributes_topic":"zigbee2mqtt/sensor_energy","name":"sensor_energy","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/sensor_energy","unique_id":"0x0015bc001b021afe_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
	Line 168: info  2021-02-01 17:26:58: MQTT publish: topic 'homeassistant/sensor/0x0015bc001b021afe/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc001b021afe"],"manufacturer":"Develco","model":"Energy meter (ZHEMI [... cut ...]
754	212.795628	0x0000	0x7557	ZigBee ZDP	67	Match Descriptor Response, Nwk Addr: 0x0000, Status: Success
778	214.261402	0x0000	0x7557	ZigBee ZDP	61	Match Descriptor Response, Nwk Addr: 0x0000, Status: Success
780	214.277945	0x7557	0x0000	ZigBee	56	APS: Ack, Dst Endpt: 0, Src Endpt: 0
782	214.288499	0x7557	0x0000	ZigBee	56	APS: Ack, Dst Endpt: 0, Src Endpt: 0
794	216.161211	0x0000	0x7557	ZigBee HA	67	ZCL: Configure Reporting, Seq: 8
796	216.170671	0x0000	0x7557	ZigBee HA	67	ZCL: Configure Reporting, Seq: 8
810	223.390471	0x0000	0x7557	ZigBee HA	67	ZCL: Configure Reporting, Seq: 8

`

Koenkk, 2021-02-01:
```
error 2021-02-01 17:27:28: Failed to configure 'sensor_energy', attempt 1 (Error: ConfigureReporting 0x0015bc001b021afe/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 30039 - 1 - 3 - 6 - 7 after 10000ms)
```

indicates that `await reporting.onOff(endpoint);` failed.

Can you try with:

```js
{
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.on_off, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        // await reporting.onOff(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.activePower(endpoint);
        await reporting.powerFactor(endpoint);
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()],
},
```

willew, 2021-02-02:
Thanks @Koenkk  for your support, unfortunately it then failed with `rmsVoltage` and after this with `rmsCurrent`. Now the code is
`// Frient
	{
		zigbeeModel: ['ZHEMI101'],
		model: 'ZHEMI101',
		vendor: 'Develco',
		description: 'Energy meter',
		fromZigbee: [fz.on_off, fz.electrical_measurement],
		toZigbee: [tz.on_off],
		meta: {configureKey: 1},
		configure: async (device, coordinatorEndpoint, logger) => {
			const endpoint = device.getEndpoint(1);
			await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
			// await reporting.onOff(endpoint);
			// await reporting.rmsVoltage(endpoint);
			// await reporting.rmsCurrent(endpoint);
			await reporting.activePower(endpoint);
			await reporting.powerFactor(endpoint);
		},
		exposes: [e.switch(), e.power(), e.current(), e.voltage()],
	},`

And you can guess it, also this failed with `activePower` timeout. I checked from the sniffer that the device is alive:
`ZigBee Network Layer Data, Dst: 0x7557, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Configure Reporting, Seq: 4
    Frame Control Field: Profile-wide (0x10)
    Sequence Number: 4
    Command: Configure Reporting (0x06)
    Reporting Configuration Record, Int16: 1
        Direction: Reported (0x00)
        Attribute: Active Power (0x050b)
        Data Type: 16-Bit Signed Integer (0x29)
        Int16: 1
    Minimum Interval: 5
    Maximum Interval: 3600
`
After this I can see the following sent by this device
`ZigBee Device Profile, Device Announcement, Nwk Addr: 0x7557, Ext Addr: Develco_00:1b:02:1a:fe
    Sequence Number: 57
    Nwk Addr of Interest: 0x7557
    Extended Address: Develco_00:1b:02:1a:fe (00:15:bc:00:1b:02:1a:fe)
    Capability Information: 0x00
`

willew, 2021-02-02:
Does [this](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-power-meter.src/zigbee-power-meter.groovy) help? It seems to be related to the same device and if understood, it recognizes two values the device reports:

```
   if (it.clusterInt == 0x0B04 && it.attrInt == 0x050b) {
                        log.debug "meter"
                        map.name = "power"
                        map.value = zigbee.convertHexToInt(it.value)/activePowerDivisor
                        map.unit = "W"
                    }
                    if (it.clusterInt == 0x0702 && it.attrInt == 0x0000) {
                        log.debug "energy"
                        map.name = "energy"
                        map.value = zigbee.convertHexToInt(it.value)/(energyDivisor * 1000)
                        map.unit = "kWh"
                    }
```

Koenkk, 2021-02-02:
Can you try with:

```js
{
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
    },
    exposes: [e.switch(), e.power()],
},
```

willew, 2021-02-03:
```
info  2021-02-03 02:17:22: Configuring 'sensor_energy'
...
debug 2021-02-03 02:17:26: Received Zigbee message from 'sensor_energy', type 'readResponse', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 2 with groupID 0
info  2021-02-03 02:17:26: Successfully configured 'sensor_energy'
debug 2021-02-03 02:17:26: Received Zigbee message from 'sensor_energy', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,152487],"currentTier1SummDelivered":[65535,4294967295],"currentTier2SummDelivered":[65535,4294967295],"instantaneousDemand":0,"status":0}' from endpoint 2 with groupID 0
info  2021-02-03 02:17:26: MQTT publish: topic 'zigbee2mqtt/sensor_energy', payload '{"energy":152.49,"linkquality":72,"power":0}'
```

So it works. Nice!

I changed `devices.js` as follows
```
		exposes: [e.switch(), e.power(), e.energy()],
```
and after this I can see also energy as kWh and some value collected while it was installed. Will install this back to the meter tomorrow. Thanks!

But should this be also exposed as a switch (it is not)? If I try to turn it on, I will see the following:
```
debug 2021-02-03 02:33:27: Publishing 'set' 'state' to 'sensor_energy'
error 2021-02-03 02:33:50: Publish 'set' 'state' to 'sensor_energy' failed: 'Error: Command 0x0015bc001b021afe/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 30039 - 1 - 3 - 6 - 11 after 10000ms)'
debug 2021-02-03 02:33:50: Error: Command 0x0015bc001b021afe/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 30039 - 1 - 3 - 6 - 11 after 10000ms)
```
Should I remove `e.switch()` from the list of exposed entries?

Koenkk, 2021-02-03:
Ah switch should indeed be removed, it can only meter and not actually switch.

Added this device now. Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Assuming this can be closed.

closed 2021-02-03 by Koenkk

blakadder, 2021-05-19:
@willew what mode of energy monitoring are you using? what kind of probe is attached?

alteoxCOM, 2021-08-20:
Hi, is there any possibility to configure the pulse rate?
Default is set to 1000 but I need to set it to 400.

willew, 2021-09-25:
@blakadder Sorry, I had not noticed you question until I had to look at this issue. I have [this](https://images.caruna.fi/styles/full/s3/sahkomittarin_lukeminen_1.jpg?itok=A9bIArcf) meter. The led blinks every Wh.

willew, 2021-09-25:
Home Assistant stopped showing Energy values (Unknown). In Home Assistant I can see the following MQTT message in MQTT info for the device:
```
Payload: 
energy: 8946.49
linkquality: 54
power: 7365
```
But the value for the energy entity is unknown, power is seen correctly. 
```
sensor_energy_energy: Unknown
sensor_energy_power: 7,196 W
```

I tried:
- updated latest zigbee2mqtt and restarted ha
- removed device and started ha

No help.

willew, 2021-10-18:
2021.10.6

> Home Assistant stopped showing Energy values (Unknown). In Home Assistant I can see the following MQTT message in MQTT info for the device:

After I installed the latest HA (2021.10.6), the problem disappared.