Koenkk/zigbee2mqtt #5961: Can not set the TRV in heating mode

piars777, 2021-01-28:
Hi everyone,

## What happened
I have different models of trv but there is one i can not put it in heating mode. I can see the current temperature but when i try to set the system mode to heat i have an error

## What did you expect to happen
I'd like to set the system mode to heat

## How to reproduce it (minimal and precise)
from home assistant i tap on the heat button.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2531
Adapter firmware version:
https://pastebin.com/embed_iframe/rB6Qg2tp

piars777, 2021-01-28:
I found something maybe that's why i can set to heat mode.
In the log, we can see the manufacturer is saswell but my trv manufacturer is siterwell and the model is GS361A-HO4

piars777, 2021-01-29:
I modified the file toZigee.js to add some logs before line 3038.
I log the device model: 
`{
  "zigbeeModel": [
    "ivfvd7h",
    "eaxp72v\u0000",
    "kfvq6avy\u0000"
  ],
  "fingerprint": [
    {
      "modelID": "TS0601",
      "manufacturerName": "_TZE200_zivfvd7h"
    },
    {
      "modelId": "TS0601",
      "manufacturerName": "_TZE200_kfvq6avy"
    }
  ],
  "model": "GS361A-H04",
  "vendor": "Siterwell",
  "description": "Radiator valve with thermostat",
  "fromZigbee": [
    {
      "cluster": "manuSpecificTuya",
      "type": [
        "commandGetData",
        "commandSetDataResponse"
      ]
    },
    {
      "cluster": "genBasic",
      "type": [
        "attributeReport",
        "readResponse"
      ]
    }
  ],
  "meta": {
    "tuyaThermostatSystemMode": {
      "0": "off",
      "1": "auto",
      "2": "manual",
      "3": "comfort",
      "4": "eco",
      "5": "boost",
      "6": "complex"
    },
    "tuyaThermostatPreset": {
      "0": "away",
      "1": "schedule",
      "2": "manual",
      "3": "comfort",
      "4": "eco",
      "5": "boost",
      "6": "complex"
    }
  },
  "toZigbee": [
    {
      "key": [
        "child_lock"
      ]
    },
    {
      "key": [
        "window_detection"
      ]
    },
    {
      "key": [
        "valve_detection"
      ]
    },
    {
      "key": [
        "current_heating_setpoint"
      ]
    },
    {
      "key": [
        "system_mode"
      ]
    },
    {
      "key": [
        "auto_lock"
      ]
    },
    {
      "key": [
        "local_temperature_calibration"
      ]
    },
    {
      "key": [
        "min_temperature"
      ]
    },
    {
      "key": [
        "max_temperature"
      ]
    },
    {
      "key": [
        "boost_time"
      ]
    },
    {
      "key": [
        "comfort_temperature"
      ]
    },
    {
      "key": [
        "eco_temperature"
      ]
    },
    {
      "key": [
        "force"
      ]
    },
    {
      "key": [
        "preset"
      ]
    },
    {
      "key": [
        "scene_store"
      ]
    },
    {
      "key": [
        "scene_recall"
      ]
    },
    {
      "key": [
        "scene_add"
      ]
    },
    {
      "key": [
        "scene_remove"
      ]
    }
  ],
  "whiteLabel": [
    {
      "vendor": "Essentials",
      "description": "Smart home heizkörperthermostat premium",
      "model": "120112"
    },
    {
      "vendor": "TuYa",
      "description": "Głowica termostatyczna",
      "model": "GTZ02"
    },
    {
      "vendor": "Revolt",
      "description": "Thermostatic Radiator Valve Controller",
      "model": "NX-4911"
    }
  ],
  "exposes": [
    {
      "type": "lock",
      "features": [
        {
          "type": "binary",
          "name": "state",
          "property": "child_lock",
          "access": 7,
          "value_on": "LOCK",
          "value_off": "UNLOCK",
          "description": "Enables/disables physical input on the device"
        }
      ]
    },
    {
      "type": "switch",
      "features": [
        {
          "type": "binary",
          "name": "state",
          "property": "window_detection",
          "access": 7,
          "value_on": "ON",
          "value_off": "OFF",
          "value_toggle": "TOGGLE"
        }
      ]
    },
    {
      "type": "numeric",
      "name": "battery",
      "property": "battery",
      "access": 1,
      "unit": "%",
      "description": "Remaining battery in %",
      "value_min": 0,
      "value_max": 100
    },
    {
      "type": "switch",
      "features": [
        {
          "type": "binary",
          "name": "state",
          "property": "valve_detection",
          "access": 7,
          "value_on": "ON",
          "value_off": "OFF",
          "value_toggle": "TOGGLE"
        }
      ]
    },
    {
      "type": "numeric",
      "name": "position",
      "property": "position",
      "access": 1,
      "unit": "%",
      "description": "Position"
    },
    {
      "type": "climate",
      "features": [
        {
          "type": "numeric",
          "name": "current_heating_setpoint",
          "property": "current_heating_setpoint",
          "access": 7,
          "value_min": 5,
          "value_max": 30,
          "value_step": 0.5,
          "unit": "°C",
          "description": "Temperature setpoint"
        },
        {
          "type": "numeric",
          "name": "local_temperature",
          "property": "local_temperature",
          "access": 5,
          "unit": "°C",
          "description": "Current temperature measured on the device"
        },
        {
          "type": "enum",
          "name": "system_mode",
          "property": "system_mode",
          "access": 7,
          "values": [
            "off",
            "auto",
            "heat"
          ],
          "description": "Mode of this device"
        },
        {
          "type": "enum",
          "name": "running_state",
          "property": "running_state",
          "access": 5,
          "values": [
            "idle",
            "heat"
          ],
          "description": "The current running state"
        }
      ]
    },
    {
      "type": "numeric",
      "name": "linkquality",
      "property": "linkquality",
      "access": 1,
      "unit": "lqi",
      "description": "Link quality (signal strength)",
      "value_min": 0,
      "value_max": 255
    }
  ]
}
` 

As you can in the `tuyaThermostatSystemMode` there is no value `heat` but the climate exposes `off` `auto` `heat` as value. Home assistant send to zigbee `heat` when i try to set the system_mode and this value is not recognized as a define value.

Koenkk, 2021-01-29:
Looks same as https://github.com/Koenkk/zigbee2mqtt/issues/5968

closed 2021-01-29 by Koenkk

MattWestb, 2021-01-29:
The saswell and siterwell is very different devices and the commands is not the same.
Z2M must load the right config or its not working with the mode mapping to HA.

piars777, 2021-01-29:
> Looks same as #5968

I modified the devices.js and replaced `meta: {tuyaThermostatSystemMode: tuya.thermostatSystemModes, tuyaThermostatPreset: tuya.thermostatPresets}` with `meta: {tuyaThermostatSystemMode: tuya.thermostatSystemModes3, tuyaThermostatPreset: tuya.thermostatPresets}` for Siterwell and i don't have anymore the problem but still remain an other  mapping problem. i ll provide more detail later