Koenkk/zigbee2mqtt #5962: TuYa RGB controller model TS0503B

ASSever, 2021-01-28:
Good afternoon!
I apologize for my poor English.

I recently received this controller.
On the same day, I found out that this particular model is not supported.

I don't know much about this topic.
But I was able to fit in node_modules/zigbee-herdsman-converters/devices.js
this piece of code:
{
    fingerprint: [{modelID: 'TS0503B', manufacturerName: '_TZ3000_i8l0nqdu'}],
    model: 'TS0503B',
    vendor: 'TuYa',
    description: 'Zigbee smart mini led strip controller 5V/12V/24V RGB',
    exposes: [e.light_brightness_colorxy()],
    fromZigbee: [fz.on_off, fz.brightness],
    toZigbee: [
        tz.light_onoff_brightness, tz.light_color,
        tz.light_brightness_move, tz.light_brightness_step,
    ],

},

And everything worked!
I took the model id and manufacturer name from the zigbee2mqtt logs.

There is one minor bug.

If you select "x":0.701,"y":0.299 on the color palette
then the desired color will light up and then immediately fade
out if you then turn off the controller and then switch on the desired color (red) will light up and will not go out until you change the color again and select again "x": 0.701, "y": 0.299

Any other combinations (e.g. "x":0.699, " y":0.3) work correctly.

Please add this device with the official integration.

Externally, it looks identical to similar devices already supported in zigbe2mqtt (model TS0505B and TS0501B).

If you need more information...
I can provide it to you if you tell me how and where to get it.

MilesAhead1, 2021-01-29:
Just received this device in the mail. Perfect timing!

ASSever, 2021-01-29:
> Just received this device in the mail. Perfect timing!

I hope that the developers of zigbee2mqtt will demonstrate no less good timing)

Koenkk, 2021-01-29:
Added, red color should also be fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-29 by Koenkk

CodingGhost, 2023-06-27:
I still have almost the same problem, running 1.31.2. When turning off the controller, it sometimes just goes dark instead of turning off. Also sometimes when turning on, it will fade on and then immidiately turn off until I select a differentz brightness / color.