Koenkk/zigbee2mqtt #5967: New device support: 'ZGRC-KEY-007'

jasperknol, 2021-01-29:
Hi,
I recently bought a 2 button and 8 button zigbee remote from Sunricher;
The 8-button remote is supported; --> Thnx.
The 2 button remote is not; but its 1:1 the same as the 8-button switch (only contains less switches)

But I was able to fit in node_modules/zigbee-herdsman-converters/devices.js
this piece of code:
{
        zigbeeModel: ['ZGRC-KEY-007'],
        model: 'SR-ZG9001K2-DIM2',
        vendor: 'ROBB',
        description: 'Zigbee 2 button wall switch',
        fromZigbee: [fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.battery],
        exposes: [e.battery(), e.action([
            'on_1', 'off_1', 'stop_1', 'brightness_move_up_1', 'brightness_move_down_1', 'brightness_stop_1'])],
        toZigbee: [],
        meta: {multiEndpoint: true},
        whiteLabel: [{vendor: 'Sunricher', model: 'SR-ZG9001K2-DIM2'}],
},

Could you add this as a default supported device?

jasperknol, 2021-01-29:
Image used for the: 
     ROB_200-008-0 --> https://www.zigbee2mqtt.io/devices/ROB_200-008-0.html
  is identical to my device; 
 
 In fact this image should not belong to the "ROB_200-008-0" <-- This is a 4 button device which should use the following image: --> https://www.robbshop.nl/wandzender-4v-zigbee-robb-smarrt

Koenkk, 2021-01-29:
Added your device and fixed the image.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-29 by Koenkk