Koenkk/zigbee2mqtt #5968: Siterwell GS361A-H04, system_mode and Home Assistant

creckord, 2021-01-29:
Hi there,

I'm having some trouble with the support for the Siterwell GS361A-H04 when it comes to its `system_mode`, both in zigbee2mqtt itself, and when used through Home Assistant. 

I tried to grok the mappings for the device, but since I'm fairly new to both zigbee2mqtt and HA, and there have been a bunch of changes and generalizations from the initial GS361 support to Tuya thermostats, it's a bit hard for me to judge how things are supposed to work with those device mappings in the first place. So I'll just list my observations and maybe we can split off separate issues from there:

### 1. Available `system_modes`

The initial contribution for the device had just three system modes, corresponding to what the device exposes through its display and buttons: `off`, `auto` and `manual`. Since then, the list has shifted from GS361 modes to more general Tuya modes and gained four additional modes: `comfort`, `eco`, `boost`, `complex`.

I'm not sure if those additional modes are supposed to be supported by the GS361, but they don't really work with mine (which are white-labeled re!volt models, so maybe they did something to it):
* `comfort`: crashes the device and puts it into F1-1 setup state
* `eco`, `boost`, `complex`: Those all behave identically - they are accepted and acknowledged when set and start off with whatever `current_heating_setpoint` was set before switching into this mode. On the devices display they present with the mode indicators all off. When changing the `current_heating_setpoint`, it is retained until the mode is switched to something else, but when switching back again, the value is lost.

The mode-specific configurations (`eco_temperature`, `comfort_temperature`, `boost_time`) don't seem to have any effect.

### 2. `system_mode`, `preset` and state attributes

I noticed that while I can set both `system_mode` and `preset`, only the `preset` attribute shows up in the device state. This is particularly bothersome because the HA discovery template uses the `system_mode` to read the current state. So I guess either the `system_mode` is missing from the state, or the template should refer to the `preset` instead.

The full json for my devices looks like this:
```
{
  "away_mode": "OFF",
  "battery": 77,
  "child_lock": "UNLOCKED",
  "current_heating_setpoint": 19,
  "linkquality": 30,
  "local_temperature": 19.2,
  "preset": "manual",
  "valve_detection": "ON",
  "window_detection": "ON"
}
```

### 3. `manual` mode and Home Assistant

One last issue with setting the `system_mode` through Home Assistant is that in the device config (`tuya.thermostatSystemModes`) the manual heating mode is called `manual`, while HA insists on it being `heat`.

This leads to `Error: TRV system mode heat is not recognized.` in zigbee2mqtt.

As far as I can see, it would be possible to map the name in the zigbee2mqtt->HA direction using the `mode_state_template`, but not the other way round. Since `tuya.thermostatSystemModes` is the only one of the bunch that uses `manual` instead of `heat` and the GS361 is the only device referring to `tuya.thermostatSystemModes`, maybe it should be renamed to `heat` as well?

Interestingly enough, in the `devices` list, the device description for `system_mode` lists `off`, `auto`, `heat` as supported values. This is hardcoded in the `exposes` config instead of using the `tuya.thermostatSystemModes` list.

As I said, my devices are all white-labeled Revolt thermostats, so I'm not which parts of this come back to them having messed with the firmware, and which are general problems with the GS361 definition. 

I'm happy to dig into the setup further and provide merge requests, but would need to know first which way to go with any potential fixes.

MattWestb, 2021-01-29:
Some (not all) tuya TRV mode / commands / functions can being found [here](https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395) (WIP).

piars777, 2021-01-29:
I modified the devices.js and replaced meta: {tuyaThermostatSystemMode: tuya.thermostatSystemModes, tuyaThermostatPreset: tuya.thermostatPresets} with meta: {tuyaThermostatSystemMode: tuya.thermostatSystemModes3, tuyaThermostatPreset: tuya.thermostatPresets} for Siterwell and i don't have anymore the problem but still remain an other mapping problem. i ll provide more detail later

piars777, 2021-01-29:
In home assistant the system_mode button (the operation list) is never selected. In The devices.js siterwell device is mapped with `fz.tuya_thermostat` when receive data from zigbee. When i change the system_mode i receive this message from zigbee:
`janv. 29 04:32:15 samso npm[7773]: Zigbee2MQTT:debug 2021-01-29 04:32:15: Received Zigbee message from 'Radiateur de la chambre d'ami', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":0,"transid":22}' from endpoint 1 with groupID 0`

As we can see the dp code match the `tuya.dataPoints.mode` in the `fromZigbee.js` and it change the `tuyaThermostatPreset`. 
I also have a saswell device and when i change the system_mode i receive `106` as dp value and it send to HA the tuyaThermostatSystemMode value . 
That means the tuyaThermostatSystemMode is never set for siterwell.
There is no preset mode defined for this device so i think i have to change the `fromZigbee` mapping in the devices.js file to not match `fz.tuya_thermostat` but another one

piars777, 2021-01-29:
I modified the devices.js i replaced `fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report]` with `fromZigbee: [fz.siterwell_thermostat, fz.ignore_basic_report]`
And i added the siterwell_thermostat in the fromZigbee.js file. I just duplicated the tuya_thermostat and replaced the  tuya.dataPoints.mode case with:
`case tuya.dataPoints.mode: {
                const ret = {};
                const presetOk = getMetaValue(msg.endpoint, model, 'tuyaThermostatPreset').hasOwnProperty(value);
                if (presetOk) {
                    ret.preset = getMetaValue(msg.endpoint, model, 'tuyaThermostatPreset')[value];
                    ret.away_mode = ret.preset == 'away' ? 'ON' : 'OFF'; // Away is special HA mode
                    ret.system_mode = tuya.thermostatSystemModes3[value];
                    ret.force = tuya.thermostatForceMode[value];
                } else {
                    console.log(`TRV preset ${value} is not recognized.`);
                    return;
                }
                return ret;
            }` 

And it works.

Koenkk, 2021-01-29:
Could you create a PR?

Koenkk, 2021-01-29:
@creckord can you test out the changes in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2178/files

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

drzony, 2021-01-29:
There is also https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html which can help you find more data points and make sure others are correct

creckord, 2021-01-31:
Doesn't work for me :(

The change from `thermostatSystemModes` to `thermostatSystemModes3` with the change in order of the modes messes up the `system_mode` <-> `preset` relationship (the preset corresponds to the mode shown on the device itself):

```
{
  "away_mode": "OFF",
  "battery": 75,
  "child_lock": "UNLOCKED",
  "current_heating_setpoint": 19,
  "force": "close",
  "linkquality": 33,
  "local_temperature": 19.8,
  "preset": "manual",
  "system_mode": "off",
  "valve_detection": "ON",
  "window_detection": "ON"
}

{
  "away_mode": "OFF",
  "battery": 75,
  "child_lock": "UNLOCKED",
  "current_heating_setpoint": 22,
  "force": "open",
  "linkquality": 33,
  "local_temperature": 19.8,
  "preset": "schedule",
  "system_mode": "heat",
  "valve_detection": "ON",
  "window_detection": "ON"
}

{
  "away_mode": "OFF",
  "battery": 75,
  "child_lock": "UNLOCKED",
  "current_heating_setpoint": 22,
  "force": "open",
  "linkquality": 33,
  "local_temperature": 19.8,
  "preset": "schedule",
  "system_mode": "heat",
  "valve_detection": "ON",
  "window_detection": "ON"
}
```

Initially, valve and window detection also were inconsistent with the json state, but toggling them off/on once seems to have fixed that - didn't notice them becoming messed up again when changing state.

Also, the "force: open/closed" tied to the mode in that way looks suspicious, but since it's only added to the json state on read, it doesn't do anything detrimental.

creckord, 2021-01-31:
I played around with the pull request, removed the "force" mapping in the mode mapping code and introduced a `thermostatSystemModes4`:
```
const thermostatSystemModes4 = {
    0: 'off',
    1: 'auto',
    2: 'heat'
};
```
That way, it works great.

piars777, 2021-02-01:
Can you explain me which mapping you change?
Because in the devices.js file i replace `tuya.thermostatSystemModes3` with `tuya.thermostatSystemModes4` and i create this new constan and i have again the error.
I test with this trv: [https://www.amazon.fr/gp/product/B08357D6LJ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url)
i also using home assistant, with this model i don't see any preset value.

creckord, 2021-02-02:
@piars777 see the commit on top of yours in my updated PR.

Koenkk, 2021-02-03:
Assuming this can be closed as https://github.com/Koenkk/zigbee-herdsman-converters/pull/2194 is merged

closed 2021-02-03 by Koenkk