Koenkk/zigbee2mqtt #5969: State of Philips hue bulb is never updated when controlled by Busch-Jaeger 6735

asmolarczyk, 2021-01-29:
Hi,

I'm using a Busch-Jaeger 6735 switch to control my zigbee devices. In order to use the dimmable scenes of this switch, I had to remove the bindings to the zigbee2mqtt coordinator. This setup works quite good when using e.g. Mueller Tint bulbs because they support attribute reporting. The state of the bulbs is immediately updated in zigbee2mqtt.

Unfortunately my Philips hue bulbs do not support reporting. Therefore the new state of the bulbs (on/off, brightness) is never updated. I already tried to use report_emulate, but I think this does not help here.

When using the Philips hue bridge, it takes a few seconds, but then the state of the bulbs is updated in the app. Is it somehow possible to get the same polling-based update behaviour in zigbee2mqtt?

I'm on the dev branch.

Koenkk, 2021-01-29:
As a work around you can try putting the heu bulb in a group and binding the 6735 to the group instead of directly to the device.

asmolarczyk, 2021-01-29:
Thank you! This is a nice workaround.
Unfortunately (of course) the bulbs restore to the last brightness level when using this approach. With the default dimmable scenes as described by busch jaeger it is possible to recall a scene

closed 2021-03-09 by github-actions[bot]