Koenkk/zigbee2mqtt #5970: mqtt publish json only changes first value

Gibbz, 2021-01-29:
## What happened
MQTT Json payloads are only applying the first value.

## What did you expect to happen
As per the documentation, all values sent in the json blob should be sent on to the device.
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

topic: zigbee2mqtt/Alarm A/set
payload: '{"melody":18,"volume":"low","alarm": true}'

the actual message sent from the log:
INFO: MQTT publish: topic 'zigbee2mqtt/Alarm A', payload '{"alarm":false,"humidity":54,"humidity_alarm":false,"linkquality":102,"melody":1,"temperature":"21.6","temperature_alarm":false,"volume":"medium"}'

## How to reproduce it (minimal and precise)
Send the payload to change multiple values at once...
topic: zigbee2mqtt/Alarm A/set
payload: '{"melody":18,"volume":"low","alarm": true}'

## Debug info
Zigbee2MQTT version: 1.17
Adapter hardware: zzh
Adapter firmware version:

Gibbz, 2021-03-01:
This is still an issue.

Koenkk, 2021-03-01:
What do you mean with not sent? Do you mean that they are not used by the device?

Gibbz, 2021-03-03:
That's what it shows in the log. I guess it depends what the log is showing, what was sent to the device? Or what the device accepted? 
I assume the log is showing what is sent to the device, so then in this case it is not sending the command I am giving the zigbee2mqtt, only the first attribute of my json is accepted.

Koenkk, 2021-03-03:
What was sent to the device

Gibbz, 2021-03-09:
Neo NAS-AB02B0
https://www.zigbee2mqtt.io/devices/NAS-AB02B0.html#neo-nas-ab02b0

esion-net, 2021-04-02:
Tried that today, too. If sending more than one attribute at once, log confirms only sending the first one. 
Eq:
payload: '{"melody":18,"volume":"low","alarm": true}'
Only 'melody' seems to be handled and confirmed as per logs.

closed 2021-05-11 by github-actions[bot]

weltmeyer, 2023-05-17:
I have the same Issue. Only the first Json property gets applied. Any updates on this?