Koenkk/zigbee2mqtt #5971: IKEA ota upgrade - actually downgrades firmware

a8ree, 2021-01-29:
## What happened
On the Dev branch, i have used the IKEA ota to update my uk tradfri socket. I was previously running 2.0.019 however after the update I am on a lower verions - 1.4.020

## What did you expect to happen
It would be expected that the firmware update would deliver a newer firmware version

## How to reproduce it (minimal and precise)
Pair uk tradfri socket that is on 2.0.019
Check current firmware version (about tab)
Click the icon in the Zigbee2Mqtt assistant web interface to update firmware
Allow update to complete
Check firmware version (about tab)

## Debug info
Zigbee2MQTT version: 1.17.0-dev
Adapter hardware: cc2652r (zzh!)
Adapter firmware version:

MattWestb, 2021-01-30:
The normal feed:
`{"fw_binary_url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.13.019/bin/10005777-6.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed","fw_file_version_LSB":17955,"fw_file_version_MSB":8194,"fw_filesize":202110,"fw_image_type":4353,"fw_manufacturer_id":4476,"fw_type":2},`

The Test feed:
`{"fw_binary_url":"http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.12.030/bin/10005777-4.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed","fw_file_version_LSB":17955,"fw_file_version_MSB":8194,"fw_filesize":201030,"fw_image_type":4353,"fw_manufacturer_id":4476,"fw_type":2},`

So Z2M must finding the old firmware from some other place on the "cyber space" is not amazon servers is running amok.

PM: All control-outlet is using the same OTA files but is having different factory data setting for doing the different models for different markets.

a8ree, 2021-01-30:
> The normal feed:
> {"fw_binary_url":"
> http://fw.ota.homesmart.ikea.net/global/GW1.0/01.13.019/bin/10005777-6.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed
> ","fw_file_version_LSB":17955,"fw_file_version_MSB":8194,"fw_filesize":202110,"fw_image_type":4353,"fw_manufacturer_id":4476,"fw_type":2},
>
> The Test feed:
> {"fw_binary_url":"
> http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.12.030/bin/10005777-4.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed
> ","fw_file_version_LSB":17955,"fw_file_version_MSB":8194,"fw_filesize":201030,"fw_image_type":4353,"fw_manufacturer_id":4476,"fw_type":2},
>
> So Z2M must finding the old firmware from some other place on the "cyber
> space" is not amazon servers is running amok.
>
> PM: All control-outlet is using the same OTA files but is having different
> factory data setting for doing the different models for different markets.

I found the firmware repo on a Reddit thread too. Couldn't work out where
this is defined in the code though?

smulle48, 2021-01-31:
> The normal feed: [...]

IKEA has indeed turned to an older version in their test releases. Perhaps an system wide bug??

Official: http://fw.ota.homesmart.ikea.net/feed/version_info.json
Test: http://fw.test.ota.homesmart.ikea.net/feed/version_info.json

MattWestb, 2021-01-31:
Third time i have checking and still 10005777-4.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed
[IKEATEST.txt](https://github.com/Koenkk/zigbee2mqtt/files/5900501/IKEATEST.txt)

a8ree, 2021-02-01:
I've noticed this in my debug log today

Debug Is new image available for '0x000d6ffffe2b5a69', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4353,"fileVersion":537019939}', latest meta '{"fileVersion":537019939,"url":"http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.12.030/bin/10005777-4.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed"}'
Debug Update available for '0x000d6ffffe2b5a69': NO

This indicates that it is finding the latest firmware - but that device is on version 1.4.020

I wonder whether the code determining the version is looking at the folder name rather than the file name for the version?
i.e. it is comparing 1.4.020 to 1.12.030 rather than 1.4.020 to 2.0.024

MattWestb, 2021-02-01:
@a8ree Have you trying re power the outlet so its doing one device accouterments and Z2M is reading the attributes a gen ? 
Also restarting Z2M may helping.

closed 2021-03-11 by github-actions[bot]