Koenkk/zigbee2mqtt #5976: Frequent crashes (New install)

TilloNeko, 2021-01-30:
# Bug Report
## What happened
It caches after some time

## What did you expect to happen
Not to cache

## How to reproduce it (minimal and precise)
Just running it for 24h... :-(

## Debug Info
Zigbee2MQTT version: Zigbee2MQTT version 1.17.0 (commit #435a64a),  zigbee-herdsman (0.13.46)
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: CC2652RB_20201026

You can clearly see when it caches in the logs.
I just installed it on a fresh rapi, only other program running on the setup is a mqtt server

Link to log: [https://pastebin.pl/view/47a3be73](url)

TilloNeko, 2021-01-30:
This is my config.

[https://pastebin.pl/view/e75841c0](url)

Koenkk, 2021-01-30:
Try disabling the availabilty feature first (remove `availability_timeout: 60` from your config)

TilloNeko, 2021-01-31:
I removed the `availability_timeout: 60` from the config file, but it did not help.

During the night it crashed one more time, is started with a lot of timeouts eg.
`error 2021-01-30 07:51:06: Publish 'set' 'brightness' to '0x842e14fffe3b7e0b' failed: 'Error: Command 0x842e14fffe3b7e0b/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)'
info  2021-01-30 07:51:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0x842e14fffe3b7e0b' failed: 'Error: Command 0x842e14fffe3b7e0b/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":0,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"0x842e14fffe3b7e0b"},"type":"zigbee_publish_error"}',`

and then it flagged all with payload offline

`info  2021-01-30 07:52:36: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe889a92/availability', payload 'offline'
error 2021-01-30 07:52:36: Failed to ping '0x842e14fffe889a92'`

logs included at pastbin

[https://pastebin.pl/view/ddf75eef](url)

Koenkk, 2021-01-31:
When you remove `availability_timeout: 60` from your config you should not get the failed to ping errors anymore (because zigbee2mqtt only pings when availability is enabled). How does your config look now?

TilloNeko, 2021-01-31:
Sorry my bad. I uploaded the wrong log. See link for new log [https://pastebin.pl/view/4808c28a](url)

It still crashed, but not in the same way. 

If im reading the log correctly, this is the point when it failed:
`info  2021-01-31 00:40:03: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeaec637', payload '{"brightness":0,"color":{"x":0.1497,"y":0.0619},"color_temp":492,"linkquality":69,"state":"OFF"}'`

`error 2021-01-31 00:40:09: Publish 'set' 'color' to '0x842e14fffe25e52c' failed: 'Error: Command 0x842e14fffe25e52c/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":9811,"colory":4057}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)'`

Koenkk, 2021-01-31:
Can you share more details about your system?
- Computer? (raspberry pi?)
- Other devices connected via USB
- Home automation software (e.g. home assistant/openhab)

TilloNeko, 2021-01-31:
Hi Koenkk, thanks fort taking your time to look into this.

**Computer?** Raspberry Pi 2B V2.1
**Other devices connected via USB?** No other devices
**Home automation software (e.g. home assistant/openhab)?** Domoticz on separate computer running domoticz-zigbee2mqtt-plugin, connecting to a mqtt broker (mosquitto) running on the same RaPi as zigbee2mqtt.

dlabanec, 2021-02-01:
Zigbee2MQTT:info  2021-02-01 11:39:19: Logging to console and directory: 'config/share/zigbee2mqtt/log/2021-02-01.11-39-18' filename: log.txt
Zigbee2MQTT:info  2021-02-01 11:39:19: Starting Zigbee2MQTT version 1.17.0 (commit #1.17.0)
Zigbee2MQTT:info  2021-02-01 11:39:19: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:error 2021-02-01 11:39:31: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-01 11:39:31: Failed to start zigbee
Zigbee2MQTT:error 2021-02-01 11:39:31: Exiting...
Zigbee2MQTT:error 2021-02-01 11:39:31: Error: AREQ - ZDO - activeEpRsp after 10000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-02-01T10_39_31_825Z-debug.log
2021-02-01T11:39:33: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2021-02-01T11:39:33: PM2 log: App [npm:0] starting in -fork mode-
2021-02-01T11:39:33: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js

PI4, CC2531, HA 32bit. Can somebody help? worked well, yesterday crash propably after new update, I tryed everhiting, unplug, reset, etc many times :(

TilloNeko, 2021-02-01:
> 2021-02-01T11:39:33: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
> 2021-02-01T11:39:33: PM2 log: App [npm:0] starting in -fork mode-
> 2021-02-01T11:39:33: PM2 log: App [npm:0] online
> 
> > zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> > node index.js
> > Zigbee2MQTT:info  2021-02-01 11:39:19: Logging to console and directory: 'config/share/zigbee2mqtt/log/2021-02-01.11-39-18' filename: log.txt
> > Zigbee2MQTT:info  2021-02-01 11:39:19: Starting Zigbee2MQTT version 1.17.0 (commit #1.17.0)
> > Zigbee2MQTT:info  2021-02-01 11:39:19: Starting zigbee-herdsman (0.13.46)
> > Zigbee2MQTT:error 2021-02-01 11:39:31: Error while starting zigbee-herdsman
> > Zigbee2MQTT:error 2021-02-01 11:39:31: Failed to start zigbee
> > Zigbee2MQTT:error 2021-02-01 11:39:31: Exiting...
> > Zigbee2MQTT:error 2021-02-01 11:39:31: Error: AREQ - ZDO - activeEpRsp after 10000ms
> > at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
> > at listOnTimeout (internal/timers.js:549:17)
> > at processTimers (internal/timers.js:492:7)
> > npm
> > ERR! code ELIFECYCLE
> > npm ERR! errno 1
> > npm
> > ERR! zigbee2mqtt@1.17.0 start: `node index.js`
> > npm ERR! Exit status 1
> > npm ERR!
> > npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
> > npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> > npm ERR! A complete log of this run can be found in:
> > npm ERR!     /root/.npm/_logs/2021-02-01T10_39_31_825Z-debug.log
> 
> PI4, CC2531, HA 32bit. Can somebody help? worked well, yesterday crash propably after new update, I tryed everhiting, unplug, reset, etc many times :(

**Hello, I do not think this is related to the same issue.** 

I would check the npm logs /root/.npm/_logs/2021-02-01T10_39_31_825Z-debug.log

pls, create a separate issue for this

Koenkk, 2021-02-01:
Can you try this fw: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin  (https://github.com/Koenkk/zigbee2mqtt/discussions/6019)

TilloNeko, 2021-02-01:
Okey, I updated the controller with the firmware. Will report back when I know if it worked or not.

TilloNeko, 2021-02-02:
So far everything have been running smooth, will report back in a day or so with more information.

In the meantime, could you do a short breakdown of the differences between this firmware and the other firmware that was already on the controller?

dlabanec, 2021-02-02:
yesterday i reflash my usb stick. But with same firmware like before... Now is workink back well. I dont understand how can update make crash on stick :/

TilloNeko, 2021-02-03:
Okey, after two days now the system have worked fine, the new firmware seams to have helped.

But some other minor problems have occurred, hard to explain.

@Koenkk  could you do a short breakdown of the differences between this firmware and the other firmware (CC2652RB_20201026) that was already on the controller?

And whats the next step?

Koenkk, 2021-02-03:
@TilloNeko probably you experienced the same issue as https://github.com/Koenkk/zigbee2mqtt/discussions/6019 . The new firmware uses the new z-stack  SDK in which this issue was fixed.

TilloNeko, 2021-02-07:
Have now been running it for several days without crashing. the new firmware definitely fixed it.

Koenkk, 2021-02-07:
Cool, assuming this can be closed now.

closed 2021-02-07 by Koenkk