Koenkk/zigbee2mqtt #5977: HG06336 Lidl/Silvercrest

hansgans007, 2021-01-30:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Hello,

https://pastebin.com/hqVgJBaT here is my Debuglog its only this Device for testing.

I have 1 CC2531 on Windows 10 for testing before it goes back to Server on Linux.

I test with 3 Window Sensor 

2 of them gets the Adress 3rd do nothing but that not the problem.

hansgans007, 2021-01-30:
2021-01-30 09:29:41: MQTT publish: topic 'test123/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.0-dev"}'

hansgans007, 2021-01-30:
Maybe its because that are 3.0 Zigbee window sensor ?

Koenkk, 2021-01-30:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/5814

closed 2021-01-30 by Koenkk