Koenkk/zigbee2mqtt #5978: Availability_timeout doesn't work

JanvdW304, 2021-01-30:
In the advanced section of the configuration.yaml I have added `availability_timeout: 600` to periodically check and update the status of my Tradfri bulbs (as described on https://www.zigbee2mqtt.io/information/configuration.html). This is needed because the bulbs are physically powered off now and then. Unfortunately availability_timeout makes not any difference: the Domoticz status of bulbs that are physically switched off still remain 'on'.

I am running on Domoticz  2020.2 build 12893,  Zigbee2MQTT v.1.17.0-dev, Domoticz Python Plugin v.0.2.1 and Python 3.7.3.

heggink, 2021-01-30:
Have you set report: true in the advanced section?

Herman Eggink

JanvdW304, 2021-01-30:
Yes, this the advanced section
`advanced:
  ikea_ota_use_test_url: true
  report: true
  availability_timeout: 600
  network_key:
  <network key>
`

JanvdW304, 2021-01-30:
Additional info: I am using a CC2531 USB sniffer

JanvdW304, 2021-02-27:
I have decreased the timeout value to 60, but that makes no difference: when a Tradfri device is powered on the Domoticz status is updated (due to 'report: true'), but when a Tradfri device is powered off the Domoticz value doesn't change (as should because of the 'availability_timeout: 60'). Does anyone has an idea I can solve the issue? Or any suggestions what I can test?

closed 2021-04-06 by github-actions[bot]