Koenkk/zigbee2mqtt #5991: Changed from CC2530+CC2591 over UART to slaesh’s CC2652RB stick and only getting "No such file or directory, cannot open /dev/ttyUSB0'"

hightow, 2021-01-30:
## What happened
Im running on these issues on a raspberry pi 4 running raspberry pi OS from a SSD connected over USB.

Since I changed from my old CC2530+CC2591 adapter that was connected over UART over to slaesh’s CC2652RB stick I can no longer start zigbee2mqtt. It always fails with: 
`
Zigbee2MQTT:error 2021-01-30 20:25:00: Failed to start zigbee,
Zigbee2MQTT:error 2021-01-30 20:25:00: Exiting...,
Zigbee2MQTT:error 2021-01-30 20:25:00: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0',
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)`

I followed the update guide and also flashed the firmware according to the guide here https://slae.sh/projects/cc2652/#installation myself when it wasn't working in the first place.

- I removed the database.db.
- updated the configuraton.yaml:
  - set a new pan_id 
`advanced:
  pan_id: 0x1a66`
  - set the new serial port:
  `serial:
  port: /dev/ttyUSB0`
  - verified the serial port: 
  `ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Jan 30 20:17 usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_4C_52-if00-port0 -> ../../ttyUSB0`
  - verified write access to the port:
  `test -w /dev/ttyUSB0 && echo success || echo failure
success`

did several reboots and now am out of ideas.

## What did you expect to happen

That zigbee2mqtt would start and correctly recognize the new adapter.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: CC2652RB_20201026.hex from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

nurikk, 2021-01-31:
Set serial: port to null to enable auto detection

hightow, 2021-01-31:
Nevermind, I didn't catch that my docker-compose file wasn't properly updated and still mapped the old `/dev/ttyACM0` instead of the new `/dev/ttyUSB0`. So after changing that and doing a `docker-compose up -d` it finally recognized the new port and all seems to work fine now!

closed 2021-01-31 by hightow