Koenkk/zigbee2mqtt #6011: Xiaomi SSM-U02 latency issue

Toukite, 2021-01-31:
Hello,

I am using z2m on HassOS and with ConbeeII. I have about 15 devices and it works like a charm.

There is one thing bothering me though, regarding Xiaomi SSM-U02 devices: there is a latency of about 2 seconds between the moment I send an order (turn on the light for example), and the moment the light actually turns on.

I also have sonoff basiczbr3 for other light usecase, and this one answers immediately. Also immediate motion detection on other devices, only the Xiaomi light module seems to have this issue.

Same behavior if I trigger with automation or with lovelace button or directly from z2m: about 2sec delay on this particular device.

Is there something I can do to remove that latency ?
I have a small passage room where light is supposed to turn on when motion/door event is detected, and I am often in the middle of the room before I actually have light :)

Thanks for the help ! Please tell me if I can provide other information

Toukite, 2021-03-03:
Just commenting to avoid an autoclose as this has not been treated yet

Koenkk, 2021-03-03:
Can you (or somebody else) check if this is the same behaviour as with the original gateway? I expect that because this looks like an en device (since it has no neutral)

closed 2021-04-10 by github-actions[bot]

jd1900, 2021-06-26:
I'm interested in this as I was about to buy it.

Anyone with this device?