Koenkk/zigbee2mqtt #6015: Pairing Problems after migrated CC2531 -> CC2652RB

sp11dy, 2021-02-01:
## What happened
After migrating from CC2531 to CC2652RB adapter all my Xiaomi Aqara temperature sensors won't pair with the coordinator again (WSDCGQ01LM & WSDCGQ11LM). 
There is nothing to see in the log .
Other Devices (also Aqara) are pairing with no problem.
The panid was changed and I also tried the dev branch.

## What did you expect to happen
Some output from the log, at least in debug mode. 
The devices LED doesn't even show any acknowledgement.

## How to reproduce it (minimal and precise)
I have changed the batteries, changed USB Port and used a USB extension for the adapter.
After fresh reinstalling z2m 1.17.0.1 and using the CC2531 I have the same problem.

## Debug info
Zigbee2MQTT version: 1.17.0.1 Hassio Addon
Adapter hardware: CC2652RB
Adapter firmware version: Z-Stack_3.x.0 20201026

juslex, 2021-02-01:
Did you changed tem pan_id, deleted the older database.db and reparing all devices? It’s recommended after changing the coordinator. 

Look this:

https://github.com/Koenkk/zigbee2mqtt/issues/4151

sp11dy, 2021-02-01:
Yes I did all of this. At some point I deleted all the files and made a fresh install, but the same problem occurs.

sp11dy, 2021-02-01:
I found the problem. I've got a bad batch of batteries from aliexpress. New Varta CR2032 and the devices paired again.
I guess the charge was low when I switched the adapter and the "new" batteries also didn't have enough charge.

closed 2021-02-01 by sp11dy