Koenkk/zigbee2mqtt #6021: Samsung multi

MaxVonEvil, 2021-02-01:
Koen, first thanks for starting this excellent project. I've mostly gotten things to work, yet I'm struggling to wrap my head around dealing with device comparability and  herdsman converters. I was thus hoping for a little help from the Jedi academy to deal with a sensor giving me some challenges:

**## What happened**
I've added several Samsung multiv4 sensors, which are already known in devices.js 
They pair fine and work (mostly) as as expected: Specifically the contact, movement and temperature sensors are reporting as expected. However there are 3 things which are not, and I'm not entirely sure how to fix, heardmans converters and all.

**1) Battery not being reported**
Referencing the log entries below, it would appear battery level is not being read correctly. Debug log does not indicate any battery level is being reported. While the sensor may possibly only be reporting battery when the voltage changes, I've tried changing a battery to provoke a change. As far as I could tell it made no difference

On the MQTT side, the battery payload is always 0, yet a payload attribute called voltage reports the value 2400 consistently. Once could conclude that it would be a matter of remapping attributes, however it raised my suspicion seeing four identical devices with different battery ages all report the 2400 voltage value. In other words, that value is likely not the real mV battery voltage.

On the HA side, it appears that said voltage payload, bogus or not, is being filtered out, as no entity is created for it. I guess this make sense, as a voltage attribute is seen on switch/dimmer type devices, referring to line voltage, not battery. 

**2) Tilt XYZ values not being reported**
I would like to understand how to add these attributes as I use them in some automations. I found the original SmartThings DTH and put it here: https://pastebin.com/YBX1G233
I figured it would make sense to include as it's the same device, yet if this is better dealt with in a separate issue, let me know.

Below are a what I'm seeing in the log for one of these devices, if it helps:

0|zigbee2mqtt 1.17.0  | Zigbee2MQTT:debug 2021-02-01 04:39:20: Received Zigbee message from 'door.patio', type 'attributeReport', cluster 'manuSpecificSamsungAccelerometer', data '{"acceleration":0}' from endpoint 1 with groupID 0

0|zigbee2mqtt 1.17.0  | Zigbee2MQTT:info  2021-02-01 04:39:20: MQTT publish: topic 'zigbee2mqtt/door.patio', payload '{"battery":0,"battery_low":false,"contact":false,"linkquality":76,"moving":false,"tamper":false,"temperature":24.41,"voltage":2400}'

## What did you expect to happen
Battery reporting % correct and XYZ tilt data is available

## How to reproduce it (minimal and precise)
Install a multiv4 sensor.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: [CC2538+CC2592](https://zigdobby.com/)
Adapter firmware version: not sure.

Thanks
/Max

crowbarz, 2021-02-09:
I have 3 `multiv4` sensors on my network, and they report voltages of 1600, 1700 and 1800. Looking at the [SmartThings DTH](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy#L284) it looks like the voltage is a value from 1500 to 2800 and converted to a percentage through a mapping function:
```javascript
def batteryMap = [28: 100, 27: 100, 26: 100, 25: 90, 24: 90, 23: 70,
    22: 70, 21: 50, 20: 50, 19: 30, 18: 30, 17: 15, 16: 1, 15: 0]
def minVolts = 15
def maxVolts = 28
```

In z2m, [utils.batteryVoltageToPercentage()](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/utils.js#L116) is used to convert the battery voltage to a percentage, but none of the existing algorithms look relevant. So it looks to me a new conversion option would need to be defined here, and the `voltageToPercentage` property in the `devices.js` model for `multiv4` updated accordingly.

You might want to reconfigure your sensor via the UI and see if it sets up reporting for the `powerCfg` cluster. On one of mine, I needed to remove and rejoin the sensor to get it to report this cluster after upgrading to the latest z2m (I'm running 1.17.1-dev at the moment).

On my HA instance, a `battery` entity was created for my sensors, but I `null`ed it out in `devices.yaml` because it was constantly zero. I left the `battery_low` in but tbh I can't figure out how that state is updated. You might want to go to Configuration > Devices and look at the MQTT info to see what device config is being received from z2m.

Tilt is another can of worms ... I couldn't get the relevant attributes (`x_axis`, `y_axis` and `z_axis`, defined [here](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts#L3931)) to report due to #6104 but that was fixed. I haven't looked at it since though. I am guessing that it will probably need additional converters in `fromZigbee.js` to convert the report into the equivalent values that SmartThings reports, though the SmartThings DTH does some crazy vendor dependent value swapping in their code.

declanshanaghy, 2021-02-18:
I'm moving my Zigbee sensors out of Home Assistant ZHA to z2m and just noticed this missing too. I have some sensors setup in HA which depend on these attributes.

Wouldn't mind contributing a fix for it but need a pointer or two on what's needed

o0v7r0o, 2021-02-25:
I have recently adjusted the battery conversion https://github.com/Koenkk/zigbee-herdsman-converters/pull/2291
(it does not show power source 'battery' though, only percentage)
It would be appreciated if you can checkout my branch and test it with your devices.

o0v7r0o, 2021-02-25:
Hi, the PR have been merged recently and should be available in the dev branch.
To fix the power source I would prefer to compare the data from the other sensor.
@MaxVonEvil @declanshanaghy @crowbarz if you can enable debug logs and repair one sensor, please can you post lines like
`No converter available for 'F-MLT-US-2' with cluster...` here?
AFAIU these messages are ok during the pairing as soon as they don't need to be converted to mqtt, but they are setting the device properties.

crowbarz, 2021-03-07:
@o0v7r0o looks like your PRs were merged into 1.18.0, I just upgraded to 1.18.1 and reconfigured my two F-MLT-US-2 sensors. The battery percentage shows more to expectations now, thanks for fixing that! I also re-paired one of the sensors, and the only converter warnings logged during pairing were:
```
Zigbee2MQTT:debug 2021-03-07 17:29:57: No converter available for 'F-MLT-US-2' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'
Zigbee2MQTT:debug 2021-03-07 17:29:59: No converter available for 'F-MLT-US-2' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b00219ecc7b","zoneState":1}'
```
The battery level does not display properly in the z2m UI, but you note in Koenkk/zigbee-herdsman-converters#2291 that there are issues with `powerSource` reporting on this sensor which probably causes this behaviour. (FTR the sensor that was re-paired reports `{"powerSource":0}` like yours.) The `battery` attribute is exported to HA/MQTT properly though, for me that's the important bit. There's still a superfluous `battery_low` attribute that can probably be removed now that `battery` works properly.

Note that `genBasic` related converter errors were muted in Koenkk/zigbee-herdsman-converters#2237.

o0v7r0o, 2021-03-24:
I assume that can be closed as solved?

closed 2021-03-25 by Koenkk