Koenkk/zigbee2mqtt #6027: Command with same ID is not corretly identified (?)

MSchutt, 2021-02-01:
It seems that with one of the latest updates my custom cluster is no longer recognized:

My Tuya LEDs (TS0505A) use the command "moveToHueAndSaturation" a bit differently. They additionally send the brightness. I have defined the cluster as such:

``` javascript
moveToHueAndSaturation: {
    ID: 6,
    parameters: [
        { name: 'hue', type: dataType_1.default.uint8 },
        { name: 'saturation', type: dataType_1.default.uint8 },
        { name: 'transtime', type: dataType_1.default.uint16 },
    ],
},
tuyaMoveToHueAndSaturationBrightness: {
    ID: 6,
    parameters: [
        { name: 'hue', type: dataType_1.default.uint8 },
        { name: 'saturation', type: dataType_1.default.uint8 },
        { name: 'transtime', type: dataType_1.default.uint16 },
        { name: 'brightness', type: dataType_1.default.uint8 },
    ],
},
```

When I now issue the command:
`await entity.command('lightingColorCtrl', 'tuyaMoveToHueAndSaturationBrightness', { hue: 243, saturation: 218, transtime: 0, brightness: 255 }, {disableDefaultResponse: true});`

and sniff the ZigBee traffic only `hue`, `saturation` and `transtime` are sent, thus making my LEDs immediately turn off (it seems that brightness is set to 0 automatically) (please see attached Wireshark screenshot for the payload)

This has worked previously.. Do you know what the problem might be here? Is there a way to use `entity.write(..)` instead of the command? It seems that this command is not used that often so we might be able to get rid of it (?).

Thank you for your time and this awesome project! :)

Wireshark payload: [https://prnt.sc/xz6njz](url)

Koenkk, 2021-02-01:
What is the manufacturer name of your TS0505A?

MSchutt, 2021-02-01:
The fingerprint of the device is: `{modelID: 'TS0505A', manufacturerName: '_TZ3000_sosdczdl'}`

Quick update: I have tried to manually change the order in the cluster.js file (so I switch `moveToHueAndSaturation` with `tuyaMoveToHueAndSaturationBrightness` and now brightness is sent as well and it works just fine

sjorge, 2021-02-01:
I think that fact that they share an ID is a problem. I wonder if we shoiuld have a way to define manufacturer specific attributes on the zcl commands.

MSchutt, 2021-02-01:
Yeah, you are probably right. What is the alternative so that I can pass brightness as well?

sjorge, 2021-02-01:
```
moveToHueAndSaturation: {
    ID: 6,
    parameters: [
        { name: 'hue', type: dataType_1.default.uint8 },
        { name: 'saturation', type: dataType_1.default.uint8 },
        { name: 'transtime', type: dataType_1.default.uint16 },
        { name: 'tuyaBrightness', type: dataType_1.default.uint8 },
    ],
},
```

I guess, but then we would hit problems if they ever expand the ZCL. Hence my question mostly aimed at Koen about maybe having a way to specify 'attributes' per manufacturer. Although I am not super familiar with the internals of herdsman so there might be some technical reason for that not going to work.

MattWestb, 2021-02-02:
@sjorge ZCL ver 9 is in the pipe (i think its for the upcoming Zigbee R23 that is being cooked in most zigbee stacks) so you is getting little to do in the future ;-)

Koenkk, 2021-02-02:
@MSchutt should be fixed now, please confirm.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

MSchutt, 2021-02-04:
Hi - thank you very much for the fast fix - it works just fine now :)

closed 2021-02-04 by MSchutt