Koenkk/zigbee2mqtt #6035: Failed to call 'OTAUpdate' 'onZigbeeEvent' (AssertionError [ERR_ASSERTION]: No image available for imageType '0'

70MM13, 2021-02-01:
## What happened
Running dev since last week, also updated today.
Seems like an OTA check is automatically run? didn't notice before. Correct report for one device, but I think another device is generating the following error:
```
Zigbee2MQTT:debug 2021-02-01 21:11:08: Received Zigbee message from 'floodlight_achter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":2,"imageType":0,"manufacturerCode":4687}' from endpoint 11 with groupID null
Zigbee2MQTT:debug 2021-02-01 21:11:08: Check if update available for '0xbc33acfffe8b839d' (GL-FL-004P)
Zigbee2MQTT:error 2021-02-01 21:11:08: Failed to call 'OTAUpdate' 'onZigbeeEvent' (AssertionError [ERR_ASSERTION]: No image available for imageType '0'
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:22:5)
    at process._tickCallback (internal/process/next_tick.js:68:7))
```

## What did you expect to happen
No assertion error:  seems like an unhandled exception?
This is what I get when check is triggered manually via MQTT:
```
Zigbee2MQTT:debug 2021-02-01 21:21:46: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"floodlight_achter"}'
Zigbee2MQTT:info  2021-02-01 21:21:46: Checking if update available for 'floodlight_achter'
Zigbee2MQTT:info  2021-02-01 21:21:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'floodlight_achter'","meta":{"device":"floodlight_achter","status":"checking_if_available"},"type":"ota_update"}'
Zigbee2MQTT:debug 2021-02-01 21:21:46: Check if update available for '0xbc33acfffe8b839d' (GL-FL-004P)
Zigbee2MQTT:debug 2021-02-01 21:21:46: Using endpoint '11'
Zigbee2MQTT:debug 2021-02-01 21:21:48: Received Zigbee message from 'floodlight_achter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":2,"imageType":0,"manufacturerCode":4687}' from endpoint 11 with groupID null
Zigbee2MQTT:debug 2021-02-01 21:21:48: Got OTA request '{"fieldControl":1,"manufacturerCode":4687,"imageType":0,"fileVersion":2}'
Zigbee2MQTT:info  2021-02-01 21:21:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'floodlight_achter' (No image available for imageType '0')","meta":{"device":"floodlight_achter","status":"check_failed"},"type":"ota_update"}'
Zigbee2MQTT:info  2021-02-01 21:21:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"floodlight_achter"},"error":"Failed to check if update available for 'floodlight_achter' (No image available for imageType '0')","status":"error"}'
Zigbee2MQTT:error 2021-02-01 21:21:48: Failed to check if update available for 'floodlight_achter' (No image available for imageType '0')
```

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.0-dev 
Adapter hardware:  Conbee II
Adapter firmware version: 0x26680700

closed 2021-03-11 by github-actions[bot]

krasevych, 2021-07-06:
I got same problem, hot to fix?

tenkaibr, 2021-09-17:
From time to time i get this error too

`Zigbee2MQTT:error 2021-09-17 05:03:57: Failed to call 'OTAUpdate' 'onZigbeeEvent' (AssertionError [ERR_ASSERTION]: No image available for imageType '0'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:22:5)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:264:18)
    at async Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:255:23)
    at async OTAUpdate.onZigbeeEvent (/app/lib/extension/otaUpdate.js:52:31)
    at async Controller.callExtensionMethod (/app/lib/controller.js:380:21))`