Koenkk/zigbee2mqtt #6036: OTA on TS0601 thermostat fail

copystring, 2021-02-01:
I have a few of these: https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

Wanted to perform OTA but it says no new image available although a link to the new image is visible in the log.

https://pastebin.com/MXAjp8Nh

Shouldn't there be a version visible in the web interface? I guess it is not reporting the version and fails because of that?

[image]

closed 2021-03-11 by github-actions[bot]

bruvv, 2021-06-04:
@copystring could you open this again? I have the same issue. Running 1.19.1 of zigbee

wimpie007, 2021-06-05:
same issue here.

VikeDragon, 2022-03-16:
same issue here.

GamerClassN7, 2022-09-13:
Same here any progress ?