Koenkk/zigbee2mqtt #6040: Ikea OTA update fails

mgiebels, 2021-02-01:
I want to update my Ikea PLugs but did doesnt work.
I have de latetest version of ziggbee2Mqt22
zigbee2mqtt version
1.17.1 commit: 1.17.1
coordinator type
ConBee2/RaspBee2
coordinator revision
0x26660700
frontend version
0.3.23

The error did i get is :

Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Switch.1"},"error":"Update of 'Switch.1' failed (Image is truncated: not long enough to contain a valid tag)","status":"error","transaction":"73z61-3"}'
Error Update of 'Switch.1' failed (Image is truncated: not long enough to contain a valid tag)

m0wlheld, 2021-02-01:
See https://www.zigbee2mqtt.io/information/ota_updates.html#using-the-ikea-tradfri-test-server

mgiebels, 2021-02-02:
Doenst help i have put it in my configuration.yaml of zigbee2mqtt

Koenkk, 2021-02-02:
This option is not allowed for the addon yet, please create an option here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

closed 2021-02-02 by Koenkk