Koenkk/zigbee2mqtt #6041: Lonsonho QS-Zigbee-D02-TRIAC-2C-LN - Failed to configure

melojms, 2021-02-01:
## What happened

Updated to the latest version 1.17.1 and I'm using the Hassio Zigbee2Mqtt Addon. I am using the CC2531 USB stick as the coordinator.
I am turning on and off the dimmer [Lonsonho QS-Zigbee-D02-TRIAC-2C-LN](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-2C-LN.html#lonsonho-qs-zigbee-d02-triac-2c-ln) using homeassistant, and sometimes I get this error: 

https://pastebin.com/raw/a85bmd82

But both lights turn on and off normally. Just the error comes up.
The device name is _dimmer_sala_

## What did you expect to happen
Expected to not throw this error. Previous version 1.17.0.1 didn't trow the error.

## How to reproduce it (minimal and precise)
Install latest version 1.17.1
## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127

jlgobert, 2021-02-02:
seems that I've the same problem without Hassio Zigbee2Mqtt add on.

https://github.com/Koenkk/zigbee2mqtt/issues/6050

Koenkk, 2021-02-02:
Does the device provide state feedback when changed through the wired switch?

melojms, 2021-02-02:
Yes it does. This is the log on pressing the wired switch:

```
Debug Received Zigbee message from 'dimmer_sala', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/dimmer_sala', payload '{"brightness_l1":17,"brightness_l2":17,"linkquality":47,"state_l1":"ON","state_l2":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/dimmer_sala/l1', payload '{"brightness":17,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/dimmer_sala/l2', payload '{"brightness":17,"state":"OFF"}'
```

Koenkk, 2021-02-02:
Should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

melojms, 2021-02-02:
Thanks! Will check that.

melojms, 2021-02-03:
How can I use the dev branch addon on Hassos with the same configurations?

jlgobert, 2021-02-03:
ok for configuration. No failure anymore

But still automatic published mqtt message that change the set attributes !

For example, I set brightness 248 to l1, and 1 second after another mqtt message is published setting back the brightness to 8 !!

Debug Received MQTT message on 'zigbee2mqtt/0x847127fffe11eab0/l1/set' with data '{"brightness":248}'
Debug Publishing 'set' 'brightness' to '0x847127fffe11eab0'
Debug Received Zigbee message from '0x000d6f00162c30fe', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x847127fffe11eab0', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":8}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x847127fffe11eab0', payload '{"brightness_l1":8,"brightness_l2":8,"linkquality":42,"state_l1":"ON","state_l2":"OFF"}'

Koenkk, 2021-02-03:
@melojms uninstall zigbee2mqtt and install zigbee2mqtt-edge. Happy to hear it's fixed now.

@jlgobert  Looking at your logs looks; the device publishes this message itself. We can't do anything about that from the z2m side, if it's incorrect and the actual brightness is not 8 but 248 it is likely a bug in the firmware of your device.

```
Debug Received Zigbee message from '0x847127fffe11eab0', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":8}' from endpoint 1 with groupID 0
```

closed 2021-02-03 by Koenkk