Koenkk/zigbee2mqtt #6042: 'left' and 'right' not allowed in friendly_name

nanobra1n, 2021-02-02:
## What happened
I have my 2 devices with `friendly_name` `LR/Left` und `LR/Right`(to organize them by room). Starting with 1.17.1 zigbee2mqtt would not start, throwing an error message that `left` and `right` are not allowed in `friendly_name`. Renaming them in `devices.yaml` to `LR/Lefty` & `LR/Righty` resolved this.   
## What did you expect to happen
Left & Right are not that unusual words to name devices and should be allowed 
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2652RB
Adapter firmware version: 20210120

sjorge, 2021-02-02:
https://github.com/Koenkk/zigbee2mqtt/blob/99274c72c26ba21347908af8238fa8dfdb4fb99d/lib/util/utils.js#L33-L44

It's because left and right are also endpoint names, and endpoint names can't be friendly names.

nanobra1n, 2021-02-02:
Fair enough. May I suggest to mention these reserved names in the documentation somewhere or is it felt that the error message is sufficient?

timoline, 2021-02-04:
ther are more:

- friendly_name is not allowed to end with: '/${endpointName
- friendly_name shoud atleast be 1 char long
- friendly_name is not allowed to end or start with /
- friendly_name is not allowed to contain null char
- Following friendly_name are not allowed: '${endpointNames}
- Friendly name cannot end with a "/DIGIT
- MQTT wildcard (+ and #) not allowed in friendly_name

https://github.com/Koenkk/zigbee2mqtt/blob/99274c72c26ba21347908af8238fa8dfdb4fb99d/lib/util/utils.js#L191

closed 2021-03-14 by github-actions[bot]