Koenkk/zigbee2mqtt #6043: TuYa TS0601_cover not working after update

MariusLaurinaitis, 2021-02-02:
Until the update, everything worked, now the mixed sides. The status shows closed, although it is open and cannot be managed.

Koenkk, 2021-02-02:
Can you provide the data/database.db entry of your device?

MariusLaurinaitis, 2021-02-02:
Hello, Everything works in settings, I can close and open. 
[image]
But menu items are swapped. And clicking from the menu doesn't work. Also Alexa changed when I say close, then open, and vice versa.
[image]

MariusLaurinaitis, 2021-02-02:
[image]

MariusLaurinaitis, 2021-02-02:
{"id":3,"type":"Router","ieeeAddr":"0xbc33acfffe453dba","nwkAddr":56754,"manufId":4098,"manufName":"_TZE200_cowvfni3","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_cowvfni3","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1612244304324}

juslex, 2021-02-02:
Same problem here

MariusLaurinaitis, 2021-02-02:
Hello:
_When you set position 100, does the cover open or close?_
When I set position 100 its Close curtains 100%, then position 0 its open curtains.

_Can you also share your configuration.yaml?_

MQTT discovery data:
Topic: homeassistant/cover/0xbc33acfffe453dba/cover/config
Payload
availability:
  - topic: zigbee2mqtt/bridge/state
command_topic: zigbee2mqtt/0xbc33acfffe453dba/set
device:
  identifiers:
    - zigbee2mqtt_0xbc33acfffe453dba
  manufacturer: TuYa
  model: Curtain motor/roller blind motor/window pusher/tubular motor (TS0601_cover)
  name: '0xbc33acfffe453dba'
  sw_version: Zigbee2MQTT 1.17.1
json_attributes_topic: zigbee2mqtt/0xbc33acfffe453dba
name: '0xbc33acfffe453dba'
position_topic: zigbee2mqtt/0xbc33acfffe453dba
set_position_template: '{ "position": {{ position }} }'
set_position_topic: zigbee2mqtt/0xbc33acfffe453dba/set
unique_id: 0xbc33acfffe453dba_cover_zigbee2mqtt
value_template: '{{ value_json.position }}'
platform: mqtt

ASNNetworks, 2021-02-02:
Using the remote you can change the direction of the curtain, which will also invert in HA. Doing this I now have the positions correct. It says 100 (open) when my curtains are indeed open and 0 (closed) when my actual curtains are closed as well.. However, the service call to open/close are somehow reversed. This is especially strange since HA sees it's at 100, but sending a close command actually opens it and vice versa. 

I received the manual from the manufacturer for anyone else needing it. 
[Manual Zemismart Curtain.pdf](https://github.com/Koenkk/zigbee2mqtt/files/5914107/Manual.Zemismart.Curtain.pdf)

I tried to use `reverse_direction: true` and `reverse_direction: false` in devices.yaml (and restarted the add-on) but it doesn't make a dfference. So basically the command to open/close needs to be switched.

MariusLaurinaitis, 2021-02-02:
OK, i did it (Using the remote you can change the direction of the curtain).  It says 100 (open) and 0 (closed).The command to open/close needs to be switched in HA. Thanks, will whait:)

Koenkk, 2021-02-02:
Involving @DomiStyle here, seems cover position invert shouldn't be applied for `_TZE200_cowvfni3`, do you remember who has this device?

ASNNetworks, 2021-02-02:
> Involving @DomiStyle here, seems cover position invert shouldn't be applied for `_TZE200_cowvfni3`, do you remember who has this device?

I have this device, you added this manufacturer per my request previously (see the ticket above).

Koenkk, 2021-02-02:
@ASNNetworks 

> Using the remote you can change the direction of the curtain

That means the cover shouldn't be inverted by default right? (and you should't invert it with the remote)

ASNNetworks, 2021-02-02:
> @ASNNetworks
> 
> > Using the remote you can change the direction of the curtain
> 
> That means the cover shouldn't be inverted by default right? (and you should't invert it with the remote)

When I got this curtain and you added the manufacturer it was inverted through zigbee2mqtt. The open/close command and the position were both inverted. You inverted both then for me, which worked for me untill today for some reason. So I don’t know what has changed.

Now I used the remove to invert the positions, which is reflected correctly. But the open/close command is still inverted.

So I think the best would be to leave everything as default and use the remote to alter the position?

DomiStyle, 2021-02-02:
@Koenkk Nope, I took the inverted models from the previous TS0601_roller_blind [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2069/files#diff-5520dd49755e0a353d3857b8963a91cb4b8d87a01b6b8eb7bb6cce1c180eac83L1445).

I'm not sure why it stopped working though. The behaviour should be the same, it had coverInverted set by default previously too, see [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2069/files#diff-5520dd49755e0a353d3857b8963a91cb4b8d87a01b6b8eb7bb6cce1c180eac83L1451).

MariusLaurinaitis, 2021-02-02:
Until the last update it worked ok and today it stopped working properly. I have 2 _TZE200_cowvfni3 motors at my home.

Koenkk, 2021-02-03:
Removed the inversion for _TZE200_cowvfni3 .  Please confirm if it works correctly now @MariusLaurinaitis 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

MariusLaurinaitis, 2021-02-04:
> Removed the inversion for _TZE200_cowvfni3 . Please confirm if it works correctly now @MariusLaurinaitis
> 
> Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Its working  ! 
Now its OK
Thanks

Marius

Koenkk, 2021-02-04:
Great!

closed 2021-02-04 by Koenkk

juslex, 2021-02-16:
Dear @Koenkk,

hello, I have two zm25tq covers - TS0601 zemismart white label.  I note, however, that the controllers are different.  while using them in zzh each one appeared with a different model.  the curtain with a problem in inversion in HA, as reported in this issue, appeared as ZSTY-SM-1SRZG-EU and the other that works perfectly like the zm25q itself.  after going back to conbee 2 and using 1.17.1 dev both started to appear as TuYa TS0601_cover.  even after updating the z2m after fixing this issue, the inversion problem continues at Home Assistant in one of the covers. I activated invert_cove or tried to invert the direction of the engine through publish and the problem continues in HA.

DomiStyle, 2021-02-16:
@juslex Can you confirm that your exact model (manufacturerName) is `_TZE200_cowvfni3`? You can check in your database.db file.

Also, what exactly is reversed? Position, close/open commands or both?

ASNNetworks, 2021-02-16:
Please don't change anything else for `_TZE200_cowvfni3` lol. That one is working perfectly now and I'd hate for it to be reversed yet again lol.

MariusLaurinaitis, 2021-02-16:
Is everything fixed in version 1.17.1-4? Because I installed it and it doesn't work for me again. In the edge version, meanwhile, everything works.

juslex, 2021-02-16:
> @juslex Can you confirm that your exact model (manufacturerName) is `_TZE200_cowvfni3`? You can check in your database.db file.
> 
> Also, what exactly is reversed? Position, close/open commands or both?

The defective cover:

{"id":19,"type":"Router","ieeeAddr":"0xbc33acfffe457d09","nwkAddr":43521,"manufId":4098,"manufName":"_TZE200_wmcdj3aq","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_wmcdj3aq","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613475742350}

the other cover that works correctly:

{"id":18,"type":"Router","ieeeAddr":"0x5c0272fffebe40cb","nwkAddr":42056,"manufId":4098,"manufName":"_TZE200_fzo2pocs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0006\u0000\u0000\u0000\u0000\u0006","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_fzo2pocs","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613475771374}

juslex, 2021-02-16:
> @juslex Can you confirm that your exact model (manufacturerName) is `_TZE200_cowvfni3`? You can check in your database.db file.
> 
> Also, what exactly is reversed? Position, close/open commands or both?

It’s the _TZE200_wmcdj3aq

DomiStyle, 2021-02-17:
@juslex What exactly is reversed for `_TZE200_wmcdj3aq`? Position, close/open commands or both?

ASNNetworks, 2021-02-17:
Maybe you can try reverting position with the remote first? And then see it HA works accordingly with that reverted position.

I explained here: 
https://github.com/Koenkk/zigbee2mqtt/issues/6043#issuecomment-771935721

Just to prevent changes breaking stuff for other users of that model.

DavidWicks, 2021-02-17:
I am also having problems with this.

The up/down/stop buttons are working in HA, but the position slider does not activate the movement of the motor.  It changes the state of `running:` to true, but no motor movement.  

The `position:` value also does not change when the motor is activated with the up/down buttons, however the motor rotates.

The `running:` state changes to true briefly and quickly back to false when the up/down buttons are pressed and the motor continues to run.

This is a new motor ZM25TQ, I have no remote and am not sure if I need one for initial setup. 

cc2531, Zigbee2mqtt latest-dev, Docker, Synology.

[Zigbee2MQTT Roller Blind Database Entry.txt](https://github.com/Koenkk/zigbee2mqtt/files/5995769/Zigbee2MQTT.Roller.Blind.Database.Entry.txt)
[image]
## System Health

version | 2021.1.5
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 3.10.105
arch | x86_64
timezone | Australia/Perth

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4784
Installed Version | 1.10.1
Stage | running
Available Repositories | 747
Installed Repositories | 14

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
mode | storage
views | 12
resources | 18

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

DomiStyle, 2021-02-17:
@DavidWicks Did you run any calibration yet? Position won't be reported without calibration.

DavidWicks, 2021-02-17:
@DomiStyle Thanks for the fast response. No I did not, can I run calibration without the remote ?  Any docs you can point me too ?

MariusLaurinaitis, 2021-02-17:
[Manual.Zemismart.Curtain (1).pdf](https://github.com/Koenkk/zigbee2mqtt/files/5995927/Manual.Zemismart.Curtain.1.pdf)
Manual

juslex, 2021-02-18:
> @juslex What exactly is reversed for `_TZE200_wmcdj3aq`? Position, close/open commands or both?

Both is reversed. 

I already tried invert_cover on configuration.yaml and reverse the engine publish in mqtt. After the update as I mentioned, the remote control was inverted, I had to revert it to operate with it normally, but the situation remains.

juslex, 2021-02-18:
> I am also having problems with this.
> 
> The up/down/stop buttons are working in HA, but the position slider does not activate the movement of the motor. It changes the state of `running:` to true, but no motor movement.
> 
> The `position:` value also does not change when the motor is activated with the up/down buttons, however the motor rotates.
> 
> The `running:` state changes to true briefly and quickly back to false when the up/down buttons are pressed and the motor continues to run.
> 
> This is a new motor ZM25TQ, I have no remote and am not sure if I need one for initial setup.
> 
> cc2531, Zigbee2mqtt latest-dev, Docker, Synology.
> 
> [Zigbee2MQTT Roller Blind Database Entry.txt](https://github.com/Koenkk/zigbee2mqtt/files/5995769/Zigbee2MQTT.Roller.Blind.Database.Entry.txt)
> [image]
> 
> ## System Health
> version	2021.1.5
> installation_type	Home Assistant Container
> dev	false
> hassio	false
> docker	true
> virtualenv	false
> python_version	3.8.7
> os_name	Linux
> os_version	3.10.105
> arch	x86_64
> timezone	Australia/Perth
> Home Assistant Community Store
> GitHub API	ok
> Github API Calls Remaining	4784
> Installed Version	1.10.1
> Stage	running
> Available Repositories	747
> Installed Repositories	14
> Home Assistant Cloud
> logged_in	false
> can_reach_cert_server	ok
> can_reach_cloud_auth	ok
> can_reach_cloud	ok
> Lovelace
> dashboards	3
> mode	storage
> views	12
> resources	18
> Spotify

I have one of this motor (_TZE200_fzo2pocs), for me after the calibration it's work very well in Home Assistant. My problem is with the _TZE200_wmcdj3aq.

juslex, 2021-02-20:
After changing reverse_direction to false and remove the invert_cover from configuration.yaml, the curtain is now inverted, when is closed the expose on z2m and home assistant is open with value 100.

if I invert on the remote control, the remote control is inverted, but the values become correct, that is: closed with a value of 0 and really closed, open with a value of 100 and really open. But there is the drawback that the control is reversed. Is there any way to resolve this in _TZE200_wmcdj3aq?

DomiStyle, 2021-02-22:
@DavidWicks Were you able to get your motor `_TZE200_fzo2pocs` running properly after calibration?

@juslex Can you reset everything to default and describe what exactly the current behaviour is for `_TZE200_wmcdj3aq`?

What does open do? What does close do? What does stop do? What does setting positon to 100 do? What does setting position to 0 do?

juslex, 2021-02-26:
> @DavidWicks Were you able to get your motor `_TZE200_fzo2pocs` running properly after calibration?
>
> @juslex Can you reset everything to default and describe what exactly the current behaviour is for [...]

Hello, sorry for the late reply.

I restarted the cover, deleted and re-added it to the z2m.

It closed instead of appearing 0, in the z2m and in the home assistant it appeared 100. And open instead of appearing 100, it appeared 0. Therefore, when it said to close, it did not close and vice versa.

I solved it in a single way: invert the positions on the control, pressing the set button 4 times behind the control until the number 4 appears and I pressed the up arrow.

After that, both in the Home Assistant and z2m the states appear correctly. Open-Open (100), Closed-Closed (0). And when I press Stop the cover to stop. However, my control is reversed.

dinhchinh82, 2021-03-08:
@DomiStyle @Koenkk  today I just updated new version of z2m 1.17.1 and my curtains did not works anymore.
The status of the curtain is not correct as some guys her. If I set the invert cover by the remote, it is not able to control the curtain.

As far as I understand, now there is a way to fix this is changing the following information in the lib/tuya.js file:

> const coverPositionInvert = ['_TZE200_wmcdj3aq'];

But this way is not so convenient for users who are using the z2m addon. It requires the time for updating from dev branch. Is a way to fix this like more simple like using devices.js or similar file?

Thank you.

DomiStyle, 2021-03-09:
@dinhchinh82 What model do you have?

dinhchinh82, 2021-03-09:
> @dinhchinh82 What model do you have?
My model is _TZE200_4vobcgd3
I've tried to set Motor direction but the position is still not correct. The closed and open state is not correct.

>Debug Publishing 'set' 'state' to '0x5c0272fffecad3d1'
Debug TuYa_cover_control: Using state enums for _TZE200_4vobcgd3: {"close":2,"open":0,"stop":1}
Debug Received Zigbee message from '0x5c0272fffecad3d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x5c0272fffecad3d1', payload '{"linkquality":42,"position":1,"running":true,"state":"OPEN"}'
Debug Received Zigbee message from '0x5c0272fffecad3d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0
Info

dinhchinh82, 2021-03-09:
@DomiStyle I guess many models got thiss issue, not only _TZE200_4vobcgd3 model. 
That's reason I expect the flexible solution that can change this configuration more easily than updating in the lib/tuya.js file.

DomiStyle, 2021-03-10:
@dinhchinh82 The list of models with reversed position is fairly small so far.

You can already manually reverse the position by using `invert_cover` as seen [here](https://www.zigbee2mqtt.io/devices/TS0601_cover.html#device-type-specific-configuration).

dinhchinh82, 2021-03-10:
> @dinhchinh82 The list of models with reversed position is fairly small so far.
> 
> You can already manually reverse the position by using `invert_cover` as seen [here](https://www.zigbee2mqtt.io/devices/TS0601_cover.html#device-type-specific-configuration).

I understand this way but there are several cover/blink in my house. If I set the invert_cover = true in the configuration file then the _TZE200_4vobcgd3 curtain works well (correct state) but the other roller blind got this state issue.

If the invert_cover can be set for each device, that'd be better.

DomiStyle, 2021-03-10:
@dinhchinh82 Do you have multiple `_TZE200_4vobcgd3` with different behaviour or are your other covers different models?

Device specific configurarion is set per device, not per model. If your other cover behaves differently once you set your curtain to reversed something else is going on.

dinhchinh82, 2021-03-11:
@DomiStyle 
I have several devices with model _TZE200_4vobcgd3. Of course, these devices works as same behavior. 
I also have other blind motor with model _TZE200_wmcdj3aq. This one works well out of box in 1.18.1 without setting invert_cover.

When I set the invert_cover = true  in the configuration file, the curtains _TZE200_4vobcgd3 works well but _TZE200_wmcdj3aq got this state issue again. So I guess there are many similar cases like this.

I don't know how to set the invert_cover for each device. Could you give me more detail guide?
here is my configuration in the z2m addon:

>device_options_string: '{"invert_cover": true}'

vanhao04071994, 2021-07-23:
Can I have the cover code please?

OliverTche, 2021-12-30:
Hello, @juslex and friends!

@juslex, How did you calibrate your motor? If another friend would know how to do it, I apreciate too much the colaboration (It`s a nightmare! rsrsrs)
 

I have the same motor paired in the Home Assistent by Zigbee 2Mqqt and also without remote control. I didn`t find where to do it in the Home Assistant.

P.S.: I have another RF remote control from MOES HOUSE that I use to curtain module. If it could help, I can use it!

Follow some images below. Wether further information be necessary, just let me know...

Thanks everyone in advance!

[image]

[image]

[image]

[image]

{"id":6,"type":"Router","ieeeAddr":"0x9035eafffe672520","nwkAddr":32838,"manufId":4098,"manufName":"_TZE200_fzo2pocs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\f\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\f\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0012","65506":21,"appVersion":82,"modelId":"TS0601","manufacturerName":"_TZE200_fzo2pocs","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640888562064,"useImplicitCheckin":true}

> I have one of this motor (_TZE200_fzo2pocs), for me after the calibration it's work very well in Home Assistant. My problem is with the _TZE200_wmcdj3aq.

almooj, 2022-05-11:
hi all, unfortunately since this change /fix , my blind are all mixed up and are not able to work properly stated that they re down instead of up etc.. 
i have 3 blinds cover 
 
1 X detected as _TYST11_wmcdj3aq mcdj3aq (this one works fine)
2X detected as _TZE200_fzo2pocs TS0601(both are reversed since recents updates)

so as i can understand in recents changes in zhaquirks/tuya/__init__.py , those 2 _TZE200_fzo2pocs TS0601 are now in the invert by default list wich makes no sense for me and probablly a lot of people.. 

how can it be solved ? 
for me , ive first set all up with the remote so if some need to be reversed , ive managed to set it up correctly (for exemple when the wire is on the right side , you have to set it up as reversed ) and then assume everything is in the correct direction to set it up in HA ...

may be this is the issue instead of trying to alter the zha default configuration

im also gonna open an issue cause this is not fixed at all for me :(