Koenkk/zigbee2mqtt #6044: Woox R7049 Smoke Alarm

RealBitfox, 2021-02-02:
Hello, I need help with the indecration of Woox R7049 smoke detectors in ZigbeeToMqTT. The device has been paired - the status is "Unsupported". What can I do? Thanks for help!!!

nickrout, 2021-02-09:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

cchristofr, 2021-02-10:
Hi @RealBitfox 
I don't have this device for the moment, but I'm interested to know if you have successfully integrated it with the tutorial given by @nickrout 
Regards
Christophe

RealBitfox, 2021-02-13:
Hello Christophe,
I already knew the tutorial by @nickrout. Unfortunately, it was not helpful for me. I have not yet succeeded in integrating the detectors. I have not received any further help from the community.

Many greetings!

cchristofr, 2021-02-13:
> Hello Christophe,
> I already knew the tutorial by @nickrout. Unfortunately, it was not helpful for me. I have not yet succeeded in integrating the detectors. I have not received any further help from the community.
> 
> Many greetings!

Hello @RealBitfox
from my point of view, if the requirement is to integrate a zigbee device in HA, then a gateway like ZiGate (or PiZiGate) is sufficient. And with HA you can use the data from the zigbee device to pilot other devices (zigbee or not). 
But, if you want to exchange information between devices then a broker like mqtt can be useful.

nickrout, 2021-02-13:
> I already knew the tutorial by @nickrout. Unfortunately, it was not helpful for me. I have not yet succeeded in integrating the detectors. I have not received any further help from the community.

How did you go with the tutorial? You have provided nothing about your experience of getting it working.

RealBitfox, 2021-02-16:
Hello and good day!
Yes, you are right. I have revealed too little information. I apologise for this!  First of all, the general conditions: My English is not the best, I can speak German much better. ;) That makes it quite difficult. 
The instructions say: 1. pairing the device with Zigbee2MQTT. This works, the device is registered and listed as "not supported". Then it says: 2. adding your device #

The next step is to add an entry of your device to node_modules/zigbee-herdsman-converters/devices.js

That's where it ends for me. Where can I find this file?

Many greetings!

cchristofr, 2021-02-16:
> Hello Christophe,
> I already knew the tutorial by @nickrout. Unfortunately, it was not helpful for me. I have not yet succeeded in integrating the detectors. I have not received any further help from the community.
> 
> Many greetings!

Hello @RealBitfox
Have you had a look to integrate "R7049 Smoke Alarm" in HA with a gateway like ZiGate ? Maybe the integration and use of data  in HA would be easier. 
It seems that ZigbeeToMqTT supports more devices than ZiGate (or ConBee II) so if you (and me) want to use other zigbee devices , then ZigbeeToMqTT would be a better approach.

For the moment I have only z-wave devices and I hesitate to change for zigbee (cheaper).

nickrout, 2021-02-16:
Really the instructions are very plain. How did you install z2m?

RealBitfox, 2021-02-17:
Hey Christophe,
I have to explain something: I don't want the smoke detectors to be integrated at my place, it's for a neighbour who I want to get interested in Home Assistant. I myself use a Conbee and I have many Zigbee devices running. Also Wlan devices with Tasmota and some ESP8266 and ESP32 as sensors.  I have built a smoke detector with Zigbee. Works very well and was ready in 30 minutes!
The neighbour's Woox smoke detectors are to be integrated via zigbeetomqtt, as other devices are to be registered...

RealBitfox, 2021-02-17:
@nickrout 
Hi!
Yes, Zigbee2mqtt,Current version: 1.17.1-4 is installed and working. At the moment, some devices are also registered. That all works. A Woox smoke detector is registered, but the manufacturer is shown as unsupported.

Thanks for your help!

nickrout, 2021-02-17:
Add-on, docker or none of the above?

RealBitfox, 2021-02-18:
Hassos auf Raspberry 3B.....

nickrout, 2021-02-18:
Then I think the instructions are here

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/master/zigbee2mqtt/DOCS.md#adding-support-for-new-devices

Miswu13, 2021-03-20:
I too have the same Woox smoke detector, and the same problem. I followed the instructions from link, but can't find node_modules/zigbee-herdsman-converters/devices.js in my zigbee2mqtt folder on Home Assistant. Is there a separate program that i have to install? after setting the log level to debug, it says: "Zigbee2MQTT:warn  2021-03-20 18:57:05: Device '0xbc33acfffe3317e9' with Zigbee model 'TS0601' and manufacturer name '_TZE200_aycxwiau' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:warn  2021-03-20 18:57:46: Device '0xbc33acfffe3317e9' left the network"

nickrout, 2021-03-20:
How did you install z2m?

Miswu13, 2021-03-20:
I added a new repository in Home assistant addon store, and after that i installed z2m. Why? Should i add, another repository, What repository?

nickrout, 2021-03-20:
There is now an ability to add external convertors via the gui. Click settings and follow your nose.

Miswu13, 2021-03-21:
The external convertor, it doesn't work. I have put the code between the [ ], and after restart z2m, the gui gives me error, and doesn't load - tried multiple times. To repair it I have to manually modify the configuration.yaml, and delete the code i inserted. It is a waste of time and effort - i will send back the sensor, and buy one that it is already supported. Thanks for trying to help

Miswu13, 2021-03-21:
Does anyone know, where can i find a list of smoke detectors supported by z2m?

nickrout, 2021-03-21:
I am sure the z2m authors regret you wasting your time. Have you asked for help with your external converters file? I think not.

The zigbee2mqtt project has a list of supported devices. Look there.

Miswu13, 2021-03-21:
I get your sarcasm, but it's ok. I was just saying that for me it is faster to buy another sensor, than it is to invest time to find out what is wrong with this one, especially when i haven't found a suitable file with necessary steps. I thought that it would be easy to create a file and insert some code in it, based on a example... Don't get me wrong, it is a good addon, i love it, and i will continue using it, but with the devices that it supports. Thank you again for your time and answer

nickrout, 2021-03-21:
There are examples of external converters here: https://github.com/Koenkk/zigbee2mqtt.io/tree/master/docs/externalConvertersExample

And I must say yours looks nothing like that (I think, I find js pretty tough to read!)

Miswu13, 2021-03-21:
Thank you for the link. I have succeded to make the the device supported in z2m, but at the Exposes tab, all are N/A, except e.warning(), that let's me select the sound efect - i don't know if i need that. The e.smoke() witch i think it is the most important, it is N/A.

nickrout, 2021-03-21:
What happens if you [s]light a fire[/s] expose it to some smoke?

cchristofr, 2021-03-21:
> Does anyone know, where can i find a list of smoke detectors supported by z2m?

hello
I use this one: https://www.zigbee2mqtt.io/devices/HS3SA.html#heiman-hs3sa

Miswu13, 2021-03-22:
Thank you @cchristofr, i will chek it out. I'll search for a zigbee model since i have already the conbee II stick ! 
@nickrout , I triggered the device with smoke, the alarm from the device goes on, but z2m sensor, doesn't trigger any of entities from Exposes tab. Even the battery entity, is N/A. 
The log from adon says something like this for every entity from Exposes: 
`Zigbee2MQTT:debug 2021-03-22 17:07:41: Received MQTT message on 'homeassistant/sensor/0xbc33acfffe3317e9/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe3317e9"],"manufacturer":"CR Smart Home","model":"Smart Smoke Alarm (TS0601)","name":"0xbc33acfffe3317e9","sw_version":"Zigbee2MQTT 1.18.1"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0xbc33acfffe3317e9","name":"0xbc33acfffe3317e9 battery","state_topic":"zigbee2mqtt/0xbc33acfffe3317e9","unique_id":"0xbc33acfffe3317e9_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'`

cchristofr, 2021-03-22:
heiman-hs3sa works well with zigbee2mqtt (I use a zigate as coordinator). 
But it gives only info about "smoke". 

The zwave Fibaro FGSD002, gives many more informations like temperature, delta of temperature and your can set different kind of alarm. Of course, it is more expensive. It works with zwave2mqtt.

closed 2021-04-30 by github-actions[bot]

mattleh, 2021-05-12:
i just bought some Woox R7049
i got following data on the dump:

mattleh, 2021-05-12:
i recently bougth some of these woox R7049.

i used:
modelID: 'TS0601',
manufacturerName: '_TZE200_aycxwiau'
as fingerprint

i got following dump data (removed some duplicate dp) :
1620817683836 0xbc33acfffedb71f2 00 3d 01 04 00 00
1620817857303 0xbc33acfffedb71f2 00 3e 01 04 00 01
1620818043114 0xbc33acfffedb71f2 00 3f 08 01 00 01
1620820225750 0xbc33acfffedb71f2 00 02 0b 05 00 00
1620820225965 0xbc33acfffedb71f2 00 03 0e 04 00 02
1620822863992 0xbc33acfffedb71f2 00 03 0e 04 00 01
1620823078037 0xbc33acfffedb71f2 00 03 0e 04 00 00

with the python script i got:
2021-05-12T13:08:03 0xbc33acfffedb71f2 (tid:  61, s: 0) [unknown     ] => 0 0
2021-05-12T13:10:57 0xbc33acfffedb71f2 (tid:  62, s: 0) [unknown     ] => 0 1
2021-05-12T13:14:03 0xbc33acfffedb71f2 (tid:  63, s: 0) [wndow detect] => 0 True
2021-05-12T13:50:25 0xbc33acfffedb71f2 (tid:   2, s: 0) [unknown     ] => 0 00000000
2021-05-12T13:50:25 0xbc33acfffedb71f2 (tid:   3, s: 0) [unknown     ] => 0 2
2021-05-12T14:34:23 0xbc33acfffedb71f2 (tid:   3, s: 0) [unknown     ] => 0 1
2021-05-12T14:37:58 0xbc33acfffedb71f2 (tid:   3, s: 0) [unknown     ] => 0 0

the wndow detect is when the test button is pressed
the first unknown with value 0 1 (i think dp=1) means no smoke
the second unknown with value 0 0 (i think dp=1) is smoke detected
following messages come after removing inserting batteries
the unknown(dp=14?) with value is probably related to battery status it shows:
0 2 above 2,8v
0 1 above 2.6v
0 0 at 2.6v and below (beeps ocassionally)
The unknown with  0 00000000 (dp=11) hasnt changed value yet

can someone help me implement this device? is there any more data needed?