Koenkk/zigbee2mqtt #6047: Problem zemismart cover blind

JAJBrouwers, 2021-02-02:
Link of supported device:

https://www.zigbee2mqtt.io/devices/TS0601_cover.html

I still encounter some malfunction whit this device, de state is not reporting, is always UNKNOWN. I have installed de edge version, this have me also the motor speed setting, that doesn't work either.

I have replied on a other problem on GitHub about the same issue, but no response.... Because it's closed? I'm not sure if the post is still actief then.

DomiStyle, 2021-02-05:
@JAJBrouwers Can you activate debug logging according to the documentation and post a debug log where you try all operations (close, open, position 100%, position 0%)? Always let the motor finish before sending the next command.

https://www.zigbee2mqtt.io/information/debug.html

JAJBrouwers, 2021-02-05:
@DomiStyle 

I followed your link about the debug, I will try it this evening when I'm home, I'm not expert in this stuff.... But as far as I understand I add a line to my configuration.yaml and a line in de add on config.

DomiStyle, 2021-02-05:
Yes, that's correct.

Also, just for clarification, the state is UNKNOWN in the zigbee2mqtt web interface? Not in Google Home?

JAJBrouwers, 2021-02-05:
The state is unknown on:
HA
Zigbee2mqtt
And Google home

So it's everywhere unknown 😅

JAJBrouwers, 2021-02-05:
@DomiStyle 

im working on it, but im stuck on the config... I get a error that the component Advanced is not found

JAJBrouwers, 2021-02-05:
@DomiStyle 

when I set the blind to "75" I get this:

```
Zigbee2MQTT:debug 2021-02-05 20:51:31: Received MQTT message on 'zigbee2mqtt/rolgordijn voor/set' with data '{"position":75}'
Zigbee2MQTT:debug 2021-02-05 20:51:31: Publishing 'set' 'position' to 'rolgordijn voor'
Zigbee2MQTT:debug 2021-02-05 20:51:32: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,75],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":9}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:32: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:33: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":10}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:33: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:34: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,4],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":11}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:34: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:34: Received Zigbee message from 'motion sensor trap', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:34: MQTT publish: topic 'zigbee2mqtt/motion sensor trap', payload '{"battery":100,"illuminance":6955,"illuminance_lux":5,"linkquality":190,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":17.66,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2021-02-05 20:51:35: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,6],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:35: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:36: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,6],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:36: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:36: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,9],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":13}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:36: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:37: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,11],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:37: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:38: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,11],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:38: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:38: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,13],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":13}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:38: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:39: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,16],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:39: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:40: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,16],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:51:40: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:51:40: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,18],"type":"Buffer"},"datatype":2,"dp":2 [... cut ...]
Zigbee2MQTT:info  2021-02-05 20:52:03: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
Zigbee2MQTT:debug 2021-02-05 20:52:04: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,76],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":20}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2021-02-05 20:52:04: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":190,"running":true}'
```

DomiStyle, 2021-02-06:
Can you post the full log file?

JAJBrouwers, 2021-02-06:
@DomiStyle 

How do you mean? The full log, this is what I get when I use the debug log

DomiStyle, 2021-02-06:
I mean the full log from zigbee2mqtt start to cover arriving at the end position. Can't you copy the full file from the addon? Alternatively, can you try finding out the exact version you are running?

You also need to enable Zigbee herdsman debug logging as shown here: https://www.zigbee2mqtt.io/information/debug.html#home-assistant-ossupervised-addon

JAJBrouwers, 2021-02-06:
About the ,

Advanced:
   Log_level: debug

When I want this in my configuration.yaml I get a error "integration not found"

So now I have added this in the config of the add-on, is that the same?

I'm not sure what you're after because in de add-on log you get more then only the debug

JAJBrouwers, 2021-02-06:
[image]

JAJBrouwers, 2021-02-06:
from open to full close:

```
Received MQTT message on 'zigbee2mqtt/rolgordijn voor/set' with data '{"position":100}'
Publishing 'set' 'position' to 'rolgordijn voor'
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":5}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,6],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":6}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,9],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":7}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,11],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":8}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,14],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":9}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,16],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":10}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,19],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":11}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,21],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":12}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":13}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,26],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":14}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,28],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":15}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,31],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":16}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,33],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":17}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,36],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":18}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,38],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":19}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,41],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":20}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,43],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":21}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,46],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":22}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,48],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":23}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,51],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":24}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,53],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":25}' from endpoint 1 with groupID null
[... cut ...] int 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,98],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":42}' from endpoint 1 with groupID null
Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":43}' from endpoint 1 with groupID null
```

DomiStyle, 2021-02-06:
Unfortunately I don't use the z2m addon so I'm not much use with configuring it.

Although I think I already see what the issue is I would still like to get the herdsman debug log.

https://www.zigbee2mqtt.io/information/debug.html#home-assistant-ossupervised-addon

`zigbee_herdsman_debug: true` needs to be added in addition to the regular `log_level: debug`.

JAJBrouwers, 2021-02-06:
oke, i have already done that.
but im still struggling with what you need, I don't see specific the herdsman log or something. I only see debug messages. I will try to google it

JAJBrouwers, 2021-02-06:
do you mean this:

```
2021-02-06T12:27:32.151Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 22
2021-02-06T12:27:32.152Z zigbee-herdsman:deconz:driver query aps data indication
2021-02-06T12:27:32.154Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x0 request id: 3 confirm status: 0
2021-02-06T12:27:32.155Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-02-06T12:27:32.156Z zigbee-herdsman:deconz:driver query aps data indication
2021-02-06T12:27:32.256Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 23
2021-02-06T12:27:32.259Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 23 srcAddr: 0x0 destAddr: 0x0 profile id: 0x0 cluster id: 0x8004
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:frameParser response payload: 3,0,0,0,20,1,4,1,5,0,1,3,0,0,10,0,25,0,3,1,0,32,0,0,5
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr.
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter RECEIVING SIMPLE_DESCRIPTOR - addr: 0x0 EP:1 inClusters: 0,10,25 outClusters: 1,32,1280
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter EP: 1
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter profile ID: 260
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter device ID: 5
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter input cluster: 0x0
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter input cluster: 0xa
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter input cluster: 0x19
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter output cluster: 0x1
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter output cluster: 0x20
2021-02-06T12:27:32.260Z zigbee-herdsman:deconz:adapter output cluster: 0x500
2021-02-06T12:27:32.358Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 24
2021-02-06T12:27:32.361Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr.: 24 status: 5
```

JAJBrouwers, 2021-02-07:
@DomiStyle  is what you're after?

DomiStyle, 2021-02-07:
Yes, that was it. Did the state ever work for you with z2m or was it broken with an update?

For some reason your motor behaves strangely and doesn't send a confirmation when it arrives at the target location, hence why you won't get any state or position and only `running: true`. But it sends continous events while moving, which I haven't seen before.

Can you try sending a close/open command instead of a position and operating the blind/curtain manually (if possible) and post the log again?

Also your position seems to still be reversed? Moving to position 100 should fully open the curtain, not close it as you described in your first log.

JAJBrouwers, 2021-02-07:
oke, i will try. then I have to send a mqtt message I think. sound not be a problem but its not in the add-on it self

JAJBrouwers, 2021-02-07:
I did a MQTT message CLOSE:

```
Zigbee2MQTT:info  2021-02-07 13:52:58: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":9,"running":true}'
2021-02-07T12:52:58.990Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xeaf EP:1 SeqNr. 182 request id: 132
2021-02-07T12:52:58.992Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 132 status: 0
2021-02-07T12:52:58.993Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:52:59.000Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-02-07T12:52:59.001Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:52:59.090Z zigbee-herdsman:deconz:driver query aps data confirm
2021-02-07T12:52:59.092Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 183
2021-02-07T12:52:59.094Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xeaf request id: 132 confirm status: 0
2021-02-07T12:52:59.095Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:52:59.095Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 112
2021-02-07T12:52:59.863Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-02-07T12:52:59.864Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:52:59.898Z zigbee-herdsman:deconz:driver query aps data indication
2021-02-07T12:52:59.899Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 184
2021-02-07T12:52:59.902Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 184 srcAddr: 0xeaf destAddr: 0x0 profile id: 0x104 cluster id: 0xef00
2021-02-07T12:52:59.903Z zigbee-herdsman:deconz:frameParser response payload: 9,113,1,0,24,2,2,0,4,0,0,0,44
2021-02-07T12:52:59.903Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":113,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":24,"dp":2,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,44]}}},"address":3759,"endpoint":1,"linkquality":9,"groupID":null}'
Zigbee2MQTT:debug 2021-02-07 13:52:59: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,44],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":24}' from endpoint 1 with groupID null
2021-02-07T12:52:59.904Z zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffec20a6a/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2021-02-07T12:52:59.905Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-02-07T12:52:59.905Z zigbee-herdsman:deconz:adapter { cmdId: 1, statusCode: 0 }
2021-02-07T12:52:59.905Z zigbee-herdsman:deconz:adapter no response expected
2021-02-07T12:52:59.906Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2021-02-07 13:52:59: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":9,"running":true}'
2021-02-07T12:53:00.000Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xeaf EP:1 SeqNr. 185 request id: 133
2021-02-07T12:53:00.004Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 133 status: 0
2021-02-07T12:53:00.005Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:53:00.010Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-02-07T12:53:00.011Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:53:00.100Z zigbee-herdsman:deconz:driver query aps data confirm
2021-02-07T12:53:00.102Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 186
2021-02-07T12:53:00.104Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xeaf request id: 133 confirm status: 0
2021-02-07T12:53:00.105Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:53:00.105Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 113
2021-02-07T12:53:00.504Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 187
2021-02-07T12:53:00.506Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-02-07T12:53:00.507Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

JAJBrouwers, 2021-02-07:
I did a open press on the motor itself:
```
Zigbee2MQTT:debug 2021-02-07 13:56:08: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,45],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":53}' from endpoint 1 with groupID null
2021-02-07T12:56:08.223Z zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffec20a6a/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2021-02-07T12:56:08.223Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-02-07T12:56:08.224Z zigbee-herdsman:deconz:adapter { cmdId: 1, statusCode: 0 }
2021-02-07T12:56:08.224Z zigbee-herdsman:deconz:adapter no response expected
2021-02-07T12:56:08.224Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2021-02-07 13:56:08: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":9,"running":true}'
2021-02-07T12:56:08.322Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xeaf EP:1 SeqNr. 114 request id: 185
2021-02-07T12:56:08.325Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 185 status: 0
2021-02-07T12:56:08.325Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:08.334Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-02-07T12:56:08.334Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:08.424Z zigbee-herdsman:deconz:driver query aps data confirm
2021-02-07T12:56:08.425Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 115
2021-02-07T12:56:08.427Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xeaf request id: 185 confirm status: 0
2021-02-07T12:56:08.427Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:08.427Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 31
2021-02-07T12:56:09.181Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-02-07T12:56:09.181Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:09.231Z zigbee-herdsman:deconz:driver query aps data indication
2021-02-07T12:56:09.232Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 116
2021-02-07T12:56:09.236Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 116 srcAddr: 0xeaf destAddr: 0x0 profile id: 0x104 cluster id: 0xef00
2021-02-07T12:56:09.236Z zigbee-herdsman:deconz:frameParser response payload: 9,32,1,0,54,2,2,0,4,0,0,0,44
2021-02-07T12:56:09.237Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":54,"dp":2,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,44]}}},"address":3759,"endpoint":1,"linkquality":5,"groupID":null}'
Zigbee2MQTT:debug 2021-02-07 13:56:09: Received Zigbee message from 'rolgordijn voor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,44],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":54}' from endpoint 1 with groupID null
2021-02-07T12:56:09.238Z zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffec20a6a/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2021-02-07T12:56:09.238Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-02-07T12:56:09.239Z zigbee-herdsman:deconz:adapter { cmdId: 1, statusCode: 0 }
2021-02-07T12:56:09.239Z zigbee-herdsman:deconz:adapter no response expected
2021-02-07T12:56:09.240Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2021-02-07 13:56:09: MQTT publish: topic 'zigbee2mqtt/rolgordijn voor', payload '{"linkquality":5,"running":true}'
2021-02-07T12:56:09.333Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xeaf EP:1 SeqNr. 117 request id: 186
2021-02-07T12:56:09.336Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 186 status: 0
2021-02-07T12:56:09.336Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:09.345Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-02-07T12:56:09.346Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:09.434Z zigbee-herdsman:deconz:driver query aps data confirm
2021-02-07T12:56:09.434Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 118
2021-02-07T12:56:09.438Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xeaf request id: 186 confirm status: 0
2021-02-07T12:56:09.439Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-02-07T12:56:09.439Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 32
```

DomiStyle, 2021-02-07:
I added a possible fix to my repository here: https://github.com/DomiStyle/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js

You need to replace your fromZigbe.js with the one linked above, but I'm not sure how this is done with the z2m addon.

The `running` and `state` attribute might be wrong but position should be reported now.

JAJBrouwers, 2021-02-08:
@DomiStyle 

Thank you so far for your time en effort!
On Google I found a topic about it, to replace fromzigbe.js on the add-on is not really possible.

I found this:
https://www.zigbee2mqtt.io/information/configuration.html#external-converters-configuration

The only thing is that it's only for one device.. I think? 

Maby @Koenkk knows something?

DomiStyle, 2021-03-11:
@JAJBrouwers https://github.com/Koenkk/zigbee-herdsman-converters/pull/2354 should fix your issue too, you can give it a try once it reaches edge/dev.

JAJBrouwers, 2021-03-11:
@DomiStyle , oke.. I will try because the js. Replacement is not working for me!

When does this merge?

DomiStyle, 2021-03-12:
@JAJBrouwers Should be merged by now.

JAJBrouwers, 2021-03-14:
@DomiStyle 

I have give it a try, it reports a state! Nice, thank you!
But it reports up posit, when cover is closed, it says its open en visa versa.
Open = position 100
Closed = position 0

the problem that I get is, when the state says its open, it is actually closed! so in HA I lost the button to close the cover! because the state says its closed....but the cover is open.

DomiStyle, 2021-03-16:
@JAJBrouwers What's your model/manufacturerName?

JAJBrouwers, 2021-03-16:
€ 56,77  60%OFF | Zemismart Tuya Zigbee Roller Schaduw Driver Smartthings Controle Ingebouwde Batterij Blind Motor Alexa Google Assistent Voice Control
https://a.aliexpress.com/_vtVcFD

I have this one.
Zemismart type M515EGB2TN

Rayhee, 2021-03-17:
@DomiStyle, i had the same issue with the position. take a look at the end of https://github.com/Koenkk/zigbee2mqtt/issues/6514
Maybe it can help. I will try your solution this weekend

closed 2021-04-24 by github-actions[bot]