Koenkk/zigbee2mqtt #6050: Lonsonho QS-Zigbee-D02-TRIAC-2C-LN fail to configure and automatic MQTT published after a command sent ! HELP (8 devices to be installed)

jlgobert, 2021-02-02:
## What happened
Fail to configure AND, even if it failed, everytime you send a command ON or OFF (or even directly with the manuel switch, )
another command is sent within a seconc and set brightness at 10. 

## What did you expect to happen
Interview should work !
And an OFF or ON command should not resend another message with a different brightness (always 10)

## How to reproduce it (minimal and precise)
On every modules I install.
## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II : CC2652RB Slaech
Adapter firmware version: CC2652RB_20210120.hex

**For the failure of configuration :**
Error Failed to configure '0x847127fffe11e8e6', attempt 2 (Error: ConfigureReporting 0x847127fffe11e8e6/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 41694 - 1 - 5 - 6 - 7 after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7))

**For the "changing messages"**
This is the message I send :
Info MQTT publish: topic 'zigbee2mqtt/0x847127fffe11e8e6', payload '{"brightness_l1":254,"brightness_l2":0,"linkquality":27,"state_l1":"ON","state_l2":"OFF"}'

and as soo as it is published : here is the second message automaticaly generated :
MQTT publish: topic 'zigbee2mqtt/0x847127fffe11e8e6', payload '{"brightness_l1":10,"brightness_l2":0,"linkquality":27,"state_l1":"ON","state_l2":"OFF"}'

jlgobert, 2021-02-03:
ok for configuration. No failure anymore

But still automatic published mqtt message that change the set attributes !

For example, I set brightness 248 to l1, and 1 second after another mqtt message is published setting back the brightness to 8 !!

Debug Received MQTT message on 'zigbee2mqtt/0x847127fffe11eab0/l1/set' with data '{"brightness":248}'
Debug Publishing 'set' 'brightness' to '0x847127fffe11eab0'
Debug Received Zigbee message from '0x000d6f00162c30fe', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x847127fffe11eab0', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":8}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x847127fffe11eab0', payload '{"brightness_l1":8,"brightness_l2":8,"linkquality":42,"state_l1":"ON","state_l2":"OFF"}'

jlgobert, 2021-02-03:
After searching for a while inthe past issues, i've found the same behaviour:

https://github.com/Koenkk/zigbee2mqtt/issues/2939
 Could you give me some feedback ?

PetersPlace, 2021-02-08:
Same here, today my Zigbee network stopped, after restart it fails on

Zigbee2MQTT:info  2021-02-08 20:18:52: Starting Zigbee2MQTT version 1.17.1 (commit #1.17.1)
Zigbee2MQTT:info  2021-02-08 20:18:52: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:error 2021-02-08 20:19:04: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-08 20:19:04: Failed to start zigbee
Zigbee2MQTT:error 2021-02-08 20:19:04: Exiting...
Zigbee2MQTT:error 2021-02-08 20:19:04: Error: AREQ - ZDO - simpleDescRsp after 10000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)

So what now,
I upgraded to 1.17.1   saturday and also set Herdman to true to add my Bltzwolf SS7 1 gang.

closed 2021-03-18 by github-actions[bot]