Koenkk/zigbee2mqtt #6051: Danalock v3 problem

Varming73, 2021-02-02:
After upgrading to the lastest version of Zigbee2mqtt (1.17.1) I can't control my Danalock V3-BTZB reliably any longer. I see this in the log when I push the State update button:
`Publish 'get' 'state' to 'Main Door Lock' failed: 'Error: Read 0x000b57fffe59d3d5/1 closuresDoorLock(["lockState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))'`

When trying to unlock:
`Publish 'set' 'state' to 'Main Door Lock' failed: 'Error: Command 0x000b57fffe59d3d5/1 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 7966 - 1 - 12 - 257 - 1 after 10000ms)'`

It works maybe 1 time out of 10-12 at the moment. Before upgrading it worked every time.

Varming73, 2021-02-03:
I've just read up on the issue but reconfiguring doesn't do any difference. It failed a couple of times but suddenly the reconfigure came through - but I still can't update the state for the lock.

Varming73, 2021-02-03:
**UPDATE**: I just tried something: I opened the Danalock app, Settings, Smart Home, pressed Z-wave (though it's a Zigbee version) and connect. After this Zigbee2Mqtt can retrieve the status from the lock again. Everything seems to work for now.

smulle48, 2021-02-03:
The correct way to setup the lock is:

- Reset the lock
- Setup in the App
- calibrate the lock
- include it in the zigbee network with the button (z-wave) in the app
- Remove at least 1 battery for 10-15 seconds
- Success :-)

Varming73, 2021-02-03:
Thanks @smulle48 but in this case the lock has been working for several days before Zigbee2mqtt suddenly couldn't retrieve the state any longer. After doing the above steps in the app **again** it suddenly worked. I wonder if the problem is a Danalock problem and not a Zigbee2mqtt problem.

smulle48, 2021-02-03:
> Thanks @smulle48 but in this case the lock has been working for several days before Zigbee2mqtt suddenly couldn't retrieve the state any longer. After doing the above steps in the app **again** it suddenly worked. I wonder if the problem is a Danalock problem and not a Zigbee2mqtt problem.

Danalock is quite "funny" when it comes to pairing.

If you for some reason pair it with zigbee first and then with the app afterwards, the Bluetooth in the lock, tend to kill the zigbee at some point. I have reported it to Danalock a few times, but they have no clue why

Varming73, 2021-02-03:
> Danalock is quite "funny" when it comes to pairing.
> 
> If you for some reason pair it with zigbee first and then with the app afterwards, the Bluetooth in the lock, tend to kill the zigbee at some point. I have reported it to Danalock a few times, but they have no clue why

I had to pair it with Bluetooth first as Zigbee wasn't enabled when I activated the batteries so I had to enable Zigbee from the app. So your experience with this doesn't seem to apply to my issue. If I can fix the issue by enabling Zigbee in the app again I don't see how this can be fixed in Zigbee2mqtt.

I've heard rumors that they're slowing on their way to close up shop and with things like the app looking very dated etc. I guess the rumors could be true. Maybe that's why they don't fix the issues reported to them.

Varming73, 2021-02-04:
The problem is back. I can't recreate the "fix" from yesterday so it might have been some coincidence or God knows what. I can't get state from the lock, lock or unlock it.

Varming73, 2021-02-10:
I found out that the little hole in the lock has a little switch. When you put a pin down and press once the lock goes into Zigbee inclusion mode. After doing this the lock can be lock/unlocked again - but the state doesnt update unless I press the "state" button and forces and update - then I get the correct state. I wonder if something has changed in Zigbee2Mqtt since that suddenly doesnt work any longer..?

Varming73, 2021-02-19:
After adding a repeater/router to my setup the locks has been working. Strange thing is I have other devices further away which has been working all the time but maybe the Danalock is more sensitive somehow.

closed 2021-02-19 by Varming73