Koenkk/zigbee2mqtt #6065: Lidl / Silvercrest smart motion sensor doesn't send any data after battery change

mrTuomoK, 2021-02-03:
Hi. The problem with this [motion sensor from Lidl](https://www.zigbee2mqtt.io/devices/HG06335.html#lidl-hg06335) is such that everything works just fine until I swap the battery (even quickly power it off). After battery change the only thing in logging I see is just that it announces itself but doesn't send any tamper/battery or occupancy events/data (nothing else on zigbee2mqtt log). Currently the only solution is to remove it from the network & repair to get it working again.

How should I start debugging this properly? I have tried with CC2531 and currently using Slaesh CC2652RB stick (and newest zigbee2mqtt 1.17.1). All other devices work just fine (lots of bulbs, sensors, switches and so on). I have a couple of these motion sensor and all are the same.

closed 2021-03-13 by github-actions[bot]

dxaudio, 2021-09-11:
I have 4 of these sensors. All behaves the same as mentioned. If battery needs replacement, i shall repair sensor otherwise it never sends data again.

dxaudio, 2021-09-11:
I see mrTuomoK from Finland, i bought my sensors in Finland too. Has anybody from other countries experience the same issue or we just in Finland were unlucky to get kind of broken sensor version..?