Koenkk/zigbee2mqtt #6066: friendly_name shoud atleast be 1 char long

mabusdogma, 2021-02-03:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

mabusdogma, 2021-02-03:
I have no friendly names on my configuration in the addon in home assistant... Not sure what to change

skipper79, 2021-02-03:
same issue here

mgrefsrud, 2021-02-03:
Check groups.yaml

skipper79, 2021-02-03:
> Check groups.yaml

This one is empty, no groups

TheLama1993, 2021-02-03:
Same issue here! Groups.yaml is empty. Friendly name is not used.

Koenkk, 2021-02-03:
The `friendly_name` is required. Can you share your `configuration.yaml`?

mgrefsrud, 2021-02-03:
I had a line in the groups.yaml file with friendly_name: ' '
I removed this and the addon started.

mabusdogma, 2021-02-03:
> 
> 
> The `friendly_name` is required. Can you share your `configuration.yaml`?

`data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user:
  password: 
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6754
  channel: 20
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
`

mabusdogma, 2021-02-03:
sorry for not using "insert code"... I think it looks more aligned like this

mabusdogma, 2021-02-03:
> 
> 
> I had a line in the groups.yaml file with friendly_name: ' '
> I removed this and the addon started.

Now it works for me, I did the same.

skipper79, 2021-02-03:
> The `friendly_name` is required. Can you share your `configuration.yaml`?

`
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

mobile_app:

homekit:
  filter:
    include_domains:
      - cover
      - sensor
    include_entities:
      - cover.rolgordijn_zijraam
      - cover.rolgordijn_achter_l_cover
      - cover.rolgordijn_achter_r_cover
      - light.ikea_testlamp_light
      - sensor.schakelaar_action
      - binary_sensor.achterdeur_senor_contact

light:
- platform: group
  name: Wandlampen boven
  entities:
    - light.verandawand_1_1
    - light.verandawand_2_1
    - light.verandawand_3_1
    
- platform: group
  name: Wandlampen onder
  entities:
    - light.verandawand_1_2
    - light.verandawand_2_2
    - light.verandawand_3_2
    
lightwave:
  host: 192.168.1.22
  switches:
    R1D1: 
      name: Verandalampjes
    R1D2:
      name: Kerstboom
    R1D4:
      name: Wandlampjes

switch:
  - platform: edimax
    host: 192.168.1.11
    username: *****
    password: '*****'
    name: Edimax

sensor:
  - platform: template
    sensors:
      edimax_watts:
        value_template: '{{ states.switch.edimax.attributes["current_power_mwh"] | float * 1000000 | round(2) }}'
        unit_of_measurement: 'W'
      edimax_current_power:
        friendly_name: huidig vermogen
        unit_of_measurement: 'W'
        value_template: "{{ state_attr('switch.edimax', 'current_power_w') | replace('Geen', 0) }}"
      edimax_total_power:
        friendly_name: totaal vermogen dag
        unit_of_measurement: 'W'
        value_template: "{{ state_attr('switch.edimax', 'today_energy_kwh') | replace('Geen', 0) }}"

cover:
- platform: group
  name: Rolgordijnen achterraam
  entities:
    - cover.rolgordijn_achter_l_cover
    - cover.rolgordijn_achter_r_cover`

skipper79, 2021-02-03:
Futher, only scnes.yaml has 'friendly_name's

`- id: '1601558963744'
  name: Sfeerverlichting Veranda
  entities:
    light.verandawand_1_1:
      min_mireds: 250
      max_mireds: 454
      friendly_name: Verandawand 1-1
      supported_features: 43
      state: 'off'
    light.verandawand_2_1:
      min_mireds: 250
      max_mireds: 454
      friendly_name: Verandawand 2-1
      supported_features: 43
      state: 'off'
    light.verandawand_3_1:
      min_mireds: 250
      max_mireds: 454
      friendly_name: Verandawand 3-1
      supported_features: 43
      state: 'off'
    automation.nieuwe_automatisering:
      last_triggered: '2020-10-25T18:36:41.083347+00:00'
      mode: single
      current: 0
      id: '1603047153413'
      friendly_name: verandalampen aan
      state: 'on'`

TheLama1993, 2021-02-03:
> > The `friendly_name` is required. Can you share your `configuration.yaml`?
> 
> I checked config/groups.yaml, it was empty.
> Then I checked the groups.yaml in the folder share/zigbee2mqtt and removed the following:
> 
> ```
> '1':
> friendly_name: ''
> ```
> 
> Then I restarted the the addon and got the following error:
> 
> ```
> Zigbee2MQTT:info  2021-02-03 20:28:37: Starting Zigbee2MQTT version 1.17.1 (commit #1.17.1)
> Zigbee2MQTT:info  2021-02-03 20:28:37: Starting zigbee-herdsman (0.13.59)
> Zigbee2MQTT:error 2021-02-03 20:28:57: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2021-02-03 20:28:57: Failed to start zigbee
> Zigbee2MQTT:error 2021-02-03 20:28:57: Exiting...
> Zigbee2MQTT:error 2021-02-03 20:28:57: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>   at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
>   at Generator.throw (<anonymous>)
>   at rejected (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
> npm
> ```
> 
> **Update**: I read in [issue 5842](https://github.com/Koenkk/zigbee2mqtt/issues/5854) to change the USB to another port. And now it works again! I'm happy!!
> Thanks @Koenkk and @mgrefsrud for your support and helping me out!

I did the same as above and the error indeed changed to the below message. After changing the zigate to another usb port, the problem remains...

Zigbee2MQTT:error 2021-02-03 21:53:12: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-03 21:53:12: Failed to start zigbee
Zigbee2MQTT:error 2021-02-03 21:53:12: Exiting...
Zigbee2MQTT:error 2021-02-03 21:53:12: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 
ERR!
 code ELIFECYCLE
npm ERR! errno 1
npm
 
ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-02-03T20_53_12_920Z-debug.log
2021-02-03T21:53:13: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]

Koenkk, 2021-02-04:
I found out that the Zigbee2MQTT api allowed to add groups with an empty name, once done this error will be thrown on startup. I've fixed the api to not allow this anymore.

closed 2021-02-04 by Koenkk