Koenkk/zigbee2mqtt #6068: Adding support for Develco Smart Relay DIN (SMRZB-332)

GD-Dal, 2021-02-03:
Hi.
By following guides and some examples from this forum, I'm trying to add support for the Develco SMRZB-332 smart relay.
This is a link to the technical manual:
https://www.develcoproducts.com/media/1793/smrzb-332-technical-manual-smart-relay-din-16a.pdf

I run HASSIO, so I had to download the devices.js file and edit that.

I have come up with this:

    {
    zigbeeModel: ['SMRZB-332'],
    model: 'SMRZB-332',
    vendor: 'Develco Products A/S',
    description: 'Smart Relay DIN',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        },
    exposes: [e.switch(), e.power()],
    },

And now the web UI says it's supported.
But I get errors:

```
error 2021-02-03 12:19:48: Publish 'get' 'state' to 'Relay - Varmtvannstank' failed: 'Error: Read 0x0015bc00140301d0/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
debug 2021-02-03 12:19:48: Error: Read 0x0015bc00140301d0/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)
    at DeconzAdapter.<anonymous> (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:606:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)
info  2021-02-03 12:19:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'state' to 'Relay - Varmtvannstank' failed: 'Error: Read 0x0015bc00140301d0/1 genOnOff([\"onOff\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response received)'","meta":{"friendly_name":"Relay - Varmtvannstank"},"type":"zigbee_publish_error"}'
```

And when i try to switch ON / OFF, I get these:

```
Publish 'set' 'state' to 'Relay - Varmtvannstank' failed: 'Error: Command 0x0015bc00140301d0/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Publish 'set' 'state' to 'Relay - Varmtvannstank' failed: 'Error: Command 0x0015bc00140301d0/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response `received)'`

```

What am I doing wrong?

Other questions:
Where can I find out exactly what this relay exposes?
What must be done to add this device permanently to future Zigbee2MQTT versions?

Thank you

GD-Dal, 2021-02-10:
Hi.
An update on this.
I'm getting readings from this device now, and it looks like this:

[image]
The energy value seems to be all power consumed since last reset?
The power value seems to be the power usage right now.

The code used looks like this:
`    {
    zigbeeModel: ['SMRZB-332'],
    model: 'SMRZB-332',
    vendor: 'Develco',
    description: 'Smart Relay DIN',
    supports: 'On/Off, Metering',
	fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
	exposes: [e.switch(), e.power(), e.energy()],
	},`

But the On/Off does not work. And this devices is supposed to support OTA.
But I have no clue how to implement those 2 things.

Any tips?

Thanks.

sjorge, 2021-02-10:
I think it might be using the wrong end point

`Smart relay DIN –End Point 0x02 ` seems to use endpoint 2, and you do configure the other reporting on that which seem to work. Try with:

```
{
zigbeeModel: ['SMRZB-332'],
model: 'SMRZB-332',
vendor: 'Develco Products A/S',
description: 'Smart Relay DIN',
fromZigbee: [fz.on_off, fz.metering],
toZigbee: [tz.on_off],
meta: {configureKey: 1},
endpoint: (device) => {
    return {'default': 2};
},
configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(2);
    await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
    await reporting.instantaneousDemand(endpoint);
    await reporting.readMeteringMultiplierDivisor(endpoint);
    },
exposes: [e.switch(), e.power()],
},
```

Koenkk, 2021-02-10:
> And this devices is supposed to support OTA.

You will need to get ota file in order to support this. Don't know if Develco provides these, you can try asking them.

GD-Dal, 2021-02-10:
Thank you both for your answers.
The On/Off seems to work fine now :))

It runs this code now:
` {
zigbeeModel: ['SMRZB-332'],
model: 'SMRZB-332',
vendor: 'Develco Products A/S',
description: 'Smart Relay DIN',
fromZigbee: [fz.on_off, fz.metering],
toZigbee: [tz.on_off],
meta: {configureKey: 1},
endpoint: (device) => {
    return {'default': 2};
},
configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(2);
    await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
    await reporting.instantaneousDemand(endpoint);
    await reporting.readMeteringMultiplierDivisor(endpoint);
    },

I'm not sure what this all means, tho.
I can email Develco and ask them.
But if we were to obtain such file, how would the upgrade work?

According to the manual, this device supports Phase dropout alarm.
Do you know how to add support for that?

And; how to permanently add support for this device in future zigbee2MQTT upgrades?

Thank you both so much.

sjorge, 2021-02-10:
If we get the files, they need to be added to https://github.com/Koenkk/zigbee-OTA and then we need to set the `ota` property for the device.

Koenkk, 2021-02-10:
> And; how to permanently add support for this device in future zigbee2MQTT upgrades?

Added it. 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

regarding ota, just post the file here and we can add it.

closed 2021-03-20 by github-actions[bot]