Koenkk/zigbee2mqtt #6069: Feature request: central enabling of simulated properties

flatsiedatsie, 2021-02-03:
## Is your feature request related to a problem? Please describe

The WebThings Gateway now has an easy to use Zigbee2MQTT plugin, thanks to the exposes functionality. Since this plugin is fully GUI based, end users can't really modify the `configuration.yaml` file.

When I learnt about the option to have `simulated_brightness` on things like IKEA buttons, I realised it would be great to have that option turned on for all devices. But it seems currently this has to be enabled on a per-device basis in that configuration file?

## Describe the solution you'd like

It would be great to have an option to automatically enable this feature whenever a device supports it. Then I could enable this by default with something like:

`process.env.ZIGBEE2MQTT_CONFIG_SIMULATE_PROPERTIES = true;`

.. before I run Zigbee2MQTT.

## Describe alternatives you've considered
I've concidered programming this feature into the addon. It actually already has some of these heuristics. I'm very glad I discovered Zigbee2MQTT already supports this - really cool. Now it's just a question of making it easy to enable.

I could also perhaps use more `process.env` overrides to force this feature on for each individual device. The downside to that is that a restart of Zigbee2MQTT would be required after a new device is paired before it would appear.

## Additional context
Are there more simulated properties available? I've coded a simulated binary property based on `enum` properties that have both `on` and `off` in them. But I now realise perhaps a `simulated_binary` feature already exists :-D

flatsiedatsie, 2021-02-03:
I tried to set the default of `simulated_brightness` to true using this command, but it didn't work:

`process.env.ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_SIMULATED_BRIGHTNESS = true;`

Koenkk, 2021-02-03:
See `device_options` on how to enable a certain option for all devices: https://www.zigbee2mqtt.io/information/configuration.html

e.g. add to your configuration.yaml:

```yaml
device_options:
  simulated_brightness: true
```

flatsiedatsie, 2021-02-03:
Happy to hear it's possible!

I tried setting that value through an environment variable (see post above), but had no luck. Does this one only work if it's in the actual `configuration.yaml` file?

flatsiedatsie, 2021-02-03:
As a test I added your suggestion to the configuration.yaml file, but saw no effect.

```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
advanced:
  log_level: debug
  log_directory: /home/pi/.webthings/data/zigbee2mqtt-adapter/log
  log_file: Zigbee2MQTT-adapter-%TIMESTAMP%.txt
device_options:
  simulated_brightness: true
```

It I add if to a single device, it does work.

Koenkk, 2021-02-03:
And how does your `devices` section look like?

flatsiedatsie, 2021-02-03:
I was wrong. It's working! Perfect!

flatsiedatsie, 2021-02-03:
Out of curiosity, should `ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_SIMULATED_BRIGHTNESS = true` work?

Koenkk, 2021-02-04:
Ideally yes, but not even sure if that config works correctly at all (it's not used by many users and not well tested). Closing this as OP issue is solved.

closed 2021-02-04 by Koenkk